pyocd/pyOCD #306: question: why the board “frdmkl27z” need to specify "TARGET" when using the gdbserver

Hoohaha, 2017-05-15:
Programming to "frdmkl27z" with gdbserver(). 
if not set "-t kl27z4", there will occur KeyError:
```
uncaught exception: 'kl27z'
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyOCD\tools\gdb_server.py", line 263, in r
un
    frequency=self.args.frequency)
  File "c:\python27\lib\site-packages\pyOCD\board\mbed_board.py", line 211, in c
hooseBoard
    target_override, frequency)
  File "c:\python27\lib\site-packages\pyOCD\board\mbed_board.py", line 185, in g
etAllConnectedBoards
    new_mbed = MbedBoard(dap_access, target_override, frequency)
  File "c:\python27\lib\site-packages\pyOCD\board\mbed_board.py", line 124, in _
_init__
    super(MbedBoard, self).__init__(target, target, link, frequency)
  File "c:\python27\lib\site-packages\pyOCD\board\board.py", line 30, in __init_
_
    self.target = TARGET[target](self.link)
   KeyError: 'kl27z'
```

```bash
C:\Dpc\stationscript>pyocd-gdbserver --version
0.8.0
```
What's the problem?

flit, 2017-05-15:
That's a bug, thanks for reporting!

The problem is that the board list in `pyOCD/board/mbed_board.py` says the target name is "kl27z". But the target dictionary in `pyOCD/target/__init__.py` lists "kl27z4".

flit, 2017-05-15:
Fyi, fixed in #307.

Hoohaha, 2017-05-16:
yes, you are right. Thanks

closed 2017-05-16 by Hoohaha