pyocd/pyOCD #313: Problem with using on frdm-K64F

masoudr, 2017-06-07:
Hi,
I'm trying to use pyOCD to run a debug server, but when I want to start it with `pyocd-gdbserver` or `pyocd-gdbserver -t k64f` but the same error comes up.
```
pyocd-gdbserver
uncaught exception:
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyOCD\tools\gdb_server.py", line 263, in run
    frequency=self.args.frequency)
  File "c:\python27\lib\site-packages\pyOCD\board\mbed_board.py", line 262, in chooseBoard
    mbed.link.open()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 428, in open
    self._interface.open()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\interface\pywinusb_backend.py", line 93, in open
    assert False
AssertionError
```
Thanks.

flit, 2017-06-08:
This is the section of code containing the assert:
```py
            if time() - start > OPEN_TIMEOUT_S:
                # If this timeout has elapsed then another process
                # has locked this device in shared mode. This should
                # not happen.
                assert False
                break
```

This comment suggests that you have another process that is trying to use the board.

@c1728p9 Is this a valid interpretation?

c1728p9, 2017-06-08:
Yep @flit, that is correct. Furthermore, it isn't another instance of pyOCD locking the board since it opens it for exclusive access rather than shared access.

masoudr, 2017-06-09:
Thanks for the answers.
But how can I solve it ? I'm using `mbed-cli` to compile and upload to the board or `online compiler`.

flit, 2019-02-02:
Closing old issue.

closed 2019-02-02 by flit