pyocd/pyOCD #320: mbed-os thread awareness?

paulharris77, 2017-07-12:
Does pyOCD support mbed-os thread awareness?

I've tried debugging an app that spawns a separate thread to main(), but gdb only reports back one thread main().

I'm trying to work out if I'm doing something wrong in gdb (via Eclipse CDT) or if pyOCD doesn't support the mbed-os RTOS threads.

Thanks

Paul

c1728p9, 2017-07-12:
Hi @paulharris77, pyOCD does not support mbed-os thread awareness at the moment, but I'm working on these changes. If you want to try what I have you can find the changes on the rtx5_support branch of my repo - https://github.com/c1728p9/pyOCD/tree/rtx5_support

paulharris77, 2017-07-12:
Great, thanks Ross, will give your version a go tomorrow.

paulharris77, 2017-07-12:
The rtx5_support branch of your repo shows the latest commit is to add rtx2 support. What about rtx5?

c1728p9, 2017-07-12:
Its a typo. It is actually for RTX5. When I created that branch I had cmsis-os2 and rtx5 mixed up with names.

paulharris77, 2017-07-13:
Ah, ok great thanks :-)

paulharris77, 2017-07-14:
Hi Russ,

I built the rtx5_support branch successfully in to a standalone exe, however, it doesn't seem to be able to load/init the RTX2ThreadProvider properly.  It carries on and flashes the target ok, but I only see one thread (main.cpp) as a result in Eclipse.  

I get the following debug out of PyOCD:

DEBUG:root:GDB received query: ['Symbol', '', '#5b']
INFO:root:Attempting to load Argon
INFO:root:Attempting to load RTX2
DEBUG:root:Value for symbol osRtxInfo = <none>
INFO:root:Attempting to load FreeRTOS
DEBUG:root:Value for symbol uxCurrentNumberOfTasks = <none>
DEBUG:root:GDB received query: ['Symbol', '', '#5b']
DEBUG:root:flash op: FlashErase
DEBUG:root:flash op: FlashWrite
.........

Any ideas?

FYI, I'm debugging an mbed LPC1768 target.

Thanks

Paul

c1728p9, 2017-07-15:
How are you building the image you are loading? The elf file needs to have symbols, otherwise, the RTX5 plugin won't work. Symbols should be built in by default.

If you can send me an elf file which exhibits this problem I can give it a try on my end.

flit, 2018-08-10:
RTX5 thread awareness has been added in PR #392.