pyocd/pyOCD #321: The latest version of 0.8.1a1 could not work on windows10

tz-arm, 2017-07-15:
Seems the version of 0.8.1a1 could not work on windows10, it just blocked with no message while execute pyocd-gdbserver directly, but pyocd-gdbserver --version/-h is working well.

tz-arm, 2017-07-15:
It is working now after update mbed-cli to 1.2.0, K64F firmware to 0242 and reset computer.

c1728p9, 2017-07-15:
Hi @tz-arm, if there are no boards connected then pyocd-gdbserver will block until one is connected, which might explain why you were seeing blocking.

You mention that you upgraded your K64F firmware to 0242. What version of firmware did you have before that? If you reload the previous version do you see the same hanging? Older firmware (like 0203) might require you to have the mbed serial driver installed to get a serial port and pyOCD support. Newer firmware does not need this driver.

robotdad, 2017-10-03:
I'm seeing a problem a bit worse than this on Win10. I installed pyOCD per the toolchain docs, but on launching pyocd-gdbserver I get the following error. I can't even run --version.

`pyocd-gdbserver

Traceback (most recent call last):
  File "C:\Users\mgoodner\AppData\Local\Programs\Python\Python36-32\Scripts\pyocd-gdbserver-script.py", line 11, in <module>
    load_entry_point('pyOCD==0.8.1a1', 'console_scripts', 'pyocd-gdbserver')()
  File "c:\users\mgoodner\appdata\local\programs\python\python36-32\lib\site-packages\pkg_resources\__init__.py", line 565, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "c:\users\mgoodner\appdata\local\programs\python\python36-32\lib\site-packages\pkg_resources\__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "c:\users\mgoodner\appdata\local\programs\python\python36-32\lib\site-packages\pkg_resources\__init__.py", line 2291, in load
    return self.resolve()
  File "c:\users\mgoodner\appdata\local\programs\python\python36-32\lib\site-packages\pkg_resources\__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "c:\users\mgoodner\appdata\local\programs\python\python36-32\lib\site-packages\pyOCD\tools\gdb_server.py", line 26
    from . import pyOCD.board.mbed_board
                       ^
SyntaxError: invalid syntax`

c1728p9, 2017-10-03:
Hi @robotdad, it look like you might be using python 3. Currently only pyOCD only works with python 2. Do you see this same problem when using python 2?

robotdad, 2017-10-04:
@c1728p9 that was it. I can launch it now, but it doesn't automatically recognize any of my boards. I've tried the frdm-k64f and a Nucleo L053R8 so far. Those should be recognized with pyocd-gdbserver -l right?

c1728p9, 2017-10-04:
Hi @robotdad, it should be able to recognize the frdm-k64f, but will not be able to recognize the Nucleo L053R8, as this uses STLink which is not supported by pyOCD.

For the FRDM-K64F which version is shown in mbed.htm or details.txt? Older versions of software (before 0242) require the windows serial driver. you can either [install the windows serial driver](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/getting_started/what_need/) or [update your firmware](https://www.nxp.com/products/microcontrollers-and-processors/arm-based-processors-and-mcus/kinetis-cortex-m-mcus/developer-resources/ides-for-kinetis-mcus/opensda-serial-and-debug-adapter:OPENSDA?&tid=vanOpenSDA#FRDM-K64F) if this is the case.

robotdad, 2017-10-04:
@c1728p9 thanks! I forgot to update the firmware on the K64F, it's working fine now on latest Win10. Looks like there is no issue here.

flit, 2018-06-18:
Closing, everything seems resolved.

closed 2018-06-18 by flit