pyocd/pyOCD #328: PyOCD monitor reset do not reset xpsr K64F

alzix, 2017-08-10:
when sending `monitor reset`  command from `arm-none-eabi-gdb` to my `pyocd-gdbserver` the `ipsr` is not cleared as expected.

`ipsr` is also not cleared after flashing MCU.

Tested on K64F: Firmware `0243_k20dx_frdmk64f_0x5000.bin`
pyocd-gdbserver version: 0.8.1a1

GDB commands:
```
p/x $xpsr
$2 = 0x810f000b
monitor reset
monitor halt
p/x $xpsr
$3 = 0x810f000b
```
pyoscd-gdbserver output:
```
DEBUG:root:flash op: FlashWrite
DEBUG:root:flash write addr: 0x1d818
DEBUG:root:GDB vCommand: FlashDone
DEBUG:root:flash op: FlashDone#ea
DEBUG:root:reset stop on Reset
DEBUG:root:Analyze time: 0.128054

[                    ]   0%DEBUG:root:Chip erase count 30, Page erase est count 0
DEBUG:root:Chip erase weight 7.098000, Page erase weight 3.072000
DEBUG:root:Using double buffer page erase program

[                    ]   3%
[=                   ]   7%
[==                  ]  10%
[==                  ]  13%
[===                 ]  17%
[====                ]  20%
[====                ]  23%
[=====               ]  27%
[======              ]  30%
[======              ]  33%
[=======             ]  37%
[========            ]  40%
[========            ]  43%
[=========           ]  47%
[==========          ]  50%
[==========          ]  53%
[===========         ]  57%
[============        ]  60%
[============        ]  63%
[=============       ]  67%
[==============      ]  70%
[==============      ]  73%
[===============     ]  77%
[================    ]  80%
[================    ]  83%
[=================   ]  87%
[==================  ]  90%
[==================  ]  93%
[=================== ]  97%DEBUG:root:Estimated page erase count: 0

[====================] 100%
DEBUG:root:Actual page erase count: 0
DEBUG:root:reset stop on Reset
INFO:root:Programmed 389120 bytes (30 pages) at 58.41 kB/s
DEBUG:root:GDB: write reg pc: 0xC1D0
DEBUG:root:GDB received query: ['Xfer', 'threads', 'read', '', '0,7fb#d0']
DEBUG:root:GDB query threads: offset: 0, size: 2043
DEBUG:root:GDB received query: ['Xfer', 'threads', 'read', '', '0,7fb#d0']
DEBUG:root:GDB query threads: offset: 0, size: 2043
DEBUG:root:GDB received query: ['Rcmd,72657365742068616c74#72']
DEBUG:root:Remote command: reset halt
DEBUG:root:reset stop on Reset
DEBUG:root:GDB received query: ['Rcmd,68616c74#fc']
DEBUG:root:Remote command: halt
DEBUG:root:GDB received query: ['Rcmd,7265736574#37']
DEBUG:root:Remote command: reset
DEBUG:root:reset stop on Reset
DEBUG:root:GDB received query: ['Rcmd,68616c74#fc']
DEBUG:root:Remote command: halt
DEBUG:root:GDB received query: ['Symbol', '', '#5b']
DEBUG:root:GDB: write reg lr: 0xC1D1
DEBUG:root:GDB: write reg sp: 0x1FFF1FC8
DEBUG:root:GDB: write reg pc: 0x751
DEBUG:root:GDB breakpoint Z1 @ c1d0
DEBUG:root:set bkpt type 1 at 0xc1d0
DEBUG:root:using hw bp instead because addr is flash
DEBUG:root:fpb has been enabled
DEBUG:root:GDB breakpoint Z1 @ 15356
DEBUG:root:set bkpt type 1 at 0x15356
DEBUG:root:using hw bp instead because addr is flash
DEBUG:root:GDB vCommand: Cont?
DEBUG:root:GDB vCommand: Cont;c
DEBUG:root:thread_actions={1: None}; default_action=c
DEBUG:root:target resumed
DEBUG:root:state halted
DEBUG:root:GDB: write reg r0: 0x2
DEBUG:root:GDB: write reg r1: 0xA
DEBUG:root:GDB: write reg r2: 0x22400000
DEBUG:root:GDB: write reg r3: 0x2
DEBUG:root:GDB: write reg r4: 0x0
DEBUG:root:GDB: write reg r5: 0x0
DEBUG:root:GDB: write reg r6: 0x0
DEBUG:root:GDB: write reg r7: 0x1FFF1FCC
DEBUG:root:GDB: write reg r8: 0x0
DEBUG:root:GDB: write reg r9: 0x0
DEBUG:root:GDB: write reg r10: 0x0
DEBUG:root:GDB: write reg r11: 0x0
DEBUG:root:GDB: write reg r12: 0xFFFFFFFF
DEBUG:root:GDB: write reg sp: 0x1FFF1FCC
DEBUG:root:GDB: write reg lr: 0x5CC9
DEBUG:root:GDB: write reg xpsr: 0x810F000B
DEBUG:root:GDB: write reg msp: 0x1FFF1FCC
DEBUG:root:GDB: write reg psp: 0x2002FFD8
DEBUG:root:GDB: write reg primask: 0x0
DEBUG:root:GDB: write reg control: 0x1
DEBUG:root:GDB: write reg basepri: 0x20
DEBUG:root:GDB: write reg faultmask: 0x0
DEBUG:root:GDB: write reg fpscr: 0x0
DEBUG:root:GDB: write reg s0: 0x0
DEBUG:root:GDB: write reg s1: 0x0
DEBUG:root:GDB: write reg s2: 0x0
DEBUG:root:GDB: write reg s3: 0x0
DEBUG:root:GDB: write reg s4: 0x0
DEBUG:root:GDB: write reg s5: 0x0
DEBUG:root:GDB: write reg s6: 0x0
DEBUG:root:GDB: write reg s7: 0x0
DEBUG:root:GDB: write reg s8: 0x0
DEBUG:root:GDB: write reg s9: 0x0
DEBUG:root:GDB: write reg s10: 0x0
DEBUG:root:GDB: write reg s11: 0x0
DEBUG:root:GDB: write reg s12: 0x0
DEBUG:root:GDB: write reg s13: 0x0
DEBUG:root:GDB: write reg s14: 0x0
DEBUG:root:GDB: write reg s15: 0x0
DEBUG:root:GDB: write reg s16: 0x0
DEBUG:root:GDB: write reg s17: 0x0
DEBUG:root:GDB: write reg s18: 0x0
DEBUG:root:GDB: write reg s19: 0x0
DEBUG:root:GDB: write reg s20: 0x0
DEBUG:root:GDB: write reg s21: 0x0
DEBUG:root:GDB: write reg s22: 0x0
DEBUG:root:GDB: write reg s23: 0x0
DEBUG:root:GDB: write reg s24: 0x0
DEBUG:root:GDB: write reg s25: 0x0
DEBUG:root:GDB: write reg s26: 0x0
DEBUG:root:GDB: write reg s27: 0x0
DEBUG:root:GDB: write reg s28: 0x0
DEBUG:root:GDB: write reg s29: 0x0
DEBUG:root:GDB: write reg s30: 0x0
DEBUG:root:GDB: write reg s31: 0x0
DEBUG:root:GDB breakpoint z1 @ c1d0
DEBUG:root:remove bkpt at 0xc1d0
DEBUG:root:GDB breakpoint z1 @ 15356
DEBUG:root:remove bkpt at 0x15356
DEBUG:root:GDB breakpoint Z1 @ 15356
DEBUG:root:set bkpt type 1 at 0x15356
DEBUG:root:using hw bp instead because addr is flash
DEBUG:root:GDB vCommand: Cont;c
DEBUG:root:thread_actions={1: None}; default_action=c
DEBUG:root:target resumed
DEBUG:root:state halted
DEBUG:root:GDB received query: ['Xfer', 'threads', 'read', '', '0,7fb#d0']
DEBUG:root:GDB query threads: offset: 0, size: 2043
DEBUG:root:GDB breakpoint z1 @ 15356
DEBUG:root:remove bkpt at 0x15356
DEBUG:root:GDB received query: ['Xfer', 'threads', 'read', '', '0,7fb#d0']
DEBUG:root:GDB query threads: offset: 0, size: 2043
DEBUG:root:GDB received query: ['Xfer', 'threads', 'read', '', '0,7fb#d0']
DEBUG:root:GDB query threads: offset: 0, size: 2043
DEBUG:root:GDB getting register context
DEBUG:root:GDB reg: r0 = 0x15351
DEBUG:root:GDB reg: r1 = 0x0
DEBUG:root:GDB reg: r2 = 0x1D2
DEBUG:root:GDB reg: r3 = 0x40052000
DEBUG:root:GDB reg: r4 = 0xC1E1
DEBUG:root:GDB reg: r5 = 0x0
DEBUG:root:GDB reg: r6 = 0x0
DEBUG:root:GDB reg: r7 = 0x1FFF1FBC
DEBUG:root:GDB reg: r8 = 0x0
DEBUG:root:GDB reg: r9 = 0x0
DEBUG:root:GDB reg: r10 = 0x0
DEBUG:root:GDB reg: r11 = 0x0
DEBUG:root:GDB reg: r12 = 0xFFFFFFFF
DEBUG:root:GDB reg: sp = 0x1FFF1FBC
DEBUG:root:GDB reg: lr = 0xC1E1
DEBUG:root:GDB reg: pc = 0x15356
DEBUG:root:GDB reg: xpsr = 0x810F000B
DEBUG:root:GDB reg: msp = 0x1FFF1FBC
DEBUG:root:GDB reg: psp = 0x2002FFD8
DEBUG:root:GDB reg: primask = 0x1
DEBUG:root:GDB reg: control = 0x1
DEBUG:root:GDB reg: basepri = 0x20
DEBUG:root:GDB reg: faultmask = 0x0
DEBUG:root:GDB reg: fpscr = 0x0
DEBUG:root:GDB reg: s0 = 0x0
DEBUG:root:GDB reg: s1 = 0x0
DEBUG:root:GDB reg: s2 = 0x0
DEBUG:root:GDB reg: s3 = 0x0
DEBUG:root:GDB reg: s4 = 0x0
DEBUG:root:GDB reg: s5 = 0x0
DEBUG:root:GDB reg: s6 = 0x0
DEBUG:root:GDB reg: s7 = 0x0
DEBUG:root:GDB reg: s8 = 0x0
DEBUG:root:GDB reg: s9 = 0x0
DEBUG:root:GDB reg: s10 = 0x0
DEBUG:root:GDB reg: s11 = 0x0
DEBUG:root:GDB reg: s12 = 0x0
DEBUG:root:GDB reg: s13 = 0x0
DEBUG:root:GDB reg: s14 = 0x0
DEBUG:root:GDB reg: s15 = 0x0
DEBUG:root:GDB reg: s16 = 0x0
DEBUG:root:GDB reg: s17 = 0x0
DEBUG:root:GDB reg: s18 = 0x0
DEBUG:root:GDB reg: s19 = 0x0
DEBUG:root:GDB reg: s20 = 0x0
DEBUG:root:GDB reg: s21 = 0x0
DEBUG:root:GDB reg: s22 = 0x0
DEBUG:root:GDB reg: s23 = 0x0
DEBUG:root:GDB reg: s24 = 0x0
DEBUG:root:GDB reg: s25 = 0x0
DEBUG:root:GDB reg: s26 = 0x0
DEBUG:root:GDB reg: s27 = 0x0
DEBUG:root:GDB reg: s28 = 0x0
DEBUG:root:GDB reg: s29 = 0x0
DEBUG:root:GDB reg: s30 = 0x0
DEBUG:root:GDB reg: s31 = 0x0
DEBUG:root:GDB received query: ['Rcmd,7265736574#37']
DEBUG:root:Remote command: reset
DEBUG:root:reset stop on Reset
DEBUG:root:GDB received query: ['Rcmd,68616c74#fc']
DEBUG:root:Remote command: halt

```

Hsiangkai, 2017-08-11:
There is register cache in GDB internal. GDB does not expect register values to be changed by monitor command. So, GDB will not fetch register values from target again after monitor command. Instead, it shows internal cached values to users. You could use flushregs to tell GDB to fetch registers forcibly from target.

GDB commands:

p/x $xpsr
$2 = 0x810f000b
monitor reset
monitor halt
flushregs
p/x $xpsr
$3 = 0x1000000

alzix, 2017-08-12:
i'll try the `flushreg` command. But from what i saw, it seems to be a real issue and not just the GDB internal cache. During boot process `rtx_kernel.c: osKernelInitialize()` asserts `IPSR` value and returns the error.

In addition the log shown above clearly shows that register values are explicitly written after flashing is done.

flit, 2017-08-12:
@Hsiangkai 's description of the root cause is correct. (Good catch, btw!) pyOCD does not (yet) do any caching of register values.

If you look at the log above, it shows gdb writing xpsr to 0x810f000b, then reading back that value:
```
DEBUG:root:GDB: write reg xpsr: 0x810F000B
```
then
```
DEBUG:root:GDB reg: xpsr = 0x810F000B
```

So gdb is setting the xpsr to the "incorrect" value prior to reading it into its cache. The log above where pyOCD is reading the xpsr is the value read directly from the CPU's xpsr.

If RTX is complaining about the xpsr, it's because gdb wrote the invalid value. (I have no idea why gdb would do that.)

Also, you can see that gdb is not reading registers again from pyOCD when you print the xpsr value after you do the reset.
```
DEBUG:root:GDB reg: s31 = 0x0
DEBUG:root:GDB received query: ['Rcmd,7265736574#37']
DEBUG:root:Remote command: reset
DEBUG:root:reset stop on Reset
DEBUG:root:GDB received query: ['Rcmd,68616c74#fc']
DEBUG:root:Remote command: halt
```
That's the end of the log. Notice no read of registers after the reset and halt, where you print xpsr.

Note that wih #296 pyOCD will do internal register caching. But gdb will still have its own cache, and this issue will remain.

closed 2018-06-18 by flit

alzix, 2019-02-03:
CC @avolinski @NirSonnenschein