pyocd/pyOCD #334: Can not program nRF52832 by pyOCD

iotvietmember, 2017-10-09:
Hi all,

I have a development board that has Nordic nRF52832 SoC and CMSIS-DAP Interface.

I tried to use pyocd-flashtool to program for nrf52832 but the tool raise an error as the below image.

[image]

Thanks

flit, 2018-06-18:
@iotvietmember Do you still see this issue with the latest version of pyOCD at the HEAD of master? If so, could you send the hex file you are trying to program? Specifically, I'm wondering if the hex file attempts to program the UICR.

henrikbrixandersen, 2018-11-05:
Does pyOCD support programming/erasing the UICR?

flit, 2019-01-26:
The UICR region has been added to the nRF targets, but it wasn't present at the time this bug was created. Programming the UICR works, since it uses the same procedure as normal flash. But I'm not sure if erasing it works as that requires use of a different register. Chip erase will erase the UICR though.

flit, 2019-02-02:
Closing this issue since the failing hex file is not available. If the problem is seen again and you can provide the information to reproduce it, please reopen this issue.

closed 2019-02-02 by flit