pyocd/pyOCD #335: using pyOCD for resetting K66F board via Raspberry PI

zambs76, 2017-10-10:
Hi,
I want to do soem test automation on a K66 Freedom board and I woud like to use the pyOCD  library to reset the board.
I wrote the following pythoin script:

```
from pyOCD.board import MbedBoard

import logging
logging.basicConfig(level=logging.INFO)

board = MbedBoard.chooseBoard(return_first=True)

target = board.target
target.reset()
```

If I run the script from Windows 10 everything works nicely, but as soon as I am trying to use the script from a Raspberry Pi 3 (using Raspbian) I often have reading errorr like the following one

```
root@raspberrypi:/home/pi/zamb# python reset_board.py
Traceback (most recent call last):
  File "reset_board.py", line 6, in <module>
    board = MbedBoard.chooseBoard(return_first=True)
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 262, in chooseBoard
    mbed.link.open()
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 430, in open
    self._packet_count = self._protocol.dapInfo("PACKET_COUNT")
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/cmsis_dap_core.py", line 91, in dapInfo
    resp = self.interface.read()
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/interface/pyusb_backend.py", line 202, in read
    self.serial_number)
pyOCD.pyDAPAccess.dap_access_api.DeviceError: Device 0311000041114e45000d3003b60f0045a6e1000097969900 read thread exited

```

Does anybody have any hints on how to overcoem thsi problem?

Cheers,
Zambs

c1728p9, 2017-10-10:
Hi @zambs76, I have had a similar experience with any board running on the Raspberry PI. I think it had USB stability issues.

zambs76, 2017-10-25:
thanks for the feedback!

gimcuan, 2018-04-19:
try to lower dap frequency with `-t` option.

cederom, 2018-04-19:
You can also use CDC / Virtual COM Port channel of DAPLink and simply send 'BREAK' command to reset Target..

flit, 2019-02-02:
Closing this issue as it seems to be a Raspberry Pi problem with USB. If encountered again, you might try passing `--daparg limit_packets=True`, which is also necessary when running under some VMs (notably VirtualBox under Windows).

closed 2019-02-02 by flit