pyocd/pyOCD #346: Trying to communicate with the FRDM-K64F using pyOCD

hellgheast, 2017-12-24:
Here's the error i got when trying to communicate with the FRDM-K64F.
I'm on Fedora 26.

[root@localhost bin]# pyocd-gdbserver 
INFO:root:DAP SWD MODE initialised
INFO:root:K64F not in secure state
uncaught exception: 
Traceback (most recent call last):
  File "/root/.local/lib/python3.6/site-packages/pyOCD/tools/gdb_server.py", line 263, in run
    frequency=self.args.frequency)
  File "/root/.local/lib/python3.6/site-packages/pyOCD/board/mbed_board.py", line 272, in chooseBoard
    mbed.init()
  File "/root/.local/lib/python3.6/site-packages/pyOCD/board/board.py", line 51, in init
    self.target.init()
  File "/root/.local/lib/python3.6/site-packages/pyOCD/target/target_kinetis.py", line 123, in init
    self.aps[0].init(bus_accessible=True)
  File "/root/.local/lib/python3.6/site-packages/pyOCD/coresight/ap.py", line 111, in init
    super(MEM_AP, self).init(bus_accessible)
  File "/root/.local/lib/python3.6/site-packages/pyOCD/coresight/ap.py", line 87, in init
    self.init_rom_table()
  File "/root/.local/lib/python3.6/site-packages/pyOCD/coresight/ap.py", line 92, in init_rom_table
    self.rom_table.init()
  File "/root/.local/lib/python3.6/site-packages/pyOCD/coresight/rom_table.py", line 178, in init
    self.read_id_registers()
  File "/root/.local/lib/python3.6/site-packages/pyOCD/coresight/rom_table.py", line 116, in read_id_registers
    regs = self.ap.readBlockMemoryAligned32(self.top_address + PIDR4, IDR_COUNT)
  File "/root/.local/lib/python3.6/site-packages/pyOCD/coresight/ap.py", line 344, in readBlockMemoryAligned32
    resp += self._readBlock32(addr, n/4)
  File "/root/.local/lib/python3.6/site-packages/pyOCD/coresight/ap.py", line 205, in _readBlock32
    resp = self.link.reg_read_repeat(size, reg)
  File "/root/.local/lib/python3.6/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 591, in reg_read_repeat
    assert isinstance(num_repeats, six.integer_types)
AssertionError

Has anyone got any similar problem ?

hellgheast, 2018-01-03:
Got fixed by using openOCD instead of pyOCD

flit, 2018-06-20:
@hellgheast Fyi, the problem is that you were using Python 3.6. pyOCD does not yet support Python 3, so please use the latest Python 2.7 (2.7.15 at the time of this writing).

closed 2018-06-26 by flit