pyocd/pyOCD #350: Does pyocd support Cortex-M7 target?

Hoohaha, 2018-03-20:
I am trying to support a new board with arm cortex-m7 core.

That raise an exception when i test gdb-server:

```
uncaught exception: 3111
Traceback (most recent call last):
  File "C:\Mycode\pyOCD\pyOCD\tools\gdb_server.py", line 299, in run
    frequency=self.args.frequency)
  File "pyOCD\board\mbed_board.py", line 274, in chooseBoard
    mbed.init()
  File "pyOCD\board\board.py", line 50, in init
    self.target.init()
  File "pyOCD\target\target_MIMXRT1021.py", line 83, in init
    super(MIMXRT1020, self).init()
  File "pyOCD\core\coresight_target.py", line 96, in init
    core0.init()
  File "pyOCD\coresight\cortex_m.py", line 313, in init
    self.readCoreType()
  File "pyOCD\coresight\cortex_m.py", line 362, in readCoreType
    logging.info("CPU core is %s", CORE_TYPE_NAME[self.core_type])
KeyError: 3111
```

This is pre-defined core types i found in cortex_m.py:
```python
# CPUID PARTNO values
ARM_CortexM0 = 0xC20
ARM_CortexM1 = 0xC21
ARM_CortexM3 = 0xC23
ARM_CortexM4 = 0xC24
ARM_CortexM0p = 0xC60

# User-friendly names for core types.
CORE_TYPE_NAME = {
                 ARM_CortexM0 : "Cortex-M0",
                 ARM_CortexM1 : "Cortex-M1",
                 ARM_CortexM3 : "Cortex-M3",
                 ARM_CortexM4 : "Cortex-M4",
                 ARM_CortexM0p : "Cortex-M0+"
               }
```

Besides i want to know does we have any documentation about the pyocd porting?

Thanks

Hoohaha, 2018-03-20:
@c1728p9 @flit

flit, 2018-03-20:
pyOCD doesn't support the CM7 on the master branch, but I have support implemented on the [dev_cm7](https://github.com/flit/pyOCD/tree/dev_cm7) branch of my fork. The only thing remaining is finishing and testing support for double-precision FPU registers (D0-D15).

Hoohaha, 2018-03-21:
@flit Thanks for your reply.

hakehuang, 2018-06-07:
@flit Hi Chris, do you have plan to support RT1020 board?

hakehuang, 2018-06-11:
Thanks @flit

markfirmware, 2018-07-03:
@flit Chris, how can I get started with https://github.com/flit/pyOCD/tree/dev_cm7
and the m7 board [atsame70-xpld](http://www.microchip.com/developmenttools/ProductDetails/atsame70-xpld)?

    WARNING:root:Unsupported board found ATML                        
    INFO:root:DAP SWD MODE initialized                               
    INFO:root:ROM table #0 @ 0xe00fd000 cidr=b105100d pidr=9f000
    ...

I get stuck on ram_regions.

Thanks, Mark

flit, 2018-07-03:
@markfirmware Basically, you need to add a new target for the ATSAME70. Within a few months, we'll be moving to support any target with a CMSIS-Pack. But for now you still have to manually add new targets. (This also should be officially documented, of course, but it doesn't make much sense to write docs for something that will change shortly.)

These are the steps:

1. Fork my pyOCD fork, so you get the dev_cm7 branch.
2. Create a new child branch of dev_cm7 to work on.
3. Create a new CoreSightTarget subclass and Flash subclass in a file under `pyOCD/target/`. You can copy one of the existing target files like `pyOCD/target/target_ncs36510.py` and rename the classes.
4. For the flash algo, please use the attached [py_blob_orig.txt](https://github.com/mbedmicro/pyOCD/files/2159708/py_blob_orig.txt). Just replace the `flash_algo` dict in your new target support file with the one from this file. (I generated this from the .FLM file in the [SAME70 CMSIS-Pack](http://www.keil.com/dd2/pack/) using `tools/generate_blobs.py` from the [FlashAlgo repo](https://github.com/mbedmicro/FlashAlgo).)
5. Edit `pyOCD/target/__init__.py` to add your new target and flash classes to the `TARGET` and `FLASH` dicts.
6. Test and create a PR 😄 

Some targets require special handling for resets, remapping of memory (usually to move a ROM out of the way), or similar. Hopefully the SAME70 doesn't need any of this, but if it does I can help you.

Since (presumably) you aren't using DAPLink firmare for the debug probe, there isn't a way that pyOCD can auto-detect the target. Thus, you'll have to pass the `-t/--target` option to the pyOCD tools.

markfirmware, 2018-07-04:
Thanks Chris - so far so good. I copied basic_test.py to same70_test.py and commented out the last two steps of flash r/w testing and file flashing. I'll tackle those next. Before that though, everything works except I had to comment out
https://github.com/markfirmware/pyOCD/blob/dev_cm7_same70/test/same70_test.py#L85-#L90
regarding 'primask'. It was producing:

```
------ TEST READ / WRITE CORE REGISTER ------
initial pc: 0x400244
now pc: 0x3D82
initial pc value rewritten: 0x400244
MSP = 0x2045ffe0; PSP = 0x00000000
CONTROL = 0x00; FAULTMASK = 0x00; BASEPRI = 0x00; PRIMASK = 0x00
Traceback (most recent call last):
  File "test/same70_test.py", line 223, in <module>
    basic_test(None, file)
  File "test/same70_test.py", line 85, in basic_test
    target.writeCoreRegister('primask', 1)
  File "/home/pi/github.com/markfirmware/pyOCD/pyOCD/core/coresight_target.py", line 158, in writeCoreRegister
    return self.selected_core.writeCoreRegister(id, data)
  File "/home/pi/github.com/markfirmware/pyOCD/pyOCD/coresight/cortex_m.py", line 802, in writeCoreRegister
    self.writeCoreRegisterRaw(regIndex, data)
  File "/home/pi/github.com/markfirmware/pyOCD/pyOCD/coresight/cortex_m.py", line 810, in writeCoreRegisterRaw
    self.writeCoreRegistersRaw([reg], [data])
  File "/home/pi/github.com/markfirmware/pyOCD/pyOCD/coresight/cortex_m.py", line 859, in writeCoreRegistersRaw
    data = (specialRegValue & mask) | ((data & 0xff) << shift)
TypeError: unsupported operand type(s) for &: 'NoneType' and 'long'

```

Can you advise regarding primask? Thanks, Mark.

P.S. I also changed https://github.com/markfirmware/pyOCD/blob/dev_cm7_same70/pyOCD/board/mbed_board.py#L54-L56
to handle the unknown board. Is there a better way?

Hoohaha, 2018-07-16:
Hi @flit ,

I tested the branch dev_cm7 with evkbimxrt1050, but there someting is wrong. I have no idea about this, Can you help to fix that? I would really appreciate. Thanks.

>  pyocd-flashtool -a 0x60000000 C:/Dpc/FreeMV/system/app_data/binaries/evkbimxrt1050.bin
INFO:root:DAP SWD MODE initialized
INFO:root:ROM table #0 @ 0xe00fd000 cidr=b105100d pidr=8e88c
INFO:root:[0]<e00fe000: cidr=b105100d, pidr=4000bb4c8, class=1>
INFO:root:ROM table #1 @ 0xe00fe000 cidr=b105100d pidr=4000bb4c8
INFO:root:[0]<e00ff000: cidr=b105100d, pidr=4000bb4c7, class=1>
INFO:root:ROM table #2 @ 0xe00ff000 cidr=b105100d pidr=4000bb4c7
INFO:root:[0]<e000e000:SCS-M4 cidr=b105e00d, pidr=4000bb00c, class=14>
INFO:root:[1]<e0001000:DWT cidr=b105e00d, pidr=4000bb002, class=14>
INFO:root:[2]<e0002000:FPB cidr=b105e00d, pidr=4000bb00e, class=14>
INFO:root:[3]<e0000000:ITM cidr=b105e00d, pidr=4000bb001, class=14>
INFO:root:[1]<e0041000:ETM-M7 cidr=b105900d, pidr=4001bb975, class=9, devtype=13, devid=0>
INFO:root:[2]<e0042000:ECT cidr=b105900d, pidr=4004bb906, class=9, devtype=14, devid=40800>
INFO:root:[1]<e0040000:TPIU-M7 cidr=b105900d, pidr=4000bb9a9, class=9, devtype=11, devid=ca1>
INFO:root:[2]<e0043000:CSTF cidr=b105f00d, pidr=4001bb101, class=15>
INFO:root:CPU core is Cortex-M7
INFO:root:FPU present: FPv5-DP
INFO:root:8 hardware breakpoints, 1 literal comparators
INFO:root:4 hardware watchpoints
ERROR:root:target exception during uninit:
Traceback (most recent call last):
  File "c:\mycode\flit-pyocd\pyOCD\board\board.py", line 64, in uninit
    self.target.resume()
  File "c:\mycode\flit-pyocd\pyOCD\core\coresight_target.py", line 120, in resume
    return self.selected_core.resume()
  File "c:\mycode\flit-pyocd\pyOCD\coresight\cortex_m.py", line 650, in resume
    if self.getState() != Target.TARGET_HALTED:
  File "c:\mycode\flit-pyocd\pyOCD\coresight\cortex_m.py", line 618, in getState
    dhcsr = self.readMemory(CortexM.DHCSR)
  File "c:\mycode\flit-pyocd\pyOCD\coresight\cortex_m.py", line 462, in readMemory
    result = self.ap.readMemory(addr, transfer_size, now)
  File "c:\mycode\flit-pyocd\pyOCD\coresight\ap.py", line 194, in readMemory
    result = readMemCb()
  File "c:\mycode\flit-pyocd\pyOCD\coresight\ap.py", line 176, in readMemCb
    res = result_cb()
  File "c:\mycode\flit-pyocd\pyOCD\coresight\dap.py", line 284, in readAPCb
    result = result_cb()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 658, in read_reg_cb
    res = transfer.get_result()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 119, in get_result
    self.daplink.flush()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 589, in flush
    self._read_packet()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 744, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 419, in decode_data
    data = self._decode_transfer_data(data)
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 315, in _decode_transfer_data
    raise DAPAccessIntf.TransferFaultError()
TransferFaultError: SWD/JTAG Transfer Fault @ 0xe000edf0
ERROR:root:link exception during link disconnect:
Traceback (most recent call last):
  File "c:\mycode\flit-pyocd\pyOCD\board\board.py", line 74, in uninit
    self.link.disconnect()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 627, in disconnect
    self.flush()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 589, in flush
    self._read_packet()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 744, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 419, in decode_data
    data = self._decode_transfer_data(data)
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 315, in _decode_transfer_data
    raise DAPAccessIntf.TransferFaultError()
TransferFaultError: SWD/JTAG Transfer Fault
Traceback (most recent call last):
[... 4 lines cut ...]
  File "c:\mycode\flit-pyocd\pyOCD\flash\flash.py", line 289, in flashBlock
    info = fb.program(chip_erase, progress_cb, smart_flash, fast_verify)
  File "c:\mycode\flit-pyocd\pyOCD\flash\flash_builder.py", line 212, in program
    self.flash.init()
  File "c:\mycode\flit-pyocd\pyOCD\flash\flash.py", line 113, in init
    self.target.setTargetState("PROGRAM")
  File "c:\mycode\flit-pyocd\pyOCD\core\coresight_target.py", line 191, in setTargetState
    return self.selected_core.setTargetState(state)
  File "c:\mycode\flit-pyocd\pyOCD\coresight\cortex_m.py", line 612, in setTargetState
    self.resetStopOnReset(True)
  File "c:\mycode\flit-pyocd\pyOCD\coresight\cortex_m.py", line 601, in resetStopOnReset
    self.reset(software_reset)
  File "c:\mycode\flit-pyocd\pyOCD\coresight\cortex_m.py", line 583, in reset
    self.dp.flush()
  File "c:\mycode\flit-pyocd\pyOCD\coresight\dap.py", line 125, in flush
    self.link.flush()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 589, in flush
    self._read_packet()
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 744, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 417, in decode_data
    data = self._decode_transfer_block_data(data)
  File "c:\mycode\flit-pyocd\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 381, in _decode_transfer_block_data
    raise DAPAccessIntf.TransferFaultError()
pyOCD.pyDAPAccess.dap_access_api.TransferFaultError: SWD/JTAG Transfer Fault

Hoohaha, 2018-07-31:
Looks like the problem is occured after the `self.reset(software_reset)`, i try to fix but i failed. @flit Could you spare time to look on the imxrt1050 target? I think this is a beginning or example that i can porting pyOCD to other comming RT boards, like RT1020/RT1060 etc. Thank you very mcuh.

flit, 2018-08-11:
Apologies for the delayed reply. 

@markfirmware The latest commit on the dev_cm7 branch should fix the problem with primask. To force the target, use the `target_override` parameter on `MbedBoard.chooseBoard()`.

@Hoohaha I'm sorry, I don't have an answer for why it's failing to read the DHCSR register. However, I'm about to start debugging RT1050 and CM7 support.

Note that I'm also going to be rebasing dev_cm7 onto master shortly (probably this weekend), to pick up the init sequence improvements.

unsignedint, 2018-12-02:
I noticed there was _some_ support for the RT1050 device landed recently. I gave it a try on my Eval Kit, but didn't have any success (it's configured to run from HyperFlash). I was wondering if there is  anything I could help with to assist with the development for the 1050 target?

flit, 2018-12-03:
@unsignedint Yep. The problems seem to be related to the flash algorithm, as working with code running from RAM seems ok. I'm working with NXP folks to get the flash algo source and fix this. I've created #463 to track this particular issue with RT1050. The same problem probably exists with RT1020, but I don't have a board to test with.

Since basic Cortex-M7 support is implemented, I'm closing this issue.

closed 2018-12-03 by flit