pyocd/pyOCD #351: "Remote failure reply: E01" with FRDM-K64F

stefanhoelzl, 2018-04-15:
Hi,

I am using ```pyOCD (0.8.0)``` with a ```FRDM-K64F``` and ```CLion (2018.1)``` under ```macOS 10.13.3``` for debugging.

I start the GDB Server via
```pyocd-gdbserver -t k64f -o``` 
and then start a remote GDB Session in CLion.
It stops at program entry and I can step through with ```(gdb) si```
But when entering ```(gdb) continue``` I get the following error:
```Remote failure reply: E01``` in my GDB remote session.

In the terminal where the pyocd-gdbserver runs, is the following exception:
(It is always a ```ERROR:root:Unhandled exception in handleMsg```which results in a ```TransferFaultError``` but it also occurs in different places.)
```console
ERROR:root:Unhandled exception in handleMsg: 
Traceback (most recent call last):
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 413, in handleMsg
    return self.stopReasonQuery(), 0
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 540, in stopReasonQuery
    return self.createRSPPacket(self.target.getTResponse())
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/target/coresight_target.py", line 198, in getTResponse
    return self.selected_core.getTResponse(forceSignal)
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/coresight/cortex_m.py", line 871, in getTResponse
    response = 'T' + conversion.byteToHex2(self.getSignalValue())
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/coresight/cortex_m.py", line 885, in getSignalValue
    fault = self.readCoreRegister('xpsr') & 0xff
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/coresight/cortex_m.py", line 597, in readCoreRegister
    regValue = self.readCoreRegisterRaw(regIndex)
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/coresight/cortex_m.py", line 623, in readCoreRegisterRaw
    vals = self.readCoreRegistersRaw([reg])
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/coresight/cortex_m.py", line 666, in readCoreRegistersRaw
    dhcsr_val = dhcsr_cb()
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/coresight/cortex_m.py", line 405, in readMemoryCb
    return self.bp_manager.filter_memory(addr, transfer_size, result())
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/coresight/ap.py", line 160, in readMemCb
    res = result_cb()
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/coresight/dap.py", line 280, in readAPCb
    result = result_cb()
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 565, in read_reg_cb
    res = transfer.get_result()
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 116, in get_result
    self.daplink.flush()
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 500, in flush
    self._read_packet()
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 651, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 333, in decode_data
    data = self._decode_transfer_data(data)
  File "/Users/stefan/Development/pyocd/lib/python2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 276, in _decode_transfer_data
    raise DAPAccessIntf.TransferFaultError()
TransferFaultError
```

flit, 2018-05-21:
@stefanhoelzl Sorry for taking a while to follow up on this. Can you please provide the full log from pyocd-gdbserver? Also, can you please try with the latest version 0.9.0 release of pyOCD? Thanks.

TurhanOz, 2018-07-05:
@stefanhoelzl have you find a solution for your issue ? Als, could you share how you configured CLion for remote debugging on K64F please ?
Cheers

stefanhoelzl, 2018-07-05:
@TurhanOz no i haven't tried it further.
Switched to the OpenSDA Firmware for the FRDM-K64F and using the SEGGER JLinkGDBServer.
In CLion I am using a GDB Configuration with the gdbserver as the remote target.
Thats the same configuration I was using with the pyocd-gdbserver

TurhanOz, 2018-07-06:
Hi @stefanhoelzl , I'm struggling with Clion configuration. So far, I'm able to build and flash my board directly within Clion. I'm also able to debug with command line, but impossible to configure Clion to interactively debug (with break point on the source code from the IDE for instance). Could you help me by sharing screen shot of your configuration please? Thanks a lot, cheers

stefanhoelzl, 2018-07-08:
@TurhanOz  here is my clion configuration

[image]

TurhanOz, 2018-07-09:
@stefanhoelzl thanks. I'll try asap :) cheers