pyocd/pyOCD #365: Can’t connect to microbit from macOS

simonjwright, 2018-06-24:
Hi,

With `sudo -H pip install pyocd`, I get
```
lockheed:~ simon$ pyocd-gdbserver --version
0.10.0
lockheed:~ simon$ pyocd-gdbserver -l
0 => Microbit [nrf51] boardId => 9900023432044e45001780110000005800000000cfcf28bd
lockheed:~ simon$ pyocd-gdbserver
uncaught exception: Unable to open device
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/pyOCD/tools/gdb_server.py", line 301, in run
    frequency=self.args.frequency)
  File "/Library/Python/2.7/site-packages/pyOCD/board/mbed_board.py", line 209, in chooseBoard
    mbed.link.open()
  File "/Library/Python/2.7/site-packages/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 513, in open
    self._interface.open()
  File "/Library/Python/2.7/site-packages/pyOCD/pyDAPAccess/interface/hidapi_backend.py", line 52, in open
    raise DAPAccessIntf.DeviceError("Unable to open device")
DeviceError: Unable to open device
```
Obviously I’m doing something stupid ...

flit, 2018-06-24:
Hi @simonjwright, I just tested with macOS 10.3.5 connecting to a micro:bit without any problems. It seems to be something particular to your system, but I'm not sure what the problem might be. (I would normally say to run `pyocd-gdbserver` with `-ddebug` to get the debug log, but you're not getting far enough for that to be useful.)

Is there any chance another process has opened the micro:bit USB device?

You might double check that the latest version of hidapi is installed by running `pip install --upgrade hidapi`. This should only be a problem if you had an older version of pyOCD or hidapi previously installed and upgraded pyOCD.

(Btw, you shouldn't need to use `sudo` to install Python packages on macOS. If `pip` doesn't work without `sudo`, then something is not right with permissions for your Python install. This is true for both the official python.org releases, as well as Homebrew Python installs. I doubt it could cause this connection problem, though.)

simonjwright, 2018-06-25:
10.13.5, I guess?

You may well be right about something else having the device open, but don’t know how to find out what.

I tried opening with libusb using the vendor, product id reported by System Information - OK (after realising that System Information reports in the order product/vendor, oops)

I checked hidapi, latest version OK.

Are you sure about pip/sudo? `pip install --user` would be OK, of course (but pyocd still doesn’t work).

I’m now up and running with J-Link, so my urgency has diminished considerably.

simonjwright, 2018-06-27:
I’ve downloaded the [Community Edition 2018 Ada compiler](https://www.adacore.com/community) and it includes a version of pyocd that works just fine. Possibly because it comes with its own Python: don’t know what the difference is, both are 2.7.10, but the one that works uses `libpython2.7.dylib`, the system one doesn’t.

flit, 2018-06-27:
That's very strange, I'm not sure what it indicates. Thanks for the feedback.

Fyi, you can use the `ioreg` command to find the process IDs of processes that have opened USB devices.

Running this command:

```ioreg -n CMSIS-DAP -l -r | grep 'IOUSBHostInterface|Product|IOUserClientCreator'```

Will show output line this:

```
+-o CMSIS-DAP@3  <class IOUSBHostInterface, id 0x1001681e6, registered, matched, active, busy 0 (30 ms), retain 12>
  |   "idProduct" = 516
    |   "IOProviderClass" = "IOUSBHostInterface"
    |   "Product" = "DAPLink CMSIS-DAP"
    |   "ProductID" = 516
    |     "Product" = "DAPLink CMSIS-DAP"
    |     "ProductID" = 516
    |     "IOUserClientCreator" = "pid 106, loginwindow"
          "IOUserClientCreator" = "pid 68785, python"
```

Here you can see both loginwindow (not sure why) and python (running pyocd-tool) have opened the CMSIS-DAP HID interface on this device.

Note that this only works if the HID interface's name exactly matches "CMSIS-DAP"; `ioreg` doesn't do substring matches on object names.

simonjwright, 2018-06-28:
What I see without `pyocd-tool` is
```
+-o CMSIS-DAP@3  <class IOUSBHostInterface, id 0x1000092a3, registered, matched, active, busy 0 (682 ms), retain 12>
  |   "idProduct" = 516
    |   "IOProviderClass" = "IOUSBHostInterface"
    |   "Product" = "DAPLink CMSIS-DAP"
    |   "ProductID" = 516
    |     "IOUserClientCreator" = "pid 120, loginwindow"
          "Product" = "DAPLink CMSIS-DAP"
          "ProductID" = 516
      "idProduct" = 516
```
With the working python (here called `gpspython`), I see
```
+-o CMSIS-DAP@3  <class IOUSBHostInterface, id 0x1000092a3, registered, matched, active, busy 0 (683 ms), retain 12>
  |   "idProduct" = 516
    |   "IOProviderClass" = "IOUSBHostInterface"
    |   "Product" = "DAPLink CMSIS-DAP"
    |   "ProductID" = 516
    |     "IOUserClientCreator" = "pid 120, loginwindow"
    |     "Product" = "DAPLink CMSIS-DAP"
    |     "ProductID" = 516
    |     "IOUserClientCreator" = "pid 14812, gpspython"
          "IOUserClientCreator" = "pid 14812, gpspython"
      "idProduct" = 516
```

flit, 2018-08-09:
Following up on this… We had a case where a colleague was experiencing the same problem, so I had a chance to debug locally. It seems to have something to do with the specific Python installation. However, I was never able to determine the root cause aside from perhaps being permission related, not multiple processes opening a device. In my colleague's case, the system Python didn't work but a homebrew'd Python worked fine. So installing a new copy of Python may fix the problem. This also explains why gpspython worked for you.

tomchisholm, 2018-10-23:
Just in case anyone else has the same issue, I solved it by installing python3 with brew and pyOCD using 'pip3' instead of 'pip', forcing it to use python 3. Seems there's some issue with pyOCD using the default python 2.7 installed with macOS.

sg-, 2019-01-13:
Wonder what is different in the version from homebrew. https://github.com/mbedmicro/pyOCD/issues/371 installed from brew and it worked too. I now have the same problem and have tried a few different python execs and no dice.

This is the output after updating the system wide python from 2.7.10 to 2.7.15 on macOS Mojave 10.14.2. The 2.7.10 version had the same problem.

```
:mbed-workspace samgro01$ which python
/usr/local/bin/python
:mbed-workspace samgro01$ python --version
Python 2.7.15
:mbed-workspace samgro01$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
appdirs (1.4.3)
asn1ate (0.6.0)
asn1crypto (0.24.0)
beautifulsoup4 (4.6.3)
certifi (2018.11.29)
cffi (1.11.5)
chardet (3.0.4)
colorama (0.3.9)
cryptography (2.4.2)
ecdsa (0.13)
fasteners (0.14.1)
funcsigs (1.0.2)
functools32 (3.2.3.post2)
future (0.16.0)
futures (3.2.0)
fuzzywuzzy (0.17.0)
hidapi (0.7.99.post21)
icetea (1.0.2)
intelhex (2.2.1)
intervaltree (2.1.0)
ipaddress (1.0.22)
Jinja2 (2.10)
jsonmerge (1.5.2)
jsonschema (2.6.0)
junit-xml (1.8)
lockfile (0.12.2)
Logbook (1.4.1)
lxml (4.2.5)
manifest-tool (1.4.6)
MarkupSafe (1.1.0)
mbed-cloud-sdk (2.0.1)
mbed-flasher (0.9.2)
mbed-greentea (1.5.0)
mbed-host-tests (1.4.4)
mbed-ls (1.6.2)
mock (2.0.0)
monotonic (1.5)
pbr (5.1.1)
pip (9.0.3)
prettytable (0.7.2)
protobuf (3.5.2.post1)
py (1.7.0)
pyasn1 (0.2.3)
pycparser (2.19)
pyelftools (0.25)
pyocd (0.14.3)
pyOpenSSL (18.0.0)
pyparsing (2.3.0)
pyserial (3.4)
pyshark-legacy (0.3.8)
python-dateutil (2.7.5)
python-dotenv (0.10.1)
pyusb (1.0.2)
PyYAML (3.13)
requests (2.20.1)
semver (2.8.1)
setuptools (40.6.3)
six (1.11.0)
trollius (1.0.4)
[... 107 lines cut ...]
xlwt (1.3.0)
xmltodict (0.11.0)
yattag (1.10.0)
You are using pip version 9.0.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
(base) C02WW1FEHV2T-3:mbed-workspace samgro01$ pyocd gdb -vv
0000214:WARNING:usb:STLink probes are not supported because no libusb library was found.
0000214:INFO:board:Target type is lpc1768
0000226:ERROR:__main__:uncaught exception: Unable to open device
Traceback (most recent call last):
  File "/Users/samgro01/Library/Python/2.7/lib/python/site-packages/pyocd/__main__.py", line 309, in run
    self._commands[self._args.cmd]()
  File "/Users/samgro01/Library/Python/2.7/lib/python/site-packages/pyocd/__main__.py", line 459, in do_gdbserver
    **sessionOptions)
  File "/Users/samgro01/Library/Python/2.7/lib/python/site-packages/pyocd/core/helpers.py", line 199, in session_with_chosen_probe
    session.open()
  File "/Users/samgro01/Library/Python/2.7/lib/python/site-packages/pyocd/core/session.py", line 155, in open
    self._probe.open()
  File "/Users/samgro01/Library/Python/2.7/lib/python/site-packages/pyocd/probe/cmsis_dap_probe.py", line 140, in open
    six.raise_from(self._convert_exception(exc), exc)
  File "/Users/samgro01/Library/Python/2.7/lib/python/site-packages/six.py", line 737, in raise_from
    raise value
ProbeError: Unable to open device
(base) :mbed-workspace samgro01$ 
```

sg-, 2019-01-13:
looking at this further after downgrading to 0.11.0 (a version that was previously working) I ran into the socket in use problem which I'd come across before.

```
:wip samgro01$ pyocd-gdbserver 
0000099:INFO:cmsis_dap_core:DAP SWD MODE initialized
0000133:INFO:rom_table:ROM table #0 @ 0xe00ff000 cidr=b105100d pidr=0
0000140:INFO:rom_table:[0]<e000e000: cidr=b105e00d, pidr=4002bb000, class=14>
0000143:INFO:rom_table:[1]<e0001000: cidr=b105e00d, pidr=4002bb002, class=14>
0000146:INFO:rom_table:[2]<e0002000:FPB cidr=b105e00d, pidr=4002bb003, class=14>
0000149:INFO:rom_table:[3]<e0000000: cidr=b105e00d, pidr=4002bb001, class=14>
0000155:INFO:rom_table:[4]<e0040000: cidr=b105900d, pidr=4002bb923, class=9, devtype=11, devid=ca1>
0000162:INFO:rom_table:[5]<e0041000: cidr=b105900d, pidr=4002bb924, class=9, devtype=13, devid=0>
0000167:INFO:cortex_m:CPU core is Cortex-M3
0000171:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000179:INFO:dwt:4 hardware watchpoints
0000186:INFO:semihost:Telnet: server started on port 4444
0000186:INFO:gdbserver:GDB server started at port:3333
0000188:ERROR:gdbserver:Unexpected exception: [Errno 48] Address already in use
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 426, in run
    connected = self.abstract_socket.connect()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyOCD/gdbserver/gdb_socket.py", line 37, in connect
    self.init()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyOCD/gdbserver/gdb_socket.py", line 32, in init
    self.s.bind((self.host, self.port))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 48] Address already in use
```

I'd seen this before and was able to find the process with a handle to this port and kill it.
```
C02WW1FEHV2T-3:wip samgro01$ lsof -i :3333
COMMAND     PID     USER   FD   TYPE             DEVICE SIZE/OFF NODE NAME
JuniperSe 45820 samgro01    4u  IPv4 0xb616fdf7b8191b45      0t0  TCP localhost:dec-notes (LISTEN)
C02WW1FEHV2T-3:wip samgro01$ lsof -i :4444
C02WW1FEHV2T-3:wip samgro01$ lsof -i :3333
COMMAND     PID     USER   FD   TYPE             DEVICE SIZE/OFF NODE NAME
JuniperSe 45820 samgro01    4u  IPv4 0xb616fdf7b8191b45      0t0  TCP localhost:dec-notes (LISTEN)
C02WW1FEHV2T-3:wip samgro01$ kill 45820
C02WW1FEHV2T-3:wip samgro01$ lsof -i :3333
C02WW1FEHV2T-3:wip samgro01$ pyocd-gdbserver 
0000105:INFO:cmsis_dap_core:DAP SWD MODE initialized
0000138:INFO:rom_table:ROM table #0 @ 0xe00ff000 cidr=b105100d pidr=0
0000145:INFO:rom_table:[0]<e000e000: cidr=b105e00d, pidr=4002bb000, class=14>
0000148:INFO:rom_table:[1]<e0001000: cidr=b105e00d, pidr=4002bb002, class=14>
0000151:INFO:rom_table:[2]<e0002000:FPB cidr=b105e00d, pidr=4002bb003, class=14>
0000155:INFO:rom_table:[3]<e0000000: cidr=b105e00d, pidr=4002bb001, class=14>
0000162:INFO:rom_table:[4]<e0040000: cidr=b105900d, pidr=4002bb923, class=9, devtype=11, devid=ca1>
0000169:INFO:rom_table:[5]<e0041000: cidr=b105900d, pidr=4002bb924, class=9, devtype=13, devid=0>
0000175:INFO:cortex_m:CPU core is Cortex-M3
0000180:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000188:INFO:dwt:4 hardware watchpoints
0000196:INFO:semihost:Telnet: server started on port 4444
0000196:INFO:gdbserver:GDB server started at port:3333
^C0004711:INFO:semihost:Telnet: server stopped
0004731:INFO:gdbserver:GDB server thread killed
C02WW1FEHV2T-3:wip samgro01$ 
```

Then upgrading back to 0.14.3 and the default python installation started working again.

```
:wip samgro01$ pip install -U pyocd
Collecting pyocd
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/6c/ae/0baea35a5d061315af00b4b461efa9f1f0c897017230442939f18e9e6885/pyocd-0.14.3.tar.gz (1.0MB)
    100% |████████████████████████████████| 1.0MB 984kB/s 
Requirement already up-to-date: intelhex in /Users/samgro01/Library/Python/2.7/lib/python/site-packages (from pyocd)
Requirement already up-to-date: six in /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (from pyocd)
[... 23 lines cut ...]
You are using pip version 9.0.3, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
C02WW1FEHV2T-3:wip samgro01$ pyocd --version
0.14.3
C02WW1FEHV2T-3:wip samgro01$ pyocd gdb
0000155:WARNING:usb:STLink probes are not supported because no libusb library was found.
0000155:INFO:board:Target type is lpc1768
0000192:INFO:dap:DP IDR = 0x2ba01477
0000204:INFO:dap:AP#0 IDR = 0x24770011
0000214:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=000 part=000)
0000222:INFO:rom_table:[0]<e000e000:SCS-M3 class=14 designer=43b part=000>
0000227:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0000232:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0000238:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0000242:INFO:rom_table:[4]<e0040000:TPIU-M3 class=9 designer=43b part=923 devtype=11 archid=0000 devid=0:0:ca1>
0000247:INFO:rom_table:[5]<e0041000:ETM-M3 class=9 designer=43b part=924 devtype=13 archid=0000 devid=0:0:0>
0000252:INFO:cortex_m:CPU core is Cortex-M3 r2p0
0000257:INFO:dwt:4 hardware watchpoints
0000260:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000270:INFO:semihost:Telnet server started on port 4444
0000270:INFO:gdbserver:GDB server started on port 3333
^C:wip samgro01$ 
...

From the error unable to open device and reading other issues, I was looking at USB to be the source of problem but maybe it was the socket hiding behind an error message that isn't very descriptive?

TeoMahnic, 2026-01-07:
Closing this issue. Python 2.7 is no longer supported by current pyOCD releases.