pyocd/pyOCD #369: Cortex-M33 / Musca-A1 Support

HugoARA, 2018-06-29:
Hi, does pyOCD support Musca-A1 board? Is there documentation about porting it? Thanks

flit, 2018-07-01:
The Musca-A1 will be supported as part of v8-M support in pyOCD. You can see the work on the [dev_v8m branch of my fork](https://github.com/flit/pyOCD/tree/dev_v8m). At the moment, I can't guarantee any schedule for completing this work, except that it is relatively high priority.

flit, 2019-04-16:
CM33 support is merged into master with #603.

Musca-A1 support is not fully working yet. Flash programming has issues (partially due to working around chip errata), but you can debug the device.

danielRep, 2019-05-04:
Hi @flit 

I am using VSCode with Cortex-Debug plugin. Could you help on how to setup the launch configurations to  debug Musca-A1?

flit, 2019-05-05:
Hi @danielRep, Unfortunately I'm not familiar enough with the cortex-debug plugin to know what could be causing the problem. I think I remember another user saying that cortex-debug doesn't yet support the combined `pyocd` tool; it only supports the deprecated `pyocd-gdbserver` tool. However, `pyocd-gdbserver` should still be available, as it hasn't been removed yet.

Several questions:
- Is `pyocd-gdbserver` in your `PATH`?
- Is there a way to point cortex-debug at the pyOCD executable?
- Is there a way to pass additional arguments to pyOCD?
- Do you get any log output from pyOCD? If so, could you please copy it here?

You might also try running pyOCD directly from the command line just to verify the connection. First run `pyocd list` to print out connected debug probes. Musca-A1 should be in the list. Then run `pyocd-gdbserver, which should connect and start up the gdbserver (if you have more than one probe connected, it will ask which one you want to use).

danielRep, 2019-05-06:
Hi @flit. 

This issue as been solved. It was a problem related with the Cortex-Debug plugin. If anyone encounter this problem, the solution is on this [link](https://github.com/Marus/cortex-debug/issues/163).

Thanks !

flit, 2019-05-06:
Thanks for the link! Sorry we unintentionally broke cortex-debug when we changed the log message…

closed 2019-05-06 by flit