pyocd/pyOCD #375: Error flashing with gdbserver

MaureenHelm, 2018-07-18:
Using pyocd v0.11.0 installed via pip3, with frdm-k64f

I get the following error in the pyocd-gdbserver when I run the `load` command in a gdb client:
```
$ pyocd-gdbserver -t k64f -da='limit_packets=True' -d debug
0000314:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x80ee idProduct=0x0021. This is probably a permission issue.
0000315:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x1d6b idProduct=0x0001. This is probably a permission issue.
0000347:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x80ee idProduct=0x0021. This is probably a permission issue.
0000353:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x1d6b idProduct=0x0001. This is probably a permission issue.
0000394:DEBUG:dap_access_cmsis_dap:Limiting packet count to 1
0000409:DEBUG:board:init board <pyOCD.board.mbed_board.MbedBoard object at 0x7f938482b160>
0000429:INFO:cmsis_dap_core:DAP SWD MODE initialized
0000562:INFO:target_kinetis:K64F not in secure state
0000623:INFO:rom_table:ROM table #0 @ 0xe00ff000 cidr=b105100d pidr=4000bb4c4
0000644:INFO:rom_table:[0]<e000e000:SCS-M4 cidr=b105e00d, pidr=4000bb00c, class=14>
0000656:INFO:rom_table:[1]<e0001000:DWT cidr=b105e00d, pidr=4003bb002, class=14>
0000667:INFO:rom_table:[2]<e0002000:FPB cidr=b105e00d, pidr=4002bb003, class=14>
0000678:INFO:rom_table:[3]<e0000000:ITM cidr=b105e00d, pidr=4003bb001, class=14>
0000700:INFO:rom_table:[4]<e0040000:TPIU-M4 cidr=b105900d, pidr=4000bb9a1, class=9, devtype=11, devid=ca1>
0000722:INFO:rom_table:[5]<e0041000:ETM-M4 cidr=b105900d, pidr=4000bb925, class=9, devtype=13, devid=0>
0000743:INFO:rom_table:[6]<e0042000:ETB cidr=b105900d, pidr=4003bb907, class=9, devtype=21, devid=0>
0000769:INFO:rom_table:[7]<e0043000:CSTF cidr=b105900d, pidr=4001bb908, class=9, devtype=12, devid=28>
0000792:INFO:cortex_m:CPU core is Cortex-M4
0000817:INFO:cortex_m:FPU present
0000826:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000827:DEBUG:fpb:fpb has been disabled
0000860:INFO:dwt:4 hardware watchpoints
0000884:INFO:semihost:Telnet: server started on port 4444
0000885:INFO:gdbserver:GDB server started at port:3333
0051543:INFO:gdbserver:One client connected!
0051543:DEBUG:gdbserver:GDB received query: [b'Supported', b'multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df']
0051587:DEBUG:gdbserver:GDB general set: b'StartNoAckMode'
0051631:DEBUG:gdbserver:GDB received query: [b'Xfer', b'features', b'read', b'target.xml', b'0,7fb#4a']
0051631:DEBUG:gdbserver:GDB query b'read_feature': offset: 0, size: 2043
0051636:DEBUG:gdbserver:GDB received query: [b'Xfer', b'features', b'read', b'target.xml', b'7fb,7fb#19']
0051637:DEBUG:gdbserver:GDB query b'read_feature': offset: 2043, size: 2043
0051642:DEBUG:gdbserver:GDB general set: b'NonStop:0'
0051642:DEBUG:gdbserver:GDB received query: [b'TStatus#49']
0051642:DEBUG:gdbserver:Current thread 0 is no longer valid, switching context to target
0051674:DEBUG:cache:no accesses
0051675:DEBUG:cache:out of date run token; invalidating cache
0051698:DEBUG:gdbserver:Tresponse=b'T0507:00000000;0d:a80e0020;0e:ffffffff;0f:14190000;thread:1;core:0;'
0051698:DEBUG:gdbserver:GDB received query: [b'Xfer', b'threads', b'read', b'', b'0,7fb#d0']
0051698:DEBUG:gdbserver:GDB query b'threads': offset: 0, size: 2043
0051710:DEBUG:gdbserver:GDB received query: [b'Attached#8f']
0051711:DEBUG:gdbserver:GDB received query: [b'Offsets#4b']
0051711:DEBUG:gdbserver:GDB received query: [b'Xfer', b'threads', b'read', b'', b'0,7fb#d0']
0051711:DEBUG:gdbserver:GDB query b'threads': offset: 0, size: 2043
0051723:DEBUG:gdbserver:GDB received query: [b'Xfer', b'memory-map', b'read', b'', b'0,7fb#e9']
0051723:DEBUG:gdbserver:GDB query b'memory_map': offset: 0, size: 2043
0051728:DEBUG:cache:no reads
0051728:DEBUG:cache:out of date run token; invalidating cache
0051740:DEBUG:context_facade:GDB getting register context
0051976:DEBUG:context_facade:GDB reg: r0 = 0x0
0051976:DEBUG:context_facade:GDB reg: r1 = 0x0
0051976:DEBUG:context_facade:GDB reg: r2 = 0x0
0051976:DEBUG:context_facade:GDB reg: r3 = 0x0
0051976:DEBUG:context_facade:GDB reg: r4 = 0x0
0051976:DEBUG:context_facade:GDB reg: r5 = 0x0
0051976:DEBUG:context_facade:GDB reg: r6 = 0x0
0051976:DEBUG:context_facade:GDB reg: r7 = 0x0
0051976:DEBUG:context_facade:GDB reg: r8 = 0x0
0051977:DEBUG:context_facade:GDB reg: r9 = 0x0
0051977:DEBUG:context_facade:GDB reg: r10 = 0x0
0051977:DEBUG:context_facade:GDB reg: r11 = 0x0
0051977:DEBUG:context_facade:GDB reg: r12 = 0x0
0051977:DEBUG:context_facade:GDB reg: sp = 0x20000EA8
0051977:DEBUG:context_facade:GDB reg: lr = 0xFFFFFFFF
0051977:DEBUG:context_facade:GDB reg: pc = 0x1914
0051977:DEBUG:context_facade:GDB reg: xpsr = 0x1000000
[... 73 lines cut ...]
0125227:ERROR:gdbserver:Unknown RSP packet: b'$vFlashWrite:30fc:\x05\x00\x00\x00\x9c\x17\x00 \x04\x00\x00\x00\x00\x00\x00\x00\xff\xff\xff\xff\x80}\n\x00\x00\x18\x86a\x00\x00\x00\x00\x00\xff\xff\xff\x1f\x8b}\n\x00\x00\x10\x04A\x00\x00\x00\x03 \xff\xff\xff\xff\x98}\n\x00\x00\x10\x04A\x00\x00\x00\x00\x00\xff\xff\xff\x07\xa5}\n\x00\x00]\xd7u\x00\x00\x00\x00 \xff\xff\x02 \xad}\n\x00\x00\x10\x04Y\x00\t\x08\x00\x00\x00\t=\x00\x00\x80\x00\x00\xb1\x1a\x00\x00\x01\x00\x00\x00\x80O\x12\x00#41'
0125227:DEBUG:gdbserver:flash op: b'FlashWrite'
0125227:DEBUG:gdbserver:flash write addr: 0x3170
0125227:ERROR:gdbserver:Unknown RSP packet: b'$vFlashWrite:3170:|)\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xa0)\x00\x00\x10+\x00\x00\x00\x00\x00\x00\xac)\x00\x00\x10+\x00\x00\x00\x00\x00\x00\xb8)\x00\x00\x10+\x00\x00\x00\x00\x00\x00\xc4)\x00\x00\x10+\x00\x00\x00\x00\x00\x00\xdc)\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe8)\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x94)\x00\x00\x00+\x00\x00\x00\x00\x00\x000}\n\x00\x00\xa00\x00\x00\xb8\x00\x00 \x88)\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd0)\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf4)\x00\x00\x140\x00\x00\xac\x00\x00 \x00}\n\x00\x00\x140\x00\x00\xa0\x00\x00 \x0c}\n\x00\x00\x140\x00\x00\x94\x00\x00 \x18}\n\x00\x00\x140\x00\x00\x88\x00\x00 }\x04}\n\x00\x00\x140\x00\x00|\x00\x00 #68'
0125227:DEBUG:gdbserver:flash op: b'FlashWrite'
0125227:DEBUG:gdbserver:flash write addr: 0x3230
0125228:ERROR:gdbserver:Unknown RSP packet: b'$vFlashWrite:3230:\xa0\x01\x00 P\x06\x00 \x00\x04\x00\x00\x11\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00#05'
0125228:DEBUG:gdbserver:flash op: b'FlashWrite'
0125228:DEBUG:gdbserver:flash write addr: 0x325c
0125228:ERROR:gdbserver:Unknown RSP packet: b'$vFlashWrite:325c:\xf0\x18\x00 \xf0\x18\x00 \x01\x00\x00\x00\x01\x00\x00\x00\x00\x19\x00 \x00\x19\x00 \x00\x00\x00\x00\x01\x00\x00\x00#a5'
0125228:DEBUG:gdbserver:flash op: b'FlashDone#ea'
0125228:ERROR:gdbserver:Unhandled exception in handleMsg: 'NoneType' object has no attribute 'program'
Traceback (most recent call last):
  File "/home/maureen/.local/lib/python3.5/site-packages/pyOCD/gdbserver/gdbserver.py", line 528, in handleMsg
    reply = handler(msg[msgStart:])
  File "/home/maureen/.local/lib/python3.5/site-packages/pyOCD/gdbserver/gdbserver.py", line 744, in vCommand
    return self.flashOp(data)
  File "/home/maureen/.local/lib/python3.5/site-packages/pyOCD/gdbserver/gdbserver.py", line 857, in flashOp
    self.flashBuilder.program(chip_erase=self.chip_erase, progress_cb=progress_cb, fast_verify=self.fast_program)
AttributeError: 'NoneType' object has no attribute 'program'
```

Flashing with pyocd-flashtool works fine.

cc: @carlescufi

flit, 2018-07-19:
Hi @MaureenHelm, thanks for reporting the issue. I'll let you know if I have any trouble reproducing it.

flit, 2018-07-25:
@MaureenHelm pyOCD v0.11.1 was just release that includes a fix for this bug and a few other issues. Thanks again for reporting!

Closing this issue.

closed 2018-07-25 by flit

MaureenHelm, 2018-07-26:
Thanks @flit. I verified the fix works in v0.11.1. pyocd was the only python 2 package used by zephyr, so we can now use python 3 exclusively. See zephyrproject-rtos/zephyr#9138

flit, 2018-07-26:
@MaureenHelm Glad to hear it! 😄