pyocd/pyOCD #388: Propagate trusted-host option

jeromecoutant, 2018-08-07:
Hi

In my company network, due to security reason, we need to use some --trusted-host options during pip install in order to bypass SSL certificate check...

During pyOCD setup, some other packages are installed,
but failed because trusted-host option is not used.
 
   Download error on https://pypi.org/simple/setuptools_scm/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661) -- Some packages may not be found!

Is it possible to use the trusted-host option during all the pyOCD setup ?

Thx

flit, 2018-08-07:
Hi @jeromecoutant, does running `pip install --trusted-host=pypi.org pyocd` not work for you? All the packages are hosted on pypi.org.

If you're trying to set up pyOCD for development, you should be able to run `pip install --trusted-host=pypi.org -r dev-requirements.txt`. The `dev-requirements.txt` file will install pyOCD as editable (equivalent to running `pip install --editable .`), plus a few other dependencies for testing.

If for some reason pip is not using the trusted host setting when installing dependencies, a possible alternative would be to install all of the dependencies in a number of separate calls to pip, passing `--trusted-host=pypi.org` for each one.

(It's kind of ironic to bypass TLS certificate checks due to security concerns!)

jeromecoutant, 2018-08-17:
You're right, I have installed each requirement with a separate call to pip.
Thx

closed 2018-08-17 by jeromecoutant