pyocd/pyOCD #389: AssertionError when flashing K66F

SeppoTakalo, 2018-08-07:
```
$ pyocd-gdbserver -o
0000397:INFO:cmsis_dap_core:DAP SWD MODE initialized
0000425:INFO:target_kinetis:K66F18 not in secure state
0000440:INFO:rom_table:ROM table #0 @ 0xe00ff000 cidr=b105100d pidr=4000bb4c4
0000445:INFO:rom_table:[0]<e000e000:SCS-M4 cidr=b105e00d, pidr=4000bb00c, class=14>
0000448:INFO:rom_table:[1]<e0001000:DWT cidr=b105e00d, pidr=4003bb002, class=14>
0000452:INFO:rom_table:[2]<e0002000:FPB cidr=b105e00d, pidr=4002bb003, class=14>
0000455:INFO:rom_table:[3]<e0000000:ITM cidr=b105e00d, pidr=4003bb001, class=14>
0000460:INFO:rom_table:[4]<e0040000:TPIU-M4 cidr=b105900d, pidr=4000bb9a1, class=9, devtype=11, devid=ca1>
0000465:INFO:rom_table:[5]<e0041000:ETM-M4 cidr=b105900d, pidr=4000bb925, class=9, devtype=13, devid=0>
0000470:INFO:rom_table:[6]<e0042000:ETB cidr=b105900d, pidr=4003bb907, class=9, devtype=21, devid=0>
0000475:INFO:rom_table:[7]<e0043000:CSTF cidr=b105900d, pidr=4001bb908, class=9, devtype=12, devid=28>
0000479:INFO:cortex_m:CPU core is Cortex-M4
0000484:INFO:cortex_m:FPU present
0000487:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000494:INFO:dwt:4 hardware watchpoints
0000499:INFO:semihost:Telnet: server started on port 4444
0000499:INFO:gdbserver:GDB server started at port:3333
0006355:INFO:gdbserver:One client connected!
0006421:INFO:gdbserver:Attempting to load Argon
0006423:INFO:gdbserver:Attempting to load Zephyr
0006424:INFO:gdbserver:Attempting to load FreeRTOS
0006439:INFO:freertos:FreeRTOS: invalidating threads list: <Notification@0x104e40610 event=8 source=<pyOCD.coresight.cortex_m.CortexM object at 0x104e5d590> data=None>
0006888:INFO:freertos:FreeRTOS: invalidating threads list: <Notification@0x1049f05d0 event=8 source=<pyOCD.coresight.cortex_m.CortexM object at 0x104e5d590> data=None>
[=================== ]  98%0018874:ERROR:gdbserver:Unhandled exception in handleMsg: 
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 528, in handleMsg
    reply = handler(msg[msgStart:])
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 744, in vCommand
    return self.flashOp(data)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 857, in flashOp
    self.flashBuilder.program(chip_erase=self.chip_erase, progress_cb=progress_cb, fast_verify=self.fast_program)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/flash/flash_builder.py", line 262, in program
    flash_operation = self._page_erase_program_double_buffer(progress_cb)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/flash/flash_builder.py", line 620, in _page_erase_program_double_buffer
    progress_cb(float(progress) / float(self.page_erase_weight))
  File "/usr/local/lib/python2.7/site-packages/pyOCD/utility/progress.py", line 37, in __call__
    assert progress <= 1.0
AssertionError

```

I'm using `arm-none-eabi-gdb` to flash Mbed Cloud Client example to K66F board.

SeppoTakalo, 2018-08-07:
Seems that the problem appeared in `0.11.*` versions.

I switched back with `pip install -U pyocd==0.10.0` and problem is not visible anymore.

flit, 2018-08-15:
Unfortunately, I'm not able to reproduce this on the K64F using the ELF you sent me.

What was the state of the target's flash prior to the load attempt that asserted?

Btw, the reason this appeared in v0.11.0 is that progress reporting was commonized into the `pyOCD.utility.progress` module in this version. As part of the refactoring, the assert was included from one of the previously multiple progress implementations.

flit, 2018-10-13:
Fyi, the [0.13.0b2 release](
https://github.com/mbedmicro/pyOCD/releases/tag/v0.13.0b2) disables the problematic assert. It won't be re-enabled until the underlying root cause is fixed. There don't seem to be any issues with programming flash even when the assert fires, so I'm closing this issue.

closed 2018-10-13 by flit