pyocd/pyOCD #399: Nordic nRF52_pca10052 board not appearing with "pyocd-gdbserver -l" command

nwsetec, 2018-08-24:
I'm having an issue with pyocd to debug a Zephyr app I'm developing with the nrf52832_pca10040 development board.  I'm running windows 7 and have tried using Python2 and Python3.  If I run the command "pyocd-gdbserver -l" I get "No available boards are connected" although when using the Segger Jlink software I can see the segger device.  Is this an issue or am I missing something?

I'm using pyocd 0.11.2

flit, 2018-08-26:
Hi @nwsetec, pyOCD only supports the CMSIS-DAP protocol at the moment. So you'll need to install the CMSIS-DAP IF firmware ([downloadable here](
https://www.nordicsemi.com/eng/nordic/download_resource/53481/6/50120613/64337) from Nordic's site) before pyOCD can see the board.

nixward, 2018-08-29:
Thanks flit, working now!

closed 2018-08-29 by nwsetec