pyocd/pyOCD #406: Unhandled exception in handlemsg

maclobdell, 2018-09-18:
While debugging LPC546XX, using pyocd 0.12.0, I get the following error.

```
0000246:WARNING:svd:Failed to load SVD file LPC54608.svd
0000247:INFO:cmsis_dap_core:DAP SWD MODE initialized
0000263:INFO:dap:DP IDR = 0x2ba01477
0000276:INFO:dap:AP#0 IDR = 0x24770011
0000287:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=43b part=4c4)
0000296:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000302:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0000308:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0000315:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0000320:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0000325:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0000332:INFO:cortex_m:CPU core is Cortex-M4
0000338:INFO:cortex_m:FPU present
0000345:INFO:dwt:4 hardware watchpoints
0000348:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000358:INFO:semihost:Telnet: server started on port 4444
0000358:INFO:gdbserver:GDB server started at port:3333
Started by GNU MCU Eclipse
0000803:INFO:gdbserver:One client connected!
0000957:INFO:gdbserver:Semihosting enabled
0001157:INFO:gdbserver:Attempting to load Argon
0001158:INFO:gdbserver:Attempting to load RTX5
0001158:INFO:gdbserver:RTX5 loaded successfully
[---|---|---|---|---|---|---|---|---|----]
[========================================]
0027144:INFO:flash_builder:Programmed 444348 bytes (14 pages) at 17.02 kB/s
0035197:ERROR:gdbserver:Unhandled exception in handleMsg: 
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyOCD\gdbserver\gdbserver.py", line 530, in handleMsg
    reply = handler()
  File "c:\python27\lib\site-packages\pyOCD\gdbserver\gdbserver.py", line 964, in getRegisters
    return self.createRSPPacket(self.target_facade.getRegisterContext())
  File "c:\python27\lib\site-packages\pyOCD\gdbserver\context_facade.py", line 55, in getRegisterContext
    vals = self._context.readCoreRegistersRaw(reg_num_list)
  File "c:\python27\lib\site-packages\pyOCD\rtos\rtx5.py", line 138, in readCoreRegistersRaw
    inException = self._get_ipsr() > 0
  File "c:\python27\lib\site-packages\pyOCD\rtos\rtx5.py", line 195, in _get_ipsr
    return self._parent.readCoreRegister('xpsr') & 0xff
  File "c:\python27\lib\site-packages\pyOCD\debug\context.py", line 87, in readCoreRegister
    regValue = self.readCoreRegisterRaw(regIndex)
  File "c:\python27\lib\site-packages\pyOCD\debug\context.py", line 99, in readCoreRegisterRaw
    vals = self.readCoreRegistersRaw([reg])
  File "c:\python27\lib\site-packages\pyOCD\debug\cache.py", line 484, in readCoreRegistersRaw
    return self._regcache.readCoreRegistersRaw(reg_list)
  File "c:\python27\lib\site-packages\pyOCD\debug\cache.py", line 111, in readCoreRegistersRaw
    self._check_cache()
  File "c:\python27\lib\site-packages\pyOCD\debug\cache.py", line 88, in _check_cache
    if self._context.core.isRunning():
  File "c:\python27\lib\site-packages\pyOCD\coresight\cortex_m.py", line 630, in isRunning
    return self.getState() == Target.TARGET_RUNNING
  File "c:\python27\lib\site-packages\pyOCD\coresight\cortex_m.py", line 607, in getState
    dhcsr = self.readMemory(CortexM.DHCSR)
  File "c:\python27\lib\site-packages\pyOCD\coresight\cortex_m.py", line 441, in readMemory
    result = self.ap.readMemory(addr, transfer_size, now)
  File "c:\python27\lib\site-packages\pyOCD\coresight\ap.py", line 258, in readMemory
    result = readMemCb()
  File "c:\python27\lib\site-packages\pyOCD\coresight\ap.py", line 240, in readMemCb
    res = result_cb()
  File "c:\python27\lib\site-packages\pyOCD\coresight\dap.py", line 316, in readAPCb
    result = result_cb()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 663, in read_reg_cb
    res = transfer.get_result()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 119, in get_result
    self.daplink.flush()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 594, in flush
    self._read_packet()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 749, in _read_packet
[... 158 lines cut ...]
    return self.getState() == Target.TARGET_RUNNING
  File "c:\python27\lib\site-packages\pyOCD\coresight\cortex_m.py", line 607, in getState
    dhcsr = self.readMemory(CortexM.DHCSR)
  File "c:\python27\lib\site-packages\pyOCD\coresight\cortex_m.py", line 441, in readMemory
    result = self.ap.readMemory(addr, transfer_size, now)
  File "c:\python27\lib\site-packages\pyOCD\coresight\ap.py", line 258, in readMemory
    result = readMemCb()
  File "c:\python27\lib\site-packages\pyOCD\coresight\ap.py", line 240, in readMemCb
    res = result_cb()
  File "c:\python27\lib\site-packages\pyOCD\coresight\dap.py", line 316, in readAPCb
    result = result_cb()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 663, in read_reg_cb
    res = transfer.get_result()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 119, in get_result
    self.daplink.flush()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 594, in flush
    self._read_packet()
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 749, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 419, in decode_data
    data = self._decode_transfer_data(data)
  File "c:\python27\lib\site-packages\pyOCD\pyDAPAccess\dap_access_cmsis_dap.py", line 317, in _decode_transfer_data
    raise DAPAccessIntf.TransferTimeoutError()
TransferTimeoutError
```

flit, 2018-09-24:
@maclobdell How reproducible is this exception? (I've actually never seen a TransferTimeoutError before!)

fepz, 2018-10-15:
I am having the same error, running or debugging a simple project on a mbed LPC 1768 board, using GNU MCU Eclipse. Using pyOCD manually to upload and debug works ok.

fepz, 2018-10-16:
No exception when using pyOCD 0.10. It seems to be a problem when using pyOCD version 0.11 or later along with GNU MCU Eclipse.

flit, 2018-10-17:
Thanks for the info, @fepz.

flit, 2019-10-12:
@thegecko @maclobdell My guess is that this is another sleep related issue. The Mbed OS app is causing the device to enter sleep and therefore lose debug connectivity. So when pyOCD tries to read the core state, the transaction hangs and times out.

Please try disabling the call to `hal_sleep()` in `sleep_manager_sleep_auto()` in `platform/mbed_sleep_manager.c`. (You should already be building using the Debug profile.)

thegecko, 2019-10-17:
@flit, disabling the `hal_sleep()` seemed to have little effect:

```
Selected port 50000 for debugging
0000845:INFO:board:Target type is lpc1768
0000975:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0001042:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001145:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=000 part=000)
0001181:INFO:rom_table:[0]
0001217:INFO:rom_table:[1]
0001265:INFO:rom_table:[2]
0001341:INFO:rom_table:[3]
0001402:INFO:rom_table:[4]
0001470:INFO:rom_table:[5]
0001574:INFO:cortex_m:CPU core #0 is Cortex-M3 r2p0
0001626:INFO:dwt:4 hardware watchpoints
0001633:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001660:INFO:server:Semihost server started on port 4444
0001660:INFO:gdbserver:GDB server started on port 50000
Reading symbols from /Users/thegecko/Mbed Programs/mbed-os-example-blinky/BUILD/LPC1768/ARMC6/mbed-os-example-blinky.elf...
done.
0001705:INFO:gdbserver:One client connected!
0x00006b40 in sleep_manager_unlock_deep_sleep_internal () at ./mbed-os/platform/mbed_sleep_manager.c:183
183	    if (core_util_atomic_decr_u16(&deep_sleep_lock, 1) == 0xFFFF) {
0001777:INFO:gdbserver:Attempting to load Argon
0001779:INFO:gdbserver:Attempting to load RTX5
0001780:INFO:gdbserver:RTX5 loaded successfully
Attached to debugger on port 50000
0007545:INFO:loader:Erased chip, programmed 37888 bytes (37 pages), skipped 3072 bytes (3 pages) at 7.06 kB/s
Image loaded: /Users/thegecko/Mbed Programs/mbed-os-example-blinky/BUILD/LPC1768/ARMC6/mbed-os-example-blinky.elf
Note: automatically using hardware breakpoints for read-only addresses.
0007758:ERROR:gdbserver:Unhandled exception in handle_message: 
Traceback (most recent call last):
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 547, in handle_message
    reply = handler()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 982, in get_registers
    return self.create_rsp_packet(self.target_facade.get_register_context())
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/context_facade.py", line 69, in get_register_context
    vals = self._context.read_core_registers_raw(reg_num_list)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/debug/cache.py", line 531, in read_core_registers_raw
    return self._regcache.read_core_registers_raw(reg_list)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/debug/cache.py", line 133, in read_core_registers_raw
    self._check_cache()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/debug/cache.py", line 110, in _check_cache
    if self._core.is_running():
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 943, in is_running
    return self.get_state() == Target.TARGET_RUNNING
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 920, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 605, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 564, in _read_memory
    result = read_mem_cb()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 546, in read_mem_cb
    res = result_cb()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 276, in read_ap_cb
    result = result_cb()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 307, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferError
```

flit, 2019-10-17:
You are running with a different target than @maclobdell. He was using an LPC546xx rather than the LPC1768 you're using. So I think these are separate bugs. Note that different exceptions are raised: `TransferError` for LPC1768, but `TransferTimeoutError` for LPC56xx.

flit, 2019-10-21:
Created #745 to cover the LPC1768 issue.

flit, 2019-10-21:
@maclobdell What application were you testing with that caused this issue? Can you reproduce it with mbed-os-example-blinky?