pyocd/pyOCD #412: Please migrate to Python3

codehelp, 2018-09-25:
With Python 3 being the future of Python while Python 2 is still in active use, it is good to have your project available for both major releases of Python.

If pyocd is to be packaged, it will only be packaged as Python3.

codehelp, 2018-09-25:
The current release depends on setuptools-scm-git-archive which is only available as a package for Python3.
https://tracker.debian.org/pkg/setuptools-scm-git-archive

codehelp, 2018-09-25:
setup.py for 0.12 declares support for python3 but the README states only Python2.7 - what is the current state of Python3 support in pyocd?

flit, 2018-09-26:
As of v0.11, PyOCD fully supports Python 3.6+ (though it should actually work ok on 3.4+), as well as still maintaining compatibility with Python 2.7. I mostly develop and test with 3.7 these days, and Travis CI runs both 2.7 and 3.6. However, our internal CI system still uses 2.7 (planning to address this soon). 

Thanks for the pointer on setuptools-scm-git-archive only being available for Python 3, I had missed that! And I'll update the readme.

codehelp, 2018-09-27:
I can't build a Debian package for pyocd for Python2 and new packages aren't being added for Python2 support anymore, so there won't be a Python2 package for setuptools-scm-git-archive. Is there a reason to stick with Python2?

flit, 2018-09-27:
Ok. I'll see about making setuptools-scm-git-archive optional.

Yes, supporting Python 2 is important because a lot of users still have only that version installed. There are a wide range of users of pyOCD, and it's best if it just works with as many systems as reasonable.

If/when we move to binary executables as the primary release artifact for most users, it may be less important to support Python 2. But that will be a while from now.

codehelp, 2018-10-05:
I can package pyocd to use Python3 for the installed scripts but provide Python2.7 module support. I'll reference this issue to cover why the Python2 code is still needed.

Thanks.

codehelp, 2018-10-05:
Packaging anything for Python2 still needs the support for setuptools-scm-git-archive, at least currently. - so I'll skip including the Python2 module - leave that support for upstream.

If the README can be updated to show Python3 support, this issue can be closed.

flit, 2018-10-11:
Closing this issue based on the changes in #427. If you don't think the changes are sufficient, please re-open this issue.

closed 2018-10-11 by flit