pyocd/pyOCD #442: Error installing pyocd 0.13.0 from pip

MaureenHelm, 2018-11-09:
```
Collecting pyocd (from -r zephyr/scripts/requirements.txt (line 13))
  Downloading https://files.pythonhosted.org/packages/12/66/686bb88d1137cce5c7f374f97bdde0361804221fbeca4758fdc8c2a4c2ad/pyocd-0.13.0.tar.gz (942kB)
    100% |████████████████████████████████| 952kB 5.4MB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-qifmoy4p/pyocd/setup.py", line 76, in <module>
        include_package_data=True,  # include files from MANIFEST.in
      File "/home/maureen/zephyr-test/lib/python3.5/site-packages/setuptools/__init__.py", line 140, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.5/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/home/maureen/zephyr-test/lib/python3.5/site-packages/setuptools/dist.py", line 370, in __init__
        k: v for k, v in attrs.items()
      File "/usr/lib/python3.5/distutils/dist.py", line 281, in __init__
        self.finalize_options()
      File "/home/maureen/zephyr-test/lib/python3.5/site-packages/setuptools/dist.py", line 529, in finalize_options
        ep.load()(self, ep.name, value)
      File "/tmp/pip-install-qifmoy4p/pyocd/.eggs/setuptools_scm-3.1.0-py3.5.egg/setuptools_scm/integration.py", line 23, in version_keyword
        dist.metadata.version = get_version(**value)
      File "/tmp/pip-install-qifmoy4p/pyocd/.eggs/setuptools_scm-3.1.0-py3.5.egg/setuptools_scm/__init__.py", line 152, in get_version
        template=write_to_template,
      File "/tmp/pip-install-qifmoy4p/pyocd/.eggs/setuptools_scm-3.1.0-py3.5.egg/setuptools_scm/__init__.py", line 72, in dump_version
        with open(target, "w") as fp:
    FileNotFoundError: [Errno 2] No such file or directory: 'pyocd/_version.py'
```

cc: @nashif

jupe, 2018-11-10:
Please fix this asap - breaks also Icetea and mbed-flasher installation.

CC: @bridadan , @studavekar

aunsbjerg, 2018-11-10:
I am seeing this issue with both python 3.5 and python 3.6 in linux. Interestingly, I am _not_ seeing this issue in windows at all.

flit, 2018-11-12:
We're having multiple issues here that seem to be caused by the (necessary) package name change to all-lowercase `pyocd`. See #444 too.

Can you try uninstalling and reinstalling pyOCD with pip? I'll also test some more on Linux.

MaureenHelm, 2018-11-13:
> Can you try uninstalling and reinstalling pyOCD with pip?

I saw this error installing pyocd in a new virtualenv. @nashif, when you saw this in zephyr CI, was it a fresh install or upgrade from a previous version?

nashif, 2018-11-13:
It was fresh install while creating  a container image.

flit, 2018-11-14:
I'm able to reproduce this on both Python 2.7.15 and 3.6.6 under Linux (Ubuntu 18.0.1 LTS). Note that it does not happen on either macOS or Windows, which is why I didn't see it. (I'm taking a note to add automated installation testing for released versions.)

Now looking into the cause… It seems like it's somehow related to the setuptools_scm package that creates the `_version.py` file. I'm not sure why this would have changed compared to previous versions.

flit, 2018-11-14:
I think I found the problem: I use a separate, clean working copy of the pyOCD repo to perform releases to PyPI. Because I use macOS with a case-insensitive APFS filesystem, when I pulled the 0.13.0 changes into this working copy, git did not rename the `pyOCD` directory to `pyocd`. Nor did setuptools have any issues building the sdist. So the tarball released to PyPI has the incorrect mixed case name for the pyOCD package. This causes different failures depending on whether the end user's filesystem is case-sensitive. Will fix and release 0.13.1 very shortly.

flit, 2018-11-15:
Just released [v0.13.1](
https://github.com/mbedmicro/pyOCD/releases/tag/v0.13.1) that fixes the issues on macOS, Linux, and Windows.

Closing this issue. If you continue to have problems please reopen.

closed 2018-11-15 by flit