pyocd/pyOCD #452: Support flash write page sizes smaller than the sector size

flit, 2018-11-15:
Some targets, notably some STM32 devices, have crazy large erase sector sizes (up to 128 kB!). This can prevent double buffering from being enabled since there is not enough RAM to fit two of these sectors.

The solution is to support a flash write page size that is smaller than the sector size. Most MCU flash controllers allow programming granularity of somewhere between 1 and 16 bytes (depends on flash technology and controller IP). The main constraint is that you want to have as large programming buffers as possible to reduce buffer swap and command issuing latency.

This change should come after #357 since that task will automatically allocate memory for the flash algos and page buffers. With that in place, it should be straightforward to reduce buffer sizes and update `FlashBuilder` to support write pages != erase pages.

stppgit, 2019-03-28:
This also impacts external memory programming using QSPI interface in Cypress PSoC 6 devices.
S25FL512S memory, for example, has erase page == 256K, what requires 256K of PSoC 6 SRAM to be allocated as the data buffer. Algorithms data + stack also requires up to 32Kbytes of SRAM.

This tightly works for PSoC6A-BLE2 devices with 288 Kbytes of SRAM but does not work for PSoC6 devices with SRAM size < 288KBytes and for secure devices where SRAM access is limited to specific regions.
In addition, we cannot use double-buffering to speed-up programming, what is critical for large QSPI memory devices such as S25FL512S (64M).

flit, 2019-03-28:
Exactly right. Please see my [feature/flash_erase_program_size](https://github.com/flit/pyOCD/tree/feature/flash_erase_program_size) branch. It's mostly done, except that the `flash_test.py` functional test fails. This is most likely due to assumptions the test makes that are no longer valid, rather than real bugs, but of course it must be fixed before the branch can be merged.

stppgit, 2019-03-28:
Sounds promising. Waiting for mainline intervention.
Thanks a lot Chris!

PR #602 (2019-04-02): Flash support for erase sectors
This PR adds support for separate erase sector and program page sizes. If the sizes are different, a `page_size` attribute can be set on the `FlashRegion`. The `blocksize` attribute remains, and is the erase sector size. Builtin targets that used a custom `Flash` class to implement different erase/program sizes were updated to use the new support and remove the custom class.

An important included change is that failed flash programming calls will raise either `FlashEraseFailure` or `FlashProgramFailure` exceptions now instead of just logging an error.

In addit [... cut]
closed 2019-04-02 by flit