pyocd/pyOCD #463: RT1050: flash related issues

flit, 2018-12-03:
Debugging a RAM-resident app on the NXP i.MX RT1050 MCU works successfully. However, programming flash and debugging an application residing in flash fails with various errors.

flit, 2019-01-17:
@AntoineZen Thanks for the additional information!

Hoohaha, 2019-01-21:
When will you plan to fix this issue? We really hope we can use pyocd to debug NXP RT boards.

flit, 2019-01-23:
I'm working on it with NXP, but I'm afraid I don't have a date for you.

flit, 2019-01-24:
Thanks. Unfortunately there's nothing revealing in the exception log.

baudiffred, 2019-10-16:
I can confirm the same failure as the posted log on an i.MXRT1021 EVK board.  I am seeing a failure of CMSIS transfer acknowledgement as the immediate cause of the exception.

danielRep, 2019-11-11:
I can confirm the same problem on an i.MXRT1020 EVK board. I am able to connect to the board and open a gdb session. When I try to "load" a FLASH resident program (led_blinky example) I have an output on pyocd terminal attesting the loading task (0019507:INFO:loader:Erased 16384 bytes (4 sectors), programmed 16384 bytes (4 pages), skipped 0 bytes (0 pages) at 8.26 kB/s). Nevertheless when I try to reset the board, the LED doesn't blink and I loose the connection to the board. To regain access I need to do a mass erase in Fuses boot mode and flash again through MCUXpressoIDE  and in internal boot mode. 

Any news from NXP @flit ?
Thank you!

flit, 2019-11-11:
Nothing really worth reporting yet, sorry. Recently I did get the RT1052 flash algo source code from NXP (unfortunately I'm not allowed to distribute it to others), but haven't had time to debug it. I did also get a RT1021 EVK recently, and have asked for the source for the RT1021 flash algo.

danielRep, 2019-11-11:
Thank you for the quick feedback @flit ! I will keep track this issue to check any advances on this. If anyone has a Linux-based tool to flash the i.MXRT1020 EVK w/o depending on the MCUXpresso tools, please let me know.

MateuszKlatecki, 2019-11-11:
@flit  flash algo source codes are distributed with MCUXpresso in ide\Examples\Flashdrivers\NXP\iMXRT folder.

flit, 2019-11-11:
@MateuszKlatecki Holy cow! Thanks for pointing that out. 😀 

It's not quite the same as what I was given. This project builds a .cfx file (used by the MCUX IDE), while the one I was given builds an .FLM. The latter format is used by CMSIS-Packs and pyOCD. However, from a quick review it looks like they mostly have wrapped the [.FLM API](https://arm-software.github.io/CMSIS_5/Pack/html/flashAlgorithm.html) with the LPCXFlashDriverLib API.

danielRep, 2019-11-11:
@flit , one last question.

Do you think that if I my binary is NOT FLASH resident, I could easily load the program to the board by using pyocd and a gdb a session?

flit, 2019-11-11:
You mean RAM resident? Yes, that should work.

Hoohaha, 2019-11-12:
Hi @flit , All NXP IMX-RT serires has same problem to load elf to flash. like EVK-MIMXRT1060, EVK-MIMXRT1015. It's good to see you get start on this issue.