pyocd/pyOCD #465: ImportError: No module named pyocd.tools.gdb_server

yennster, 2018-12-06:
I used `pip install -U pyocd` to upgrade my pyocd version today.

I try to run a GDB server with `pyocd-gdbserver` and I get the following error output:

```
~ $ pyocd-gdbserver
Traceback (most recent call last):
  File "/usr/local/bin/pyocd-gdbserver", line 7, in <module>
    from pyocd.tools.gdb_server import main
ImportError: No module named pyocd.tools.gdb_server
```

I am using pyocd version 0.13.2 and arm-none-eabi-gdb version `GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 8.1.0.20180315-git`

flit, 2018-12-06:
Hi @yennster, what OS are you using, and what pip and setuptools versions? You might need to uninstall pyOCD and reinstall.

The package name was changed from "pyOCD" to "pyocd" in order to be compliant with Python naming standards. Unfortunately, on case-insensitive filesystems, upgrading may not correctly rename the folder. What's weird is that I tried this on macOS, Windows, and Linux (Ubuntu) and it seemed ok (after some initial trouble in 0.13.0). Maybe it's and older pip or setuptools? I'll try reproducing it once I know your OS and versions.

JojoS62, 2018-12-07:
got the same issue, running on windows 10. With pyocd 0.13, mbetgt fiails. A 'mbedgt --version' throws an error that pyOCD.board is missing.
Downgrading to 0.12 fixes the problem, upgrading to 0.13.2 raises the problem again.
```
D:\Projects\Sn\LocalGit\test\mbed-SDIO-V2>mbedgt --version
mbedgt: Not all required Python modules were imported!
mbedgt: No module named pyOCD.board
```

yennster, 2018-12-07:
@flit Here's my version info:

- Mac OS 10.14.1
- pip 18.1
- python 2.7.15
- setuptools 39.2.0

flit, 2018-12-08:
@yennster I just tried reproducing your upgrade problem with no luck. I created a virtualenv with exactly the same versions as you have, installed pyOCD 0.12.0, then ran `pip install -U pyocd` to upgrade to the latest. `pyocd-gdbserver` still runs fine. So I'm not sure what the problem you are seeing could be caused by! Can you try using a virtualenv?

@JojoS62 In the meantime, I recommend using two virtualenvs. One for debugging with pyOCD, and one for running greentea. I've moved the issue you are having to #467 since it's different from @yennster's upgrade problem (but with similar errors).

yennster, 2019-02-13:
Issue is resolved now

closed 2019-02-13 by yennster