pyocd/pyOCD #467: Greentea version incompatibility

flit, 2018-12-08:
_Reported by @JojoS62 in #465._

running on windows 10. With pyocd 0.13, mbetgt fiails. A 'mbedgt --version' throws an error that pyOCD.board is missing.
Downgrading to 0.12 fixes the problem, upgrading to 0.13.2 raises the problem again.

```
D:\Projects\Sn\LocalGit\test\mbed-SDIO-V2>mbedgt --version
mbedgt: Not all required Python modules were imported!
mbedgt: No module named pyOCD.board
```

flit, 2018-12-08:
This looks like it might be caused by conflicting version requirements between htrun (used by greentea) and pyOCD. I'm working on fixing these tools so they can use the latest pyOCD.

flit, 2018-12-11:
Created pull requests to update pyOCD:
- ARMmbed/htrun#233
- ARMmbed/mbed-flasher#149

flit, 2018-12-16:
@JojoS62 ARMmbed/htrun#233 was merged and released in htrun version 1.4.4. Could you please test and close this issue if the problem is fixed? Thanks!

JojoS62, 2018-12-18:
@flit : I've just upgraded pyOCD to 0.14.2 but I stell get the same error. Is there is something else missing?

```
C:\Users\sn>pip install -U pyocd
Collecting pyocd
  Downloading https://files.pythonhosted.org/packages/74/2b/618c5ac18961770b26f429c78946cc9003860f1036a6d51902ba23f90811/pyocd-0.14.2.tar.gz (1.0MB)
    100% |################################| 1.0MB 3.9MB/s
Requirement already satisfied, skipping upgrade: intelhex in c:\python27\lib\site-packages (from pyocd) (2.2.1)
Requirement already satisfied, skipping upgrade: six in c:\python27\lib\site-packages (from pyocd) (1.11.0)
Requirement already satisfied, skipping upgrade: future in c:\python27\lib\site-packages (from pyocd) (0.16.0)
Requirement already satisfied, skipping upgrade: websocket-client in c:\python27\lib\site-packages (from pyocd) (0.51.0)
Requirement already satisfied, skipping upgrade: intervaltree<3.0 in c:\python27\lib\site-packages (from pyocd) (2.1.0)
Requirement already satisfied, skipping upgrade: colorama in c:\python27\lib\site-packages (from pyocd) (0.3.9)
Requirement already satisfied, skipping upgrade: pyelftools in c:\python27\lib\site-packages (from pyocd) (0.25)
Requirement already satisfied, skipping upgrade: pyusb>=1.0.0b2 in c:\python27\lib\site-packages (from pyocd) (1.0.2)
Requirement already satisfied, skipping upgrade: pyyaml in c:\python27\lib\site-packages (from pyocd) (3.13)
Requirement already satisfied, skipping upgrade: pywinusb>=0.4.0 in c:\python27\lib\site-packages (from pyocd) (0.4.2)
Requirement already satisfied, skipping upgrade: enum34 in c:\python27\lib\site-packages (from pyocd) (1.1.6)
Requirement already satisfied, skipping upgrade: sortedcontainers in c:\python27\lib\site-packages (from intervaltree<3.0->pyocd) (2.0.4)
Building wheels for collected packages: pyocd
  Running setup.py bdist_wheel for pyocd ... done
  Stored in directory: C:\Users\sn\AppData\Local\pip\Cache\wheels\80\7b\02\7c3dc5ed7661a196f06b0cdc029298d5674a048737c6cd2f37
Successfully built pyocd
Installing collected packages: pyocd
  Found existing installation: pyOCD 0.12.0
    Uninstalling pyOCD-0.12.0:
      Successfully uninstalled pyOCD-0.12.0
Successfully installed pyocd-0.14.2

C:\Users\sn>pip show pyocd
Name: pyocd
Version: 0.14.2
Summary: Cortex-M debugger for Python
Home-page: https://github.com/mbedmicro/pyOCD
Author: Chris Reed, Martin Kojtal, Russ Butler
Author-email: 
License: Apache 2.0
Location: c:\python27\lib\site-packages
Requires: intervaltree, intelhex, pyelftools, pyyaml, websocket-client, colorama, future, enum34, pywinusb, six, pyusb
Required-by:

C:\Users\sn>mbedgt --version
mbedgt: Not all required Python modules were imported!
mbedgt: No module named pyOCD.board
mbedgt: Check if:
        1. You've correctly installed dependency module using setup tools or pip:
                * python setup.py install
                * pip install <module-name>
        2. There are no errors preventing import in dependency modules
        See: https://github.com/ARMmbed/greentea#installing-greentea```

flit, 2018-12-18:
You also need to upgrade mbed-host-tests to the latest 1.4.4 version. Upgrading mbed-greentea should also upgrade mbed-host-tests.

flit, 2018-12-18:
Great to hear that it's working for you now!

closed 2018-12-18 by flit