pyocd/pyOCD #479: Broken python API example documentation

YifeiZuo01, 2018-12-19:
https://github.com/mbedmicro/pyOCD/blob/057528698f69cbd4e333334b6acb31ef4e3e5c7e/docs/API_EXAMPLES.md#hello-world-example-code

For mbedBoard there is no "flash" attribute already.
```
with ConnectHelper.session_with_chosen_probe() as session:
    board = session.board
    target = board.target
    flash = board.flash
```

flit, 2018-12-19:
Hi @YifeiZuo01, thanks for reporting this issue.

YifeiZuo01, 2018-12-20:
another place which has similar issue 
https://github.com/mbedmicro/pyOCD/blob/master/docs/PYTHON_API.md#notes
>pyocd.flash.flash: flash programming API in the Flash class, accessible from the 'flash' attribute on a target (i.e., session.board.target.flash).

anthrax-0, 2019-02-02:
Hi, not sure if this issue should be reopened or newly created: the current example in the https://github.com/mbedmicro/pyOCD/blob/636721dcdf17c56667374477d7ec9812695f9c67/docs/API_EXAMPLES.md still is not complete - it uses flash_binary() method that was removed if I'm not mistaken?

flit, 2019-02-02:
@anthrax-0 Thanks for catching that!

flit, 2019-04-13:
All docs issues described in this ticket have been fixed.

closed 2019-04-13 by flit