pyocd/pyOCD #489: pyocd causes Linux Desktop to lockup

GlenFlint, 2018-12-31:
Any pyocd (version 0.14.2.dev12) command, including list, causes the Linux Mint 17.3 running the MATE desktop to lockup to the point the computer must be reset.  A terminal window is available and can still run top/htop.  Before switching to MATE that the Cinnamon desktop locked up as well.  I think my problem has something to do with conflicting versions of Python, but don't know where to look for log files or some debug that could help untangle the mess.

flit, 2018-12-31:
How strange! I'm very surprised that pyOCD could cause any such system-level problems. Does it lockup after pyOCD is installed (via pip?) or only after you run the `pyocd` tool?

GlenFlint, 2018-12-31:
There is a work around.
```
arm-none-eabi-objcopy -O ihex main main.hex
```
Copies the executable to a .hex format which can be drag and dropped onto the MICROBIT file system.

flit, 2018-12-31:
I wonder if it could be USB related? It's definitely not a general Linux problem, pyOCD works fine with Ubuntu, which is distro we test with. I noticed that the latest Linux Mint version is 19.1, could you test with that and see if the problem is still reproducible?

It's good to hear you at least have a workaround for flashing.

I'll try to reproduce this when I find time to set up a VM running Linux Mint.