pyocd/pyOCD #493: ST-Link/V2 USB Error

michieldwitte, 2019-01-05:
I can not seem to connect using
0483:3748 STMicroelectronics ST-LINK/V2
on the latest pyOCD (commit: 32e06866f5ac302af3eb03339551a24b533268fb)

When debugging further, it seems a usb issue.
I've tried this on 2 seperate ST-LINK/V2's and verified both of them work.
This is the output when running the basic_test.py when the st-link is first plugged in.

~~~
/usr/bin/python3.6 /home/user/project/pyOCD/test/basic_test.py
WARNING:session:Error attempting to access config file 'test_boards.yaml': [Errno 2] No such file or directory: 'test_boards.yaml'
INFO:board:Target type is cortex_m
DEBUG:stlink.usb:  USB CMD> f1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG:stlink.usb:  USB IN < (6 bytes)
DEBUG:stlink.usb:  USB IN < 27 07 83 04 48 37
DEBUG:stlink.usb:  USB CMD> f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG:stlink.usb:  USB IN < (8 bytes)
Traceback (most recent call last):
  File "/home/user/project/pyOCD/pyocd/probe/stlink/usb.py", line 206, in transfer
    data = self._read(readSize)
  File "/home/user/project/pyOCD/pyocd/probe/stlink/usb.py", line 179, in _read
    data = self._ep_in.read(read_size, timeout)
  File "/home/user/.local/lib/python3.6/site-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/home/user/.local/lib/python3.6/site-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/project/pyOCD/pyocd/probe/stlink_probe.py", line 86, in open
    self._link.open()
  File "/home/user/project/pyOCD/pyocd/probe/stlink/stlink.py", line 67, in open
    self.get_target_voltage()
  File "/home/user/project/pyOCD/pyocd/probe/stlink/stlink.py", line 141, in get_target_voltage
    response = self._device.transfer([Commands.GET_TARGET_VOLTAGE], readSize=8)
  File "/home/user/project/pyOCD/pyocd/probe/stlink/usb.py", line 211, in transfer
    six.raise_from(STLinkException("USB Error: %s" % exc), exc)
  File "<string>", line 3, in raise_from
pyocd.probe.stlink.STLinkException: USB Error: [Errno 32] Pipe error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/project/pyOCD/test/basic_test.py", line 238, in <module>
    basic_test(None, file)
  File "/home/user/project/pyOCD/test/basic_test.py", line 47, in basic_test
    with ConnectHelper.session_with_chosen_probe(unique_id=board_id, **get_session_options()) as session:
  File "/home/user/project/pyOCD/pyocd/core/helpers.py", line 199, in session_with_chosen_probe
    session.open()
  File "/home/user/project/pyOCD/pyocd/core/session.py", line 155, in open
    self._probe.open()
  File "/home/user/project/pyOCD/pyocd/probe/stlink_probe.py", line 89, in open
    six.raise_from(self._convert_exception(exc), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.ProbeError: USB Error: [Errno 32] Pipe error

Process finished with exit code 1
~~~

Any subsequent tries, gives a shorter usb transaction.

~~~
/usr/bin/python3.6/home/user/project/pyOCD/test/basic_test.py
WARNING:session:Error attempting to access config file 'test_boards.yaml': [Errno 2] No such file or directory: 'test_boards.yaml'
INFO:board:Target type is cortex_m
DEBUG:stlink.usb:  USB CMD> f1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Traceback (most recent call last):
  File "/home/user/project/pyOCD/pyocd/probe/stlink/usb.py", line 192, in transfer
    count = self._ep_out.write(paddedCmd, timeout)
  File "/home/user/.local/lib/python3.6/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/home/user/.local/lib/python3.6/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 824, in bulk_write
    timeout)
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/project/pyOCD/pyocd/probe/stlink_probe.py", line 86, in open
    self._link.open()
  File "/home/user/project/pyOCD/pyocd/probe/stlink/stlink.py", line 66, in open
    self.get_version()
  File "/home/user/project/pyOCD/pyocd/probe/stlink/stlink.py", line 81, in get_version
    response = self._device.transfer([Commands.GET_VERSION], readSize=6)
  File "/home/user/project/pyOCD/pyocd/probe/stlink/usb.py", line 211, in transfer
    six.raise_from(STLinkException("USB Error: %s" % exc), exc)
  File "<string>", line 3, in raise_from
pyocd.probe.stlink.STLinkException: USB Error: [Errno 32] Pipe error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/project/pyOCD/test/basic_test.py", line 238, in <module>
    basic_test(None, file)
  File "/home/user/project/pyOCD/test/basic_test.py", line 47, in basic_test
    with ConnectHelper.session_with_chosen_probe(unique_id=board_id, **get_session_options()) as session:
  File "/home/user/project/pyOCD/pyocd/core/helpers.py", line 199, in session_with_chosen_probe
    session.open()
  File "/home/user/project/pyOCD/pyocd/core/session.py", line 155, in open
    self._probe.open()
  File "/home/user/project/pyOCD/pyocd/probe/stlink_probe.py", line 89, in open
    six.raise_from(self._convert_exception(exc), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.ProbeError: USB Error: [Errno 32] Pipe error

Process finished with exit code 1
~~~

The lsusb -v output:
~~~
Bus 001 Device 028: ID 0483:3748 STMicroelectronics ST-LINK/V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0x3748 ST-LINK/V2
  bcdDevice            1.00
  iManufacturer           1 STMicroelectronics
  iProduct                2 STM32 STLink
  iSerial                 3 SÿnPwRQA$
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              4 ST Link
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
~~~

flit, 2019-01-10:
Thanks for reporting the bug! What OS is this under, Linux? Are you running under a VM?

michieldwitte, 2019-01-10:
A Ubuntu 18.04 derivative, not running in a VM.

closed 2019-01-23 by flit

IzidorM, 2019-02-12:
Hi. I have similar problem. When I insert the stlink v2 in to the PC and try to use it I get the following exception:
``` python
# sequence I use in python interpreter
allProbes = ConnectHelper.get_all_connected_probes()
>>> probe = allProbes[0]
>>> session = Session(probe, target_override="NRF52")
>>> session.open()
Traceback (most recent call last):
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink/usb.py", line 206, in transfer
    data = self._read(readSize)
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink/usb.py", line 179, in _read
    data = self._ep_in.read(read_size, timeout)
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink_probe.py", line 86, in open
    self._link.open()
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink/stlink.py", line 67, in open
    self.get_target_voltage()
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink/stlink.py", line 141, in get_target_voltage
    response = self._device.transfer([Commands.GET_TARGET_VOLTAGE], readSize=8)
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink/usb.py", line 211, in transfer
    six.raise_from(STLinkException("USB Error: %s" % exc), exc)
  File "<string>", line 3, in raise_from
pyocd.probe.stlink.STLinkException: USB Error: [Errno 32] Pipe error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/core/session.py", line 155, in open
    self._probe.open()
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink_probe.py", line 89, in open
    six.raise_from(self._convert_exception(exc), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.ProbeError: USB Error: [Errno 32] Pipe error
>>> 
```

If I try to only run session.open() again I get the following exception:
``` python
>>> session.open()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/core/session.py", line 155, in open
    self._probe.open()
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink_probe.py", line 86, in open
    self._link.open()
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink/stlink.py", line 65, in open
    self._device.open()
  File "/home/im/.local/share/virtualenvs/pyocd-J3QNWU30/lib/python3.7/site-packages/pyocd/probe/stlink/usb.py", line 111, in open
    assert self._closed
AssertionError
```

If I exit the python interpreter and run a new one, same thing will happen...

If I run openocd after unsuccessfully running the pyocd, the openocd will normaly connect to the target. The fun thing is that if I close the openocd and try to run above commands again, they will work as expected (no exception)...

My pyocd version is: 0.14.3
My python version is: 3.7.1
libusb version: 1.0.22-1
OS: Arch linux

PR #517 (2019-01-23): fix st-link communication
This fixes the usb problem i was facing connecting to the ST-Link/V2
Fixes #493