pyocd/pyOCD #497: Exception when using with ST Nucleo F401RE

alanbowman, 2019-01-09:
I have just seen that pyOCD supports stlinkv2 and installed version 0.14.3 using pip. I know that the board isn't supported directly, but wanted to try using GDB to debug code. However, when I try to start pyOCD commander using the generic cortex_m target I get an exception. Is this expected to be possible (and I'm missing something) or is this a non-starter?

The exception is as below:
```
0000830:INFO:board:Target type is cortex_m
0000846:INFO:dap:DP IDR = 0x2ba01477
0000857:ERROR:__main__:uncaught exception: 0
Traceback (most recent call last):
  File "/home/alan/Documents/mbed_test/lib/python3.6/site-packages/pyocd/__main__.py", line 309, in run
    self._commands[self._args.cmd]()
  File "/home/alan/Documents/mbed_test/lib/python3.6/site-packages/pyocd/__main__.py", line 490, in do_commander
    PyOCDCommander(self._args, cmds).run()
  File "/home/alan/Documents/mbed_test/lib/python3.6/site-packages/pyocd/tools/pyocd.py", line 548, in run
    status = CORE_STATUS_DESC[self.target.get_state()]
  File "/home/alan/Documents/mbed_test/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 270, in get_state
    return self.selected_core.get_state()
  File "/home/alan/Documents/mbed_test/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 65, in selected_core
    return self.cores[self._selected_core]
KeyError: 0
```

flit, 2019-01-10:
Hi @alanbowman, no, it should work as long as you don't try to program flash. The `KeyError` exception indicates that no core was found, and there is a big chunk of the log missing where the AP and ROM table scan should be.

Can you please try running 'pyocd gdbserver -v' to enable debug logging? This will report what steps in the connect sequence are executing.

alanbowman, 2019-01-10:
Sorry, the last log was an attempt to start pyocd commander - sorry if that confused.  The following log is an attempt to start gdbserver:
```
(mbed_test) alan@cheviot:~/Documents/mbed_test$ pyocd list
## => Board Name | Unique ID
-- -- ----------------------
 0 => STM32 STLinkV2-1 | 066FFF484951775087073136
(mbed_test) alan@cheviot:~/Documents/mbed_test$ pyocd gdbserver -v
0000879:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x1d6b idProduct=0x0003. This is probably a permission issue.
0000880:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x05e3 idProduct=0x0610. This is probably a permission issue.
0000881:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x413c idProduct=0x2107. This is probably a permission issue.
0000881:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x046d idProduct=0xc05a. This is probably a permission issue.
0000884:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x1d6b idProduct=0x0002. This is probably a permission issue.
0000906:INFO:board:Target type is cortex_m
0000924:DEBUG:sequencer:Running task load_svd
0000924:DEBUG:sequencer:Running task create_flash
0000925:DEBUG:sequencer:Running task dp_init
0000927:INFO:dap:DP IDR = 0x2ba01477
0000928:DEBUG:sequencer:Running task power_up
0000930:DEBUG:sequencer:Running task find_aps
0000931:DEBUG:sequencer:Running task create_aps
0000931:DEBUG:sequencer:Running task init_ap_roms
0000932:DEBUG:sequencer:Running task create_cores
0000932:DEBUG:sequencer:Running task create_components
0000932:DEBUG:sequencer:Running task notify
0000933:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x7f70ca0a64e0>
0000933:DEBUG:board:uninit board <pyocd.board.board.Board object at 0x7f70ca0a6160>
0000936:ERROR:__main__:uncaught exception: list index out of range
Traceback (most recent call last):
  File "/home/alan/Documents/mbed_test/lib/python3.6/site-packages/pyocd/__main__.py", line 309, in run
    self._commands[self._args.cmd]()
  File "/home/alan/Documents/mbed_test/lib/python3.6/site-packages/pyocd/__main__.py", line 472, in do_gdbserver
    gdb = gdbs[0]
IndexError: list index out of range
```

flit, 2019-01-11:
Thanks for the log. No worries, commander would be ok too, I just needed the connect sequence.

It's strange, pyOCD is not seeing any CoreSight APs (Access Ports). But low-level SWD transactions must be working because it's able to read the DP's (Debug Port's) ID register. On the STM32F401, there should be a single AHB-AP that is used by the debugger to perform AHB bus transactions.

Could the device be locked?
What version of STLinkV2 firmware is on your board?

alanbowman, 2019-01-12:
The details.txt file in the STLINK drive folder says:
```
Version: 0221
Build:   Nov  4 2015 15:25:25
```
I'm pretty sure it's not locked, as I've programmed a binary using the drag 'n' drop method recently.

alanbowman, 2019-01-12:
Fixed by updating the firmware on the STLink.

Noting how the old the version was, I've updated my STLink firmware.  It now says:
```
Version: 0221
Build:   Sep 18 2018 11:09:21
```

The pyocd log now shows:
```
(mbed_test) alan@cheviot:~/Documents/mbed_test$ pyocd gdbserver -v
0000907:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x1d6b idProduct=0x0003. This is probably a permission issue.
0000908:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x05e3 idProduct=0x0610. This is probably a permission issue.
0000909:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x413c idProduct=0x2107. This is probably a permission issue.
0000910:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x046d idProduct=0xc05a. This is probably a permission issue.
0000920:DEBUG:pyusb_backend:ValueError "The device has no langid" while trying to access dev.product for idManufacturer=0x1d6b idProduct=0x0002. This is probably a permission issue.
0000933:INFO:board:Target type is cortex_m
0000948:DEBUG:sequencer:Running task load_svd
0000948:DEBUG:sequencer:Running task create_flash
0000949:DEBUG:sequencer:Running task dp_init
0000952:INFO:dap:DP IDR = 0x2ba01477
0000953:DEBUG:sequencer:Running task power_up
0000954:DEBUG:sequencer:Running task find_aps
0000955:DEBUG:sequencer:Running task create_aps
0000956:DEBUG:sequencer:Running task create_ap.0
0000957:DEBUG:ap:Using accelerated memory access interface
0000958:INFO:dap:AP#0 IDR = 0x24770011
0000958:DEBUG:sequencer:Running task init_ap_roms
0000959:DEBUG:sequencer:Running task init_ap.0
0000963:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=411)
0000966:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000971:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0000974:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0000976:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0000978:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0000980:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0000981:DEBUG:sequencer:Running task create_cores
0000984:INFO:cortex_m:CPU core is Cortex-M4 r0p1
0000988:INFO:cortex_m:FPU present: FPv4-SP
0001002:DEBUG:sequencer:Running task create_components
0001004:INFO:dwt:4 hardware watchpoints
0001008:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001009:DEBUG:fpb:fpb has been disabled
0001014:DEBUG:sequencer:Running task notify
0001020:INFO:semihost:Telnet server started on port 4444
0001020:INFO:gdbserver:GDB server started on port 3333
^C0010225:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x7f40f4c064e0>
0010226:DEBUG:board:uninit board <pyocd.board.board.Board object at 0x7f40f4c06160>
```

closed 2019-01-12 by alanbowman