pyocd/pyOCD #506: LookupError: setuptools-scm was unable to detect version for

392625227, 2019-01-10:
C:\Users\Administrator\Desktop\pyOCD-master\pyOCD-master>python setup.py install
C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\distutils\dist.py:261: UserWarning: Unknown distributi
on option: 'long_description_content_type'
  warnings.warn(msg)
Traceback (most recent call last):
  File "setup.py", line 91, in <module>
    zip_safe=True,
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\site-packages\setuptools\__init__.py", line 12
9, in setup
    return distutils.core.setup(**attrs)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\distutils\core.py", line 108, in setup
    _setup_distribution = dist = klass(attrs)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\site-packages\setuptools\dist.py", line 372, i
n __init__
    _Distribution.__init__(self, attrs)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\distutils\dist.py", line 281, in __init__
    self.finalize_options()
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\site-packages\setuptools\dist.py", line 528, i
n finalize_options
    ep.load()(self, ep.name, value)
  File "c:\users\administrator\desktop\pyocd-master\pyocd-master\.eggs\setuptools_scm-3.1.0-py3.6.egg\setuptools_scm\int
egration.py", line 23, in version_keyword
    dist.metadata.version = get_version(**value)
  File "c:\users\administrator\desktop\pyocd-master\pyocd-master\.eggs\setuptools_scm-3.1.0-py3.6.egg\setuptools_scm\__i
nit__.py", line 142, in get_version
    parsed_version = _do_parse(config)
  File "c:\users\administrator\desktop\pyocd-master\pyocd-master\.eggs\setuptools_scm-3.1.0-py3.6.egg\setuptools_scm\__i
nit__.py", line 111, in _do_parse
    "use git+https://github.com/user/proj.git#egg=proj" % config.absolute_root
LookupError: setuptools-scm was unable to detect version for 'C:\\Users\\Administrator\\Desktop\\pyOCD-master\\pyOCD-mas
ter'.

Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub
's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.

For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/
user/proj.git#egg=proj

flit, 2019-01-10:
Please update your pip and setuptools to the latest versions, 18.1 and 40.6.3 as of this writing. (Run `pip install -U pip setuptools`.)

392625227, 2019-01-11:
Already latest version of  pip and setuptools when the issue found .

flit, 2019-01-11:
Then I'm not sure what the problem could be. The warning `UserWarning: Unknown distribution option: 'long_description_content_type'` should only happen with old versions of setuptools.

Are you trying to install from a git clone, or a zip file downloaded from GitHub?

closed 2026-01-07 by TeoMahnic