pyocd/pyOCD #508: Error installing 0.14.3 from pip

MaureenHelm, 2019-01-12:
The zephyr project recently upgraded pyocd from 0.12.0 to 0.14.3 in zephyrproject-rtos/zephyr#12444. The upgrade works fine on my and @nashif's local machines, but it does not work in our CI environment:
```
Collecting pyocd==0.14.3 (from -r requirements.txt (line 13))
  Downloading https://files.pythonhosted.org/packages/6c/ae/0baea35a5d061315af00b4b461efa9f1f0c897017230442939f18e9e6885/pyocd-0.14.3.tar.gz (1.0MB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-xcgne0pi/pyocd/setup.py", line 31, in <module>
        long_description=open('README.md', 'r').read(),
      File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 6073: ordinal not in range(128)
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-xcgne0pi/pyocd/
```

More detailed console output is [here](https://app.shippable.com/github/zephyrproject-rtos/docker-image/runs/22/1/console)

MaureenHelm, 2019-01-14:
Thanks @flit. @nashif was able to work around by declaring locale before calling pip.

See docker-library/python#13

flit, 2019-01-15:
@MaureenHelm Fyi, [v0.15.0](
/mbedmicro/pyOCD/releases/tag/v0.15.0) was just released. This version should fix this problem. If you still have issues, please reopen.

closed 2019-01-15 by flit