pyocd/pyOCD #521: Backport enum34 fix

drogga, 2019-01-28:
Would it be possible to backport `'enum34;python_version<"3.4"'` requirement to pyOCD 0.12.x? Otherwise it does not work on Python>=3.4, and pyOCD 0.13+ has incompatible API.

flit, 2019-01-28:
Hi @drogga. I thought you could still install enum34 under Python >=3.4, and it just wouldn't do anything. Are you seeing an error or exception that you can provide?

drogga, 2019-01-29:
Yes, it overrides standard enum and breaks some parts of the standard library.
I encountered this type of issue: https://stackoverflow.com/questions/43124775/why-python-3-6-1-throws-attributeerror-module-enum-has-no-attribute-intflag

flit, 2019-01-29:
Ok, version [0.12.1](/mbedmicro/pyOCD/releases/tag/v0.12.1) has been released. Please let me know if you have any issues.

closed 2019-01-29 by flit