pyocd/pyOCD #529: flashing nordic nrf52 softdevice

IzidorM, 2019-02-04:
Hi,
I am trying to flash a nordic nrf52832 with pyocd. When I try to flash a fw with softdevice included in same image (or if I flash first softdevice and then fw image or vice versa), the fw doesnt start (To be more precise it stuck somewhere in softdevice address space, so it doesnt jump to the fw) . Same image works if flashed with openocd. Everything even start to work, if I only reflash the softdevice (without the app fw part) with openocd after flashing both with pyocd. I am 
using following 2 commands:

pyocd which doesnt work:
 pyocd flash -t NRF52 -e chip nrf_sd_fw.hex
pyocd version:
0.14.3

openocd which works
~/build/openocd-code2/src/openocd -s ~/build/openocd-code2/tcl/ -f target/nrf52832.cfg -c "init" -c "program nrf_sd_fw.hex" -c "reset" -c "exit"
openocd version: 
Open On-Chip Debugger 0.10.0+dev-00207-g41092636-dirty (2018-01-09-12:21)

If somebody with more experience with pyocd would point me in the direction how to fix this issue, Ill be glad to fix it myself and commit back...

flit, 2019-02-06:
Can you send me the hex files you are trying to program? I've been using an nRF52832 with softdevice (app built via Mbed OS) without any apparent issues. Although, I admit I don't know how the softdevice differs compared to user code as far as flashing and target configuration.

IzidorM, 2019-02-08:
Hi, sorry for late replay. I modified an example from nordic sdk to get a really basic app which only use ble so is board independant and did some tests with it. I figured out that this problem occurs only when using softdevice 6.1.1. So If I flash the 6.1.1 the fw wont start. If I flash softdevice 6.1.0 then everything works ok... I attached the hex files. If everything flashes ok, you will see a device advertising with a name basic_test... I also tested again with openocd and I have no problems there. 

[test_fw.tar.gz](https://github.com/mbedmicro/pyOCD/files/2846274/test_fw.tar.gz)

Here are the commands I use to flash:
For flashing app:
pyocd flash -t NRF52 nrf52832_app_basic_example.hex

For flashing softdevice one of the following commands:
pyocd flash -t NRF52 s132_nrf52_6.1.1_softdevice.hex 
pyocd flash -t NRF52 s132_nrf52_6.1.0_softdevice.hex

I use python version:
Python 3.7.1 on linux
pyocd version:
0.14.3
and I use STMicroelectronics ST-LINK/V2 
pyocd output:
(pyocd) im ~/some_path/pyocd $ pyocd list
;## => Board Name | Unique ID
;-- -- ----------------------
;0 => STM32 STLinkV2 | WÿqIUS($%

flit, 2019-02-08:
How strange that the softdevice version impact it! I'll take a look, thanks for sending the files.

flit, 2019-02-08:
I'm able to reproduce the problem. It will probably take some time to debug. Thanks for reporting the issue!

IzidorM, 2019-02-08:
ok great. So let me know if I can help somehow...