pyocd/pyOCD #530: Low performance on python 3.7

ashy-cypress, 2019-02-05:
Hi,
I am testing cypress boards with pyocd and experienced low performance when using python 3.7. For example flashing hex file takes twice more time on python 3.7 then 2.7 for the same bin:
**PYTHON ==2.7**
(venv) D:\CyP\pyOCD\pyocd_2_7\pyocd>pyocd flash "D:\CyP\pyOCD\pyocd_2_7\pyocd\binaries\l1_cy8c6xx7.bin"
[====================] 100%
0002320:INFO:loader:Programmed 10936 bytes (22 pages) at 5.89 kB/s (22 pages unchanged)

**PYTHON ==3.7**
(venv) D:\CyP\pyOCD\pyocd_3_7\pyocd>pyocd flash "D:\CyP\pyOCD\pyocd_3_7\pyocd\binaries\l1_cy8c6xx7.bin"
[====================] 100%
0006685:INFO:loader:Programmed 10936 bytes (22 pages) at 2.24 kB/s (22 pages unchanged)

Also when running gdb_test.py it happends that tests fails due to  loss of synchronization between gdb-server and client due to slow gdb-server.

[automated_test_results_CYW943012P6EVB-01_0.txt](https://github.com/mbedmicro/pyOCD/files/2831897/automated_test_results_CYW943012P6EVB-01_0.txt)
[output_cy8c6xx7_0.txt](https://github.com/mbedmicro/pyOCD/files/2831898/output_cy8c6xx7_0.txt)

flit, 2019-02-06:
Thanks for the report.

What OS are you testing under? Is the OS running as a VM?

This may somehow be specific to PSoC6 targets, as I'm not seeing the performance difference on other targets.

Target | Version | Result
--|--|--
nRF52832 | Python 3.7 | `Programmed 276960 bytes (69 pages) at 30.83 kB/s (0 pages unchanged)`
nRF52832 | Python 2.7 | `Programmed 276960 bytes (69 pages) at 28.07 kB/s (0 pages unchanged)`
K64F | Python 3.7 | `Programmed 703792 bytes (172 pages) at 18.99 kB/s (0 pages unchanged)`
K64F | Python 2.7 | `Programmed 703792 bytes (172 pages) at 19.00 kB/s (0 pages unchanged)`

Notice that in your tests, pyOCD is not actually programming flash. See the "22 pages unchanged" remark. The results you reported actually show flash read performance over AHB. These numbers are surprisingly low compared to most targets, unless the AHB clock is set very low.

Can you try re-running, but erasing the chip before each test?

For a raw data transfer performance test, try running `test/speed_test.py`. For most chips, RAM and flash read performance is comparable. Here are examples using the same targets as above:

Target | USB | Version | RAM | Flash
--|--|--|--|--
nRF52832 | High speed | Python 3.7 | 144.2 kB/s | 145.8 kB/s
nRF52832 | High speed | Python 2.7 | 144.0 kB/s | 146.0 kB/s
K64F | Full speed | Python 3.7 | 55.6 kB/s | 55.5 kB/s
K64F | Full speed | Python 2.7 | 56.3 kB/s | 56.2 kB/s

All of the above results were obtained under macOS 10.14.2 on a MacBook Pro (15-inch, 2018).

ashy-cypress, 2019-02-06:
Hi thanks for response,

> What OS are you testing under? Is the OS running as a VM?

We are running automated_tests.py on Win10, Mac and Ubuntu, all not virtual machines, the result is same for all OSs.

> Can you try re-running, but erasing the chip before each test?

Yes, here are results (on Win10):
**Python 2.7:**
(venv) D:\CyP\pyOCD\pyocd_2_7\pyocd>pyocd flash -e chip "D:\CyP\pyOCD\pyocd_2_7\pyocd\binaries\l1_cy8c6xxa.bin"
[====================] 100%
0003667:INFO:loader:Programmed 10240 bytes (20 pages) at 3.39 kB/s (0 pages unchanged)

**Python 3.7**
(venv) D:\CyP\pyOCD\pyocd_3_7\pyocd>pyocd flash -e chip "D:\CyP\pyOCD\pyocd_3_7\pyocd\binaries\l1_cy8c6xxa.bin"
[====================] 100%
0013944:INFO:loader:Programmed 10240 bytes (20 pages) at 0.98 kB/s (0 pages unchanged)

> For a raw data transfer performance test, try running `test/speed_test.py`. For most chips, RAM and flash read performance is comparable.

Here are results of speed test we have on Ubuntu:
**Python 2.7:**
------ TEST RAM READ / WRITE SPEED ------
Writing 65536 byte took 2.439 seconds: 26871.317 B/s
Reading 65536 byte took 2.414 seconds: 27144.647 B/s
TEST PASSED

------ TEST ROM READ SPEED ------
Reading 2097152 byte took 85.714 seconds: 24466.847 B/s
TEST PASSED
0102623:INFO:dap:DP IDR = 0x6ba02477
CY8CPROTO-062-4343W #0: finished Speed Test... PASSED (90.911 s)

**Python 3.7**
------ TEST RAM READ / WRITE SPEED ------
Writing 65536 byte took 8.305 seconds: 7891.147 B/s
Reading 65536 byte took 8.095 seconds: 8096.221 B/s
TEST PASSED

------ TEST ROM READ SPEED ------
Reading 2097152 byte took 258.918 seconds: 8099.679 B/s
TEST PASSED
0297260:INFO:dap:DP IDR = 0x6ba02477
CY8CPROTO-062-4343W #0: finished Speed Test... PASSED (276.193 s)

Whole test log of running automated_tests.py on Mac with poth Python 2.7 and 3.7 for psoc targets attached:
[Ub_log.txt](https://github.com/mbedmicro/pyOCD/files/2835519/Ub_log.txt)

Cypress-OpenOCD, 2019-02-15:
Pull request #537 resolves the issue

PR #537 (2019-02-16): usb_backend.read: yield to other threads if receive buffer is empty
usb_backend.read method spins in endless loop waiting for receive thread to read some data from the DAPLink. This waists CPU cycles and causes high CPU load, especially on Python 3.

Yield to other threads by calling sleep(0) if receive buffer is empty.

This change improves read performance no Python 3. As a result, PSoC6 Flash programming performance is improved by a factor of three.

speed_test.py, original code:
Python2:
  Reading 65536 byte took 1.430 seconds: 45814.825 B/s
  Flash programming speed: 18 KB/s
Python3:
  Re [... cut]
closed 2019-02-16 by flit