pyocd/pyOCD #557: Objcopy conversion fails

khariya, 2019-03-04:
The [objcopy](https://github.com/mbedmicro/pyOCD/blob/master/test/test_util.py#L66) fails.
Running the command from directly from Command Prompt results in error:

```
>arm-none-eabi-objcopy -v -I binary -O ihex -B arm C:\Users\<USER>\pyocd\binaries\<bin_file> c:\Temp\<elf_file>
copy from C:\Users\<USER>\pyocd\binaries\<binFile> [binary] to c:\Temp\<elfFile> [ihex]
arm-none-eabi-objcopy: c:\Temp\<elfFile> 64-bit address 0x4b4fa300000000 out of range for Intel Hex file
arm-none-eabi-objcopy:c:\Temp\test.elf: bad value
```

OS: Windows 10
GNU tools: 8-2018-q4-major

flit, 2019-03-04:
Hi @khariya. That's really weird, I haven't seen this before. Can you send me the binary that reproduces this problem?

khariya, 2019-03-04:
No, I haven't tried yet.
Is there a specific version you would like me to try?

flit, 2019-03-04:
Not really, just wondering if it was a regression in the 8-2018-q4-major release. I'm currently using 7-2018-q2-update, mostly because 8 is not officially supported in Mbed OS yet.

khariya, 2019-03-04:
I just tried version 7-2017-q4-major and it works without issues.
Thanks for quick response and I'm sorry for the inconvenience.

closed 2019-03-04 by khariya

flit, 2019-03-04:
No worries! I also tested on Mac and Linux, and neither show the problem. It's only Windows 8-2018-q4-major. I'm in the middle of reporting the bug, will send the issue link when it's done.

flit, 2019-03-04:
Bug reported on the GNU Arm Embedded toolchain: [Bug #1818578](https://bugs.launchpad.net/gcc-arm-embedded/+bug/1818578)

flit, 2019-03-05:
Fyi, the GNU MCU Eclipse ARM Embedded GCC toolchain (based on Arm's release) has a fix for this bug: https://github.com/gnu-mcu-eclipse/arm-none-eabi-gcc/releases/tag/v8.2.1-1.2

Also, seems there was an existing bug for the toolchain: https://bugs.launchpad.net/gcc-arm-embedded/+bug/1810274