pyocd/pyOCD #570: pyocd-gdbserver fails when attempting to use a cmsis-pack

pelrun, 2019-03-12:
I can use targets supplied by a pack when running `pyocd gdbserver`, but the new targets are not available when calling `pyocd-gdbserver` instead.

This isn't a big deal for command line usage, but I'm currently using an IDE extension to launch pyocd and it still uses the old command.

flit, 2019-03-13:
That's weird. The same code is used by both tools to enable packs, so it should work fine. How are you specifying the pack? And what is the `pyocd-gdbserver` command line?

pelrun, 2019-03-14:
```
pyocd-gdbserver --pack c:/Keil_v5/ARM/PACK/.Download/SiliconLabs.EFM32G_DFP.5.7.0.pack -t efm32g890f128
usage: pyocd-gdbserver [-h] [--version] [--config PATH] [--no-config]
                       [--pack PATH] [-p PORT_NUMBER] [-sc {telnet,stdx}]
                       [-T TELNET_PORT] [--allow-remote] [-b BOARD_ID] [-l]
                       [--list-targets] [--json] [-d LEVEL] [-t TARGET] [-n]
                       [-r] [-C VECTOR_CATCH] [-s] [-f FREQUENCY] [-o] [-bh]
                       [-ce | -se] [-u] [-hp] [-fp] [-S] [-G]
                       [-c CMD [CMD ...]] [-da DAPARG [DAPARG ...]]
                       [--elf PATH] [-O OPTION] [--no-deprecation-warning]
pyocd-gdbserver: error: invalid target option 'efm32g890f128'
```
as opposed to
```
pyocd gdbserver --pack c:/Keil_v5/ARM/PACK/.Download/SiliconLabs.EFM32G_DFP.5.7.0.pack -t efm32g890f128
0000878:WARNING:pyusb_v2_backend:OS error getting USB interface string: [Errno 2] Entity not found
0000881:WARNING:pyusb_v2_backend:OS error getting USB interface string: [Errno 2] Entity not found
0000881:WARNING:pyusb_v2_backend:OS error getting USB interface string: [Errno 2] Entity not found
0000905:INFO:pack_target:Loading CMSIS-Pack: c:/Keil_v5/ARM/PACK/.Download/SiliconLabs.EFM32G_DFP.5.7.0.pack
0000909:INFO:board:Target type is efm32g890f128
0000960:WARNING:pyusb_v2_backend:OS error getting USB interface string: [Errno 2] Entity not found
0000964:WARNING:pyusb_v2_backend:OS error getting USB interface string: [Errno 2] Entity not found
0000964:WARNING:pyusb_v2_backend:OS error getting USB interface string: [Errno 2] Entity not found
0001055:INFO:dap:DP IDR = 0x2ba01477
0001059:INFO:dap:AP#0 IDR = 0x24770011
0001061:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=673 part=001)
0001063:INFO:rom_table:[0]<e000e000:SCS-M3 class=14 designer=43b part=000>
0001065:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0001066:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0001068:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0001070:INFO:rom_table:[4]<e0040000:TPIU-M3 class=9 designer=43b part=923 devtype=11 archid=0000 devid=0:0:ca0>
0001071:INFO:cortex_m:CPU core is Cortex-M3 r2p0
0001075:INFO:dwt:4 hardware watchpoints
0001076:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001087:INFO:server:Semihost server started on port 4444
0001088:INFO:gdbserver:GDB server started on port 3333
```

The only difference between the two is I changed `pyocd-gdbserver` to `pyocd gdbserver`.

Listing the available targets results in different behaviour - pyocd-gdbserver still shows the hardcoded targets, whereas pyocd includes the targets defined in the pack file.

flit, 2019-03-14:
Thanks for the detailed info! I'll try to reproduce it.

Listing targets with `pyocd-gdbserver` does not support showing any pack-based targets. The deprecated tools necessarily won't support all features.

Btw, which IDE extension are you using?

pelrun, 2019-03-14:
Yup, I noticed that in the code afterwards.

The extension is https://github.com/Marus/cortex-debug for VSCode. I've already submitted a PR to improve it's pyocd support, but I haven't yet changed it away from using pyocd-gdbserver. Guess I should do that!

flit, 2019-03-16:
Cool! I was going to start experimenting with the cortex-debug extension. I'll take a look at your PR.

flit, 2019-03-22:
Finally figured it out. 😄 `pyocd-gdbserver` uses an argparse type callable to validate the target name passed to `--target` (`validate_target()` in `pyocd/tools/gdb_server.py`). This function raises `InvalidArgumentError` for an unknown target, and is not aware of pack targets.

PR #596 (2019-03-28): Fix selection of pack targets in pyocd-gdbserver
`pyocd-gdbserver` used an argparse type callable, `validate_target()`, to validate the target name passed to `--target`. This function raised `InvalidArgumentError` for an unknown target, and was not aware of pack targets. Thus, attempting to select a pack target would always fail even when it worked in the combined `pyocd` tool. This PR removes `validate_target()` and `InvalidArgumentError`. The combined `pyocd` tool also had a leftover vestige of `InvalidArgumentError` that was removed.

Closes #570
closed 2019-03-28 by flit