pyocd/pyOCD #571: Support for LPC55S69 Needed

maclobdell, 2019-03-12:
cc @flit @mmahadevan108 

Please add support for debugging LPC55S69 with pyOCD.  This would enable debugging with Eclipse, GDB Server, and scripting with Python.

flit, 2019-03-12:
Working on it at this very moment.

galak, 2019-04-04:
I was trying this on a LPCXpresso55S69 and 2 different things:

1. Added this to the board_ids.py, not sure what to do about binary image:

```
diff --git a/pyocd/board/board_ids.py b/pyocd/board/board_ids.py
index c8229e7..9091658 100644
--- a/pyocd/board/board_ids.py
+++ b/pyocd/board/board_ids.py
@@ -45,6 +45,7 @@ BOARD_ID_TO_INFO = {
     "0227": BoardInfo(  "MIMXRT1050-EVKB",      "mimxrt1050_hyperflash", "l1_mimxrt1050-evkb_hyperflash.bin",),
     "0230": BoardInfo(  "FRDM-K20D50M",         "k20d50m",          "l1_k20d50m.bin",       ),
     "0231": BoardInfo(  "FRDM-K22F",            "k22f",             "l1_k22f.bin",          ),
+    "0236": BoardInfo(  "LPCXpresso55S69",      "lpc55s69",         "",                     ),
     "0240": BoardInfo(  "FRDM-K64F",            "k64f",             "l1_k64f.bin",          ),
     "0250": BoardInfo(  "FRDM-KW24D512",        "kw24d5",           "l1_kw24d5.bin"         ),
     "0251": BoardInfo(  "FRDM-KW36",            "kw36z4",           "l1_kw36z.bin",         ),
```

When I try to flash I get:

```
(lpc) Kumars-MacBook-Air:pyOCD galak$ pyocd flash /Users/galak/git/zephyr/samples/hello_world/build/zephyr/zephyr.bin 
0000758:INFO:rom_table:AP#0 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0000768:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0000768:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000779:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000786:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000793:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000801:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000808:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=0:0:ca0>
0000822:INFO:rom_table:AP#1 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0000831:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0000831:INFO:rom_table:  AP#1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000841:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000847:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000854:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000861:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000868:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=0:0:ca0>
0001043:ERROR:__main__:uncaught exception: 
Traceback (most recent call last):
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/__main__.py", line 300, in run
    self._commands[self._args.cmd]()
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/__main__.py", line 376, in do_flash
    file_format=self._args.format)
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/flash/loader.py", line 163, in program
    self._loader.commit()
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/flash/loader.py", line 505, in commit
    keep_unwritten=self._keep_unwritten)
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/flash/flash_builder.py", line 401, in program
    sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/flash/flash_builder.py", line 579, in _compute_sector_erase_pages_and_weight
    self._analyze_pages_with_partial_read()
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/flash/flash_builder.py", line 513, in _analyze_pages_with_partial_read
    self.flash.init(self.flash.Operation.VERIFY)
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/flash/flash.py", line 211, in init
    self.target.reset_and_halt(Target.ResetType.SW)
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/core/coresight_target.py", line 314, in reset_and_halt
    return self.selected_core.reset_and_halt(reset_type)
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/coresight/cortex_m.py", line 904, in reset_and_halt
    xpsr = self.read_core_register('xpsr')
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/coresight/cortex_m.py", line 965, in read_core_register
    regValue = self.read_core_register_raw(regIndex)
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/coresight/cortex_m.py", line 979, in read_core_register_raw
    vals = self.read_core_registers_raw([reg])
  File "/Users/galak/git/pyOCD/lpc/lib/python3.7/site-packages/pyocd-0.18.1.dev74+dirty-py3.7.egg/pyocd/coresight/cortex_m.py", line 1040, in read_core_registers_raw
    assert dhcsr_val & CortexM.S_REGRDY
AssertionError
(lpc) Kumars-MacBook-Air:pyOCD galak$ git diff
```

flit, 2019-04-04:
Flashing the LPC55S69 may not work yet. At the very least, you need to set `-Osmart_flash=0`. This is because this device will fault if you attempt to read erased flash pages, by design. But pyOCD by default will read existing flash contents to try to skip pages that don't change. The `smart_flash` option can disable this.

danielRep, 2019-05-17:
Hi @flit 

Is the support for LPC55S69 still on progress?

flit, 2019-05-17:
@danielRep Yes, it is.

PR #690 (2019-06-19): LPC55S69 flash programming bug fixes.
This patch fixes several issues with the built-in NXP LPC55S69 target:
- Reset and halt works now, for both security states.
- Flash programming works.

As part of the fixes, basic support for getting a core's current security state was added to the Python API, as well as detecting which security states are supported.

Closes #571
closed 2019-06-19 by flit