pyocd/pyOCD #575: Periodically report "The device has no langid" on windows 10

XIVN1987, 2019-03-16:
code:
`
#! python2
import time
import logging
from daplink.probe import stlink_probe

logging.basicConfig(level=logging.DEBUG, format='%(name)s - %(levelname)s - %(message)s')

while True:
    stlink = stlink_probe.StlinkProbe.get_all_connected_probes()
    print stlink, '\n'

    time.sleep(1)
`

output:
`
[<daplink.probe.stlink_probe.StlinkProbe object at 0x03008B10>] 

stlink.usb - DEBUG - ValueError "The device has no langid" while trying to access STLink USB device fields (VID=0483 PID=374b). This is probably a permission issue.
[] 

[<daplink.probe.stlink_probe.StlinkProbe object at 0x03008B10>] 

stlink.usb - DEBUG - ValueError "The device has no langid" while trying to access STLink USB device fields (VID=0483 PID=374b). This is probably a permission issue.
[] 

[<daplink.probe.stlink_probe.StlinkProbe object at 0x03008B10>] 

stlink.usb - DEBUG - ValueError "The device has no langid" while trying to access STLink USB device fields (VID=0483 PID=374b). This is probably a permission issue.
[] 
`

Periodically finding stlink probe success and fail. Does anyone know how to solve this problem?

flit, 2019-03-16:
Thanks for the report and code to reproduce. I haven't seen this issue on Windows yet. I'll try to reproduce this shortly.

What's funny is that the `ValueError` debug log message was intended to help Linux users who did not have udev permissions set to allow access to the debug probe. I don't believe we've seen this kind of permission issue on Windows before.

What type and version of STLink are you using? Is it an on-board STLinkV2, or standalone? Can you please provide the full STLink version number? You can get the full version number from pyOCD with this code:

```py
stlinks = stlink_probe.StlinkProbe.get_all_connected_probes()
for st in stlinks:
    st.open()
    print("STLink %s version: %s" % (st.unique_id, st._link.version_str)
    st.close()
```

XIVN1987, 2019-03-17:
@flit 
Thanks for your reply.

I ran that code on a Windows 7 computer, and the same thing happened

I bought the stlink on taobao. Its code may have been modified.
I will buy a NUCLEO development board with standard stlink to test this code

When I used your code to get the version number, the error disappeared

code
``` python
#! python2
import time
import logging
from pyocd.probe import stlink_probe

logging.basicConfig(level=logging.DEBUG, format='%(asctime)s - %(name)s - %(levelname)s - %(message)s')

while True:
    stlink = stlink_probe.StlinkProbe.get_all_connected_probes()
    print stlink, '\n'

    if stlink:
        stlink = stlink[0]
        stlink.open()
        print "STLink %s version: %s\n" %(stlink.unique_id, stlink._link.version_str)
        stlink.close()

    time.sleep(1)
```
output:
```
[<pyocd.probe.stlink_probe.StlinkProbe object at 0x030E2C90>] 

STLink 0671FF505154787867114458 version: V2-1 v2J32

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x03162870>] 

STLink 0671FF505154787867114458 version: V2-1 v2J32

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x03162410>] 

STLink 0671FF505154787867114458 version: V2-1 v2J32

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x031628B0>] 

STLink 0671FF505154787867114458 version: V2-1 v2J32
```

If I comment out open and close in the code, the error reappears

code:
``` python
#! python2
import time
import logging
from pyocd.probe import stlink_probe

logging.basicConfig(level=logging.DEBUG, format='%(asctime)s - %(name)s - %(levelname)s - %(message)s')

while True:
    stlink = stlink_probe.StlinkProbe.get_all_connected_probes()
    print stlink, '\n'

    if stlink:
        stlink = stlink[0]
        #stlink.open()
        print "STLink %s version: %s\n" %(stlink.unique_id, stlink._link.version_str)
        #stlink.close()

    time.sleep(1)
```
output:
```
[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02FE2C90>] 

STLink 0671FF505154787867114458 version: None

2019-03-17 10:27:59,562 - stlink.usb - DEBUG - ValueError "The device has no langid" while trying to access STLink USB device fields (VID=0483 PID=374b). This is probably a permission issue.
[] 

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02FE2C90>] 

STLink 0671FF505154787867114458 version: None

2019-03-17 10:28:01,724 - stlink.usb - DEBUG - ValueError "The device has no langid" while trying to access STLink USB device fields (VID=0483 PID=374b). This is probably a permission issue.
[] 

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02FE2C90>] 

STLink 0671FF505154787867114458 version: None

2019-03-17 10:28:03,878 - stlink.usb - DEBUG - ValueError "The device has no langid" while trying to access STLink USB device fields (VID=0483 PID=374b). This is probably a permission issue.
[] 

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02FE2C90>] 

STLink 0671FF505154787867114458 version: None
```

I don't know if there is something wrong with the stlink I bought, so I need to buy a standard stlink test

XIVN1987, 2019-03-18:
@flit 

I tested the st-link on the Nucleo development board, and the same situation occurred

But when I modified the get_all_connected_devices function in the pyocd\probe\stlink\usb.py file
from:
``` python
devices = usb.core.find(find_all=True, custom_match=cls._usb_match)
```
to:
``` python
device = usb.core.find(idVendor=0x0483, idProduct=0x374b)
devices = [device] if device else []
```

the error disappeared
```
[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02D3F030>] 

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02D93610>] 

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02D22D10>] 

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02D935F0>] 

[<pyocd.probe.stlink_probe.StlinkProbe object at 0x02D3F030>] 
```

flit, 2019-03-18:
Yes, I'm seeing the same thing, too! Thanks for reporting this issue, and for the debugging. I'm working on a permanent fix.

closed 2019-03-27 by flit

XIVN1987, 2019-03-28:
The code works now, thanks very much.

XIVN1987, 2019-03-28:
@flit 
The code works now, thanks very much.

flit, 2019-03-28:
Thank you for reporting the issue and providing a simple script to reproduce, that helped a lot.

PR #595 (2019-03-27): Debug probe fixes
This PR contains multiple bug fixes for both CMSIS-DAP and STLink debug probes.

- Workaround for apparent libusb issue on Windows that caused STLink devices to intermittently not be seen. Closes #575 
- Workaround for old "mbed CMSIS-DAP" firmware (the precursor to DAPLink) reporting an incorrect USB device class. Closes #588 
- The full range of STLink error responses to DP/AP transfers and high level memory accesses are now handled correctly. This change also addresses failures introduced by the latest STLink firmware (V2J33M25) returning more accurate errors. Closes #581
 [... cut]