pyocd/pyOCD #580: connect_test fails on DISCO_L475VG_IOT01A device

studavekar, 2019-03-18:
pyocd version : 0.18.0

Device : `DISCO_L475VG_IOT01A`

pyocd.yaml

```yaml
probes:
  0666FF555051897267232812: # Probe's unique ID.
    target_override:  stm32l475xg
    test_binary:      stm32l475vg_iot01a.bin
```

## Test command 
`python connect_test.py -d`
## Linux

```python

----- TESTING CONNECT/DISCONNECT -----
Flashing new binary
DEBUG:flash_builder:Analyze time: 0.509709
DEBUG:flash_builder:Chip erase count 30, Page erase est count 1
DEBUG:flash_builder:Chip erase weight 5.592700, Page erase weight 1.696700
DEBUG:flash_builder:Using double buffer page erase program
[====================] 100%
DEBUG:flash_builder:Estimated page erase count: 1
DEBUG:flash_builder:Actual page erase count: 1
INFO:pyocd.flash.loader:Programmed 60748 bytes (30 pages) at 8.35 kB/s (29 pages unchanged)
Verifying target is running
TEST FAILED
```

## Windows
```python
----- TESTING CONNECT/DISCONNECT -----
Flashing new binary
DEBUG:flash_builder:Analyze time: 0.568000
DEBUG:flash_builder:Chip erase count 30, Page erase est count 1
DEBUG:flash_builder:Chip erase weight 5.592700, Page erase weight 1.696700
DEBUG:flash_builder:Using double buffer page erase program
[====================] 100%
DEBUG:flash_builder:Estimated page erase count: 1
DEBUG:flash_builder:Actual page erase count: 1
INFO:pyocd.flash.loader:Programmed 60748 bytes (30 pages) at 8.12 kB/s (29 pages unchanged)
Verifying target is running
TEST FAILED

```

flit, 2019-03-22:
I'm able to reproduce this failure.

studavekar, 2019-03-22:
>On Windows I'm having the issue described in #575 that prevents connect_test.py from executing. It sees the probe once, then hangs waiting for a probe to be connected.

Yeah, I did encounter `Waiting for a debug probe to be connected...`  however wanted to check the behavior of this tests on windows so I manually unplug-plug the device for tests to start.

jeromecoutant, 2019-04-12:
Hi

Got issue while trying to connect DISCO_L475VG_IOT01A

````
$ pyocd gdbserver
0000506:ERROR:__main__:uncaught exception: The device has no langid
Traceback (most recent call last):
  File "build\bdist.win32\egg\pyocd\__main__.py", line 300, in run
    self._commands[self._args.cmd]()
  File "build\bdist.win32\egg\pyocd\__main__.py", line 460, in do_gdbserver
    **sessionOptions)
  File "build\bdist.win32\egg\pyocd\core\helpers.py", line 141, in session_with_chosen_probe
    allProbes = ConnectHelper.get_all_connected_probes(blocking=blocking, unique_id=board_id)
  File "build\bdist.win32\egg\pyocd\core\helpers.py", line 78, in get_all_connected_probes
    allProbes = DebugProbeAggregator.get_all_connected_probes(unique_id=unique_id)
  File "build\bdist.win32\egg\pyocd\probe\aggregator.py", line 32, in get_all_connected_probes
    probes += cls.get_all_connected_probes()
  File "build\bdist.win32\egg\pyocd\probe\cmsis_dap_probe.py", line 73, in get_all_connected_probes
    return [cls(dev) for dev in DAPAccess.get_connected_devices()]
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 456, in get_connected_devices
    all_interfaces = _get_interfaces()
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 56, in _get_interfaces
    v2_interfaces = INTERFACE[USB_BACKEND_V2].get_all_connected_interfaces()
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\interface\pyusb_v2_backend.py", line 179, in get_all_connected_interfaces
    for board in all_devices:
  File "c:\python27\lib\site-packages\usb\core.py", line 1249, in device_iter
    if _interop._all(tests) and (custom_match is None or custom_match(d)):
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\interface\pyusb_v2_backend.py", line 275, in __call__
    cmsis_dap_interface = usb.util.find_descriptor(config, custom_match=match_cmsis_dap_interface_name)
  File "c:\python27\lib\site-packages\usb\util.py", line 190, in find_descriptor
    return _interop._next(desc_iter(**args))
  File "c:\python27\lib\site-packages\usb\_interop.py", line 66, in _next
    return next(iter)
  File "c:\python27\lib\site-packages\usb\util.py", line 183, in desc_iter
    if _interop._all(tests) and (custom_match is None or custom_match(d)):
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\interface\pyusb_v2_backend.py", line 271, in match_cmsis_dap_interface_name
    interface_name = usb.util.get_string(desc.device, desc.iInterface)
  File "c:\python27\lib\site-packages\usb\util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
````

jeromecoutant, 2019-04-12:
Oups... I unplugged my Android phone, and it can connect again to the board...
NB: after few tests, it is not related to DISCO L475

flit, 2019-04-13:
Thanks @jeromecoutant, I created #614 based on your report.