pyocd/pyOCD #581: cortex_test fails on DISCO_L475VG_IOT01A device

studavekar, 2019-03-18:
pyocd version : 0.18.0

Device : `DISCO_L475VG_IOT01A`

pyocd.yaml

```yaml
probes:
  0666FF555051897267232812: # Probe's unique ID.
    target_override:  stm32l475xg
    test_binary:      stm32l475vg_iot01a.bin
```

## Test command 
`python cortex_test.py`
## Linux

```python
    ------ Testing Invalid Memory Access Recovery ------
    reading 0x1000 bytes at invalid address 0x3e000000
    failed to get expected fault
    reading 0x1000 bytes at invalid address 0x3e000001
    failed to get expected fault
    writing 0x00001000 bytes at invalid address 0x3e000000
    failed to get expected fault!
    writing 0x00001000 bytes at invalid address 0x3e000001
    failed to get expected fault!
    r/w 0x00000502 bytes at 0x20000000
    Aligned access pass
    r/w 0x00000502 bytes at 0x20000001
    Unaligned access pass
    TEST FAILED
```

## Windows
```python
------ Testing Invalid Memory Access Recovery ------
reading 0x1000 bytes at invalid address 0x3e000000
  got expected error: SWD/JTAG Transfer Fault @ 0x3e000000-0x3e0007ff
reading 0x1000 bytes at invalid address 0x3e000001
Traceback (most recent call last):
  File "cortex_test.py", line 537, in <module>
    cortex_test(None)
  File "cortex_test.py", line 392, in cortex_test
    target.read_memory_block8(addr_invalid + 1, 0x1000)
  File "C:\repos\pyOCD\pyocd\core\coresight_target.py", line 255, in read_memory_block8
    return self.selected_core.read_memory_block8(addr, size)
  File "C:\repos\pyOCD\pyocd\coresight\cortex_m.py", line 620, in read_memory_block8
    data = self.ap.read_memory_block8(addr, size)
  File "C:\repos\pyOCD\pyocd\core\memory_interface.py", line 73, in read_memory_block8
    mem = self.read8(addr)
  File "C:\repos\pyOCD\pyocd\core\memory_interface.py", line 64, in read8
    return self.read_memory(addr, 8, now)
  File "C:\repos\pyOCD\pyocd\probe\stlink_probe.py", line 290, in read_memory
    six.raise_from(self._convert_exception(exc), exc)
  File "C:\repos\pyOCD\venv\lib\site-packages\six-1.12.0-py2.7.egg\six.py", line 737, in raise_from
    raise value
pyocd.core.exceptions.ProbeError: STLink error (24): AP WDATA error
```

flit, 2019-03-21:
I'm able to reproduce this failure. It only happens on new STLink firmware versions. I previously had V2J28 (current version as of when I first implemented STLink support) installed on my L475 board and the test passed. Updating to V2J33 causes this failure.

Strangely, I see the same failure on Linux (Ubuntu 18.04.2 LTS) as on Windows instead of the different failures you reported.

studavekar, 2019-03-22:
>Strangely, I see the same failure on Linux (Ubuntu 18.04.2 LTS) as on Windows instead of the different failures you reported.

it might be ubuntu version I have `Ubuntu 16.04.2 LTS` 

```

------ Testing Invalid Memory Access Recovery ------
reading 0x1000 bytes at invalid address 0x3e000000
  got expected error: SWD/JTAG Transfer Fault @ 0x3e000000-0x3e0007ff
reading 0x1000 bytes at invalid address 0x3e000001
Traceback (most recent call last):
  File "cortex_test.py", line 537, in <module>
    cortex_test(None)
  File "cortex_test.py", line 392, in cortex_test
    target.read_memory_block8(addr_invalid + 1, 0x1000)
  File "/home/arm/repo/pyOCD/pyocd/core/coresight_target.py", line 266, in read_memory_block8
    return self.selected_core.read_memory_block8(addr, size)
  File "/home/arm/repo/pyOCD/pyocd/coresight/cortex_m.py", line 620, in read_memory_block8
    data = self.ap.read_memory_block8(addr, size)
  File "/home/arm/repo/pyOCD/pyocd/core/memory_interface.py", line 73, in read_memory_block8
    mem = self.read8(addr)
  File "/home/arm/repo/pyOCD/pyocd/core/memory_interface.py", line 64, in read8
    return self.read_memory(addr, 8, now)
  File "/home/arm/repo/pyOCD/pyocd/probe/stlink_probe.py", line 290, in read_memory
    six.raise_from(self._convert_exception(exc), exc)
  File "/home/arm/repo/pyOCD/env/local/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
pyocd.core.exceptions.ProbeError: STLink error (24): AP WDATA error

(env) arm@austin-ci-linux-002:~/repo/pyOCD/test$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.2 LTS
Release:        16.04
Codename:       xenial
```

PR #595 (2019-03-27): Debug probe fixes
This PR contains multiple bug fixes for both CMSIS-DAP and STLink debug probes.

- Workaround for apparent libusb issue on Windows that caused STLink devices to intermittently not be seen. Closes #575 
- Workaround for old "mbed CMSIS-DAP" firmware (the precursor to DAPLink) reporting an incorrect USB device class. Closes #588 
- The full range of STLink error responses to DP/AP transfers and high level memory accesses are now handled correctly. This change also addresses failures introduced by the latest STLink firmware (V2J33M25) returning more accurate errors. Closes #581
 [... cut]
closed 2019-03-27 by flit