pyocd/pyOCD #582: PyYAML `>=4.2b,<5.0` requirement does not seem to be workable

mbolivar, 2019-03-19:
Hi, as part of https://github.com/zephyrproject-rtos/zephyr/issues/14460 I learned the pyOCD setup.py has a strange YAML version range: `'pyyaml>=4.2b1,<5.0'`:

https://github.com/mbedmicro/pyOCD/blob/master/setup.py#L62

This is not a good idea; could you please update to 5.0 or continue working with 3.13? Either one would resolve our issue and I think is a good idea for pyOCD as well. From one of new lead maintainers of PyYAML:

```
There seems to be some suggesting that people use one of the
failed 4.2bx releases to get #74 behavior. This is a bad idea. 3.13
is the current supported release. I could delete the 4.2b-s from PyPI
but I haven't. I almost certainly will after 5.1 goes out.
```

https://github.com/yaml/pyyaml/issues/193#issuecomment-457387449

In other words, the current requirements are explicitly requesting users to install a PyYAML version that the maintainers are calling a "failed" release, and which they are actively discouraging users to install.

FWIW, the changes in the Zephyr tree to support PyYAML v5.0+ were minor and are discussed here: https://github.com/yaml/pyyaml/wiki/PyYAML-yaml.load(input)-Deprecation

Can anyone from pyOCD comment on what's holding the code back from working with 5.0?

flit, 2019-03-20:
Well, I didn't realise that 5.0 was released. We updated the requirement to the latest version when there was a security warning triggered by the `load()` API not being safe. At the time, only version 4.2b1 was released.

I'll update to 5.0 as soon as I can.

closed 2019-03-20 by flit

mbolivar, 2019-04-04:
@flit any idea when a release will be available that includes https://github.com/mbedmicro/pyOCD/pull/583? It looks like 0.18.0 still has the old requirement.

flit, 2019-04-05:
Pretty soon, within the next week. There are a few more fixes I wanted to get in. I will set a deadline for the 0.19.0 release of next Friday 12-Apr regardless of what changes are in. Is that ok?

mbolivar, 2019-04-05:
> Pretty soon, within the next week. There are a few more fixes I wanted to get in. I will set a deadline for the 0.19.0 release of next Friday 12-Apr regardless of what changes are in. Is that ok?

Sounds good, thanks! That's around when Zephyr is targeting the first LTS release, so we're hoping to get the fix into our requirements.txt by then.

MaureenHelm, 2019-04-05:
> Sounds good, thanks! That's around when Zephyr is targeting the first LTS release, so we're hoping to get the fix into our requirements.txt by then.

A day or two earlier would be helpful. cc: @galak

flit, 2019-04-06:
Then I'll get it out no later than Tuesday, maybe this weekend. I can also release further fixes in bugfix releases.

flit, 2019-04-07:
Fyi, [version 0.19.0](https://github.com/mbedmicro/pyOCD/releases/tag/v0.19.0) was just released.

PR #583 (2019-03-20): Update pyyaml requirement to >=5.1.
Closes #582