pyocd/pyOCD #584: ModuleNotFoundError: No module named 'pyocd._version'

CharlesChenArm, 2019-03-21:
window 10, board:NUCLEO_F429ZI
when run python basic_test.py in pyOCD/test directory, it will give a hint as follows:
[image]
work around:
add -version.py in pyOCD\pyocd directory.
the content of -version.py is "version='0.18.0'", the version number is the version of my machine.

Hoohaha, 2019-03-21:
@CharlesChenArm 

Run command `pip install -r requirements-dev.txt` to install pyocd in development mode. 
Maybe it can help you.

CharlesChenArm, 2019-03-21:
@Hoohaha 
I try your way, but it doesn't work.
[image]

Hoohaha, 2019-03-21:
@CharlesChenArm 
I am sorry, the correct file name should be `dev-requirements.txt`

CharlesChenArm, 2019-03-21:
@Hoohaha  it can work, thank you:)

flit, 2019-03-21:
Can this issue be closed?

@Hoohaha Thanks for your help!!

CharlesChenArm, 2019-03-22:
@flit 
Does this solution need to add to our documents?  It is just a suggestion for your reference before closing it:)

Hoohaha, 2019-03-22:
@CharlesChenArm 

https://github.com/mbedmicro/pyOCD/blob/master/docs/developers_guide.md#steps

CharlesChenArm, 2019-03-22:
@Hoohaha  Thank you very much. This issue can be closed:)

closed 2019-03-22 by CharlesChenArm