pyocd/pyOCD #585: dlls of libusb

CharlesChenArm, 2019-03-21:
window 10, board:NUCLEO_F429ZI
when run python basic_test.py in pyOCD/test directory, it will give a hint as follows:
[image]

When I place dlls in MS64 folder to main Python, it still doesn’t work.
Work around:
Place dlls in MS32 folder instead of MS64 folder, so it can work.
I don’t know why it happens. Maybe python, pip or other tools I have installed is 32bit, so it only works unless I use dll in MS32.

flit, 2019-03-21:
What do you mean by "MS32" and "MS64"?

CharlesChenArm, 2019-03-22:
[image]
@flit 
They are folders of libusb which provides dlls as above.

flit, 2019-03-22:
Ahhh… 😄 Thanks!

See [this StackOverflow question](https://superuser.com/questions/358434/how-to-check-if-a-binary-is-32-or-64-bit-on-windows) for how to check if an executable is 32- or 64-bit on Windows.

Both Python 3.7 and 2.7 are 64-bit executables on my system, and the libusb DLL is 64-bit in both cases.

I suspect you must have installed a 32-bit Python.

I'm closing this issue. If you think there is an error with pyOCD or its documentation, please reopen.

closed 2019-03-22 by flit

CharlesChenArm, 2019-03-23:
@flit 
i have check the version of python in my laptop, it is 32bit. Now i download 64bit Python version, the issue doesn't happen again. Thank you very much.

bridadan, 2019-03-23:
here's a bit of tangential question, would it ever be possible to distribute libusb dlls/libs as part of the pip package?

flit, 2019-03-23:
@bridadan It should be possible to create a package that includes the libusb libs for all OSes. You'd also need a wrapper that found the right shared lib given the OS and passed the `backend` parameter to `usb.core.find()`. Imo, it'd be better to have a separate package rather than include this directly in pyOCD. But yeah, I've thought about it 😄 . Although, with binary releases of pyOCD coming soon it probably won't matter much, as libusb will be bundled in the pyOCD executables.

bridadan, 2019-03-25:
Yeah I agree, binary releases seem to be the way to go. Thanks for the response, looking forward to that!