pyocd/pyOCD #586: when running automated_test.py, it will block in gdbserver

CharlesChenArm, 2019-03-21:
window 10, board:NUCLEO_F429ZI
when run python automated_test.py in pyOCD/test directory, it will give a hint as follows:
[image]
it will block in gdbserver and can't execute in a full progress.

flit, 2019-03-21:
Can you please provide the full console log and attach all `.txt` files in the `test/` directory? You can run `gdb_test.py` by itself. Thanks!

(As a general rule, please copy log output as text into the issue description and format as code using \`\`\` fences instead of using screenshots.)

CharlesChenArm, 2019-03-22:
[test_params0.txt](https://github.com/mbedmicro/pyOCD/files/2994611/test_params0.txt)
[output_stm32f429xi_0.txt](https://github.com/mbedmicro/pyOCD/files/2994612/output_stm32f429xi_0.txt)

CharlesChenArm, 2019-03-22:
pyocd version : 0.18.0

Device : **NUCLEO-F429ZI**

pyocd.yaml

probes:
  066AFF565251887067065926: # Probe's unique ID.
    target_override:  stm32f429xi
    test_binary:      nucleo_f429zi.bin

### Test command
python gdb_test.py

flit, 2019-03-22:
There's something wrong with your installation of gcc and/or Python 2.

The `output_stm32f429xi_0.txt` file you attached has only this one line:

```
ImportError: No module named site
```

A normal `output_*.txt` file starts with the `arm-none-eabi-gdb-py` executable printing its version and license info (example below). Then you'll start seeing `Executing command:` many times, with other output mixed in.

Because `arm-none-eabi-gdb-py` isn't even printing its version info, there must be something wrong with its installation. The `ImportError` also indicates that something might be wrong with your Python 2 installation, as `arm-none-eabi-gdb-py` requires Python 2. (Otoh, you are able to run pyOCD and `gdb_test.py`.)

Unfortunately, I can't really help much here… I suggest reinstalling GNU Arm Embedded and Python 2. Make sure you have both tools in your `PATH`.

Example start of a normal `output_*.txt` file from a `gdb_test.py` execution:
```
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
/Users/creed/.gdbinit:46: Error in sourced command file:
Cannot enable the TUI when output is not a terminal
arm-none-eabi-gdb-py is using Python 2.7.10 (default, Aug 17 2018, 19:45:58) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.0.42)]
(gdb) Executing command: set pagination off
Executing command: set confirm off
...
```

CharlesChenArm, 2019-03-23:
@flit 
it is caused by python and GNU-ARM. 
My laptop is window 10. In current GCC_ARM versions, it doesn't provide 64bit version. 
in its readme.txt which is located in \share\doc\gcc-arm-none-eabi\readme.txt.
there are two sentences to describe the relationship between python and GNU-ARM version.
1)[To use gdb python build (arm-none-eabi-gdb-py), you need install 32 bit
python2.7 no matter 32 or 64 bit Windows. Please get the package from
https://www.python.org/download/.]
2) To use gdb python build (arm-none-eabi-gdb-py), you need install 32 bit
python2.7 no matter 32 or 64 bit Windows. Please get the package from
https://www.python.org/download/.

So if you want to run python gdb_test.py on window 10,   you must  install python2 version and GNU-ARM 32bit version. Unless you do this combination, or it will run fail just as mentioned.
But in the link of https://github.com/mbedmicro/pyOCD/blob/master/docs/developers_guide.md#steps,
there has a sentence as follows:
•	Install Python. **Version 3.6.0 or above is preferred**, while version 2.7.9 or above is also supported. Add to PATH.
**it will give developer a wrong hint which can cause user to download python3 version. But as i have mentioned, it will run fail when using GNU_ARM on window 10.**

Anyway, i can run the test successfully now, test is not passed. The results are as follows, please check them. Thank you.
[output_stm32f429xi_0.txt](https://github.com/mbedmicro/pyOCD/files/2999156/output_stm32f429xi_0.txt)
[gdb_test_raw0.txt](https://github.com/mbedmicro/pyOCD/files/2999157/gdb_test_raw0.txt)

flit, 2019-03-27:
Thanks, I've taken a note to update the documentation to be more clear. Though I'm not sure how it's working on my system then, since I definitely have 64-bit Python 2.7 installed. Anyway, for development both Python 3 and Python 2 should be installed.

As for the test failure, it looks like the gdb_test_program is crashing for some reason.

This is the important section of `gdb_test_raw0.txt`. The `c` lines are gdb commands executed by `gdb_script.py`.
```
c set $pc = main
w $Pf=00000020#75
r $OK#9a
w $g#67
r $00000000[snip]#e2
w $m20000000,4#4f
r $80b582b0#fb
w $m80077f6,4#6f
r $0028fad0#25
w $m80077f6,2#6d
r $0028#ca
c c&
w $Z0,20000006,2#9c
r $OK#9a
w $vCont;c#a8
r <Timeout: 0 seconds>$T0b07:6f680020;0d:c0ff0220;0e:edffffff;0f:1a701c4b;thread:1;#3d
w $qXfer:threads:read::0,7fb#d0
r $l<?xml version="1.0"?><!DOCTYPE feature SYSTEM "threads.dtd"><threads><thread id="1">HardFault</thread></threads>#8e
```

1. Before the above sequence, the test code is loaded into RAM (the gdb test code is in `src/gdb_test_program`). 
2. Then PC is set to the beginning of the test code, at the start of RAM (0x20000000 in this case).
3. gdb installs a breakpoint on `main()` at address 0x20000006.
4. The target is resumed.
5. pyOCD immediately sees the target enter HardFault due to vector catch. This is reported to gdb. The PC is at 0x4b1c701a!

Could you help debug this? You can see the full sequence of operations in `test/gdb_script.py`.

CharlesChenArm, 2019-03-28:
[image]
when I debug gdb_script.py in visual stdio,  it has an error like this.

flit, 2019-03-28:
Thanks for trying, but `gdb_script.py` will only run from within gdb-py. 

What I actually meant was if you could help to debug by manually reproducing the commands used in `gdb_script.py` by typing them into the gdb console. You can then examine the system at various points and try to determine why it causes a HardFault.

There may be a way to debug it using VSCode, but you would have to do something like what is described in #471 to manually start up the debug server from within the gdb process and then wait for VSCode to connect. This might be worth attempting!

Thanks!

CharlesChenArm, 2019-03-29:
@flit 
Next week I will spend time continue to study how to run it if I have time. But maybe I can't provide the information on time since I am not familiar with python.

CharlesChenArm, 2019-03-29:
[image]
when I use directly use gdb to debug gdb_scripts.py， there is still a wrong message "ImportError".

flit, 2019-03-29:
Ok, no worries! I'll debug this issue, it just may take a little while to get to.

CharlesChenArm, 2019-03-30:
@flit Thank you very much. If you have results on this issue, you can teach me by writing it on comment.  I can have a chance to learn from it:)