pyocd/pyOCD #590: calling pyocd flash with file that does not exists causes unexpected exception

jupe, 2019-03-25:
In above case `payload_NUCLEO_L073RZ.hex` file does not exists in given folder so assuming cli API would return with proper error message. Now behaviour are totally unexpected because there is no any mention about missing file.

```
$ pyocd flash --pack ~/Downloads/Keil.STM32L0xx_DFP.2.0.1.pack -t stm32l073rz ../bin/payload_NUCLEO_L073RZ.hex
0000707:INFO:pack_target:Loading CMSIS-Pack: /Users/jusvat01/Downloads/Keil.STM32L0xx_DFP.2.0.1.pack
0000969:ERROR:__main__:uncaught exception: local variable 'file_obj' referenced before assignment
Traceback (most recent call last):
  File /lib/python2.7/site-packages/pyocd/__main__.py", line 304, in run
    self._commands[self._args.cmd]()
  File "/lib/python2.7/site-packages/pyocd/__main__.py", line 379, in do_flash
    file_format=self._args.format)
  File "/lib/python2.7/site-packages/pyocd/flash/loader.py", line 138, in program
    file_obj.close()
UnboundLocalError: local variable 'file_obj' referenced before assignment
```

flit, 2019-03-25:
Thanks for the bug report. Will get to it quickly.

flit, 2019-03-25:
Depends on #587 being merged. Otherwise you just get a `NameError` for missing `ArgumentError`.

PR #591 (2019-03-25): Fix pyocd flash error for invalid file
When calling `pyocd flash` with an invalid path you'd get a less than helpful exception about `file_obj` not being defined. This PR reworks `FileProgrammer` error checking so `FileNotFoundError` is raised if the given path is invalid. A couple other issues are also addressed, including the cause of the `file_obj` exception.

Closes #590
closed 2019-03-25 by flit