pyocd/pyOCD #592: Are there any examples for using pyOCD with JTAG only devices

jiez, 2019-03-25:
Hi, I'm looking for some examples for using pyOCD with JTAG only devices. I have a target with Cortex-M core which only supports JTAG. I tried to use pyOCD and a CMSIS-DAP adapter to connect to it, but not successful. I'm wondering maybe I should manually select JTAG mode. Is there an example for manually select JTAG mode? Thanks.

The CMSIS-DAP adapter I'm using does support both JTAG and SWD modes. I can use OpenOCD to connect to the target successfully.

flit, 2019-03-25:
Hi! No, I'm sorry but pyOCD doesn't truly support JTAG at the moment. We've never encountered a device that only supports JTAG and not SWD, and SWD is more efficient than JTAG. So there really has not been a reason. This is the first time it's been requested (I think).

You *probably* can get it to work with a few small changes.

- Add a user option to select between SWD or JTAG.
- Probably need user options to set the JTAG device count and IR length. Then the [DAP_JTAG_Configure](http://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__JTAG__Configure.html) CMSIS-DAP command needs to be sent with these option values. The command is already [implemented in `CMSISDAPProtocol`](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/probe/pydapaccess/cmsis_dap_core.py#L353) but would need to be exposed at higher levels.
- Ideally would also support scan-chain auto-detection using [DAP_JTAG_Sequence](http://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__JTAG__Sequence.html), but this not strictly necessary. (STLink also supports JTAG but only supports auto-detect.)
- At [pyocd/coresight/dap.py:92](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/coresight/dap.py#L92) it should check that SWD is selected before executing `self.link.swj_sequence()`.

That might be all! 😄 

If you make progress, we would love to see a pull request to enable JTAG! I'll also put JTAG support on the roadmap.

jiez, 2019-03-25:
Thanks. I saw #110 has been merged in 2015. I thought pyOCD might have already supported JTAG only devices.

flit, 2019-03-25:
Interesting! I wasn't aware of that PR. Unfortunately it was a long time ago and it seems parts of the JTAG support introduced by that PR were broken and/or dropped along the way. I'll create a new issue to fix that regression, and to get JTAG testing integrated into our CI system.

flit, 2019-03-25:
Actually, I was mostly wrong. It looks like JTAG support wasn't dropped, I was just looking at the wrong location. [`DAPAccessCMSISDAP.swj_sequence()`](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py#L666) has the logic to init JTAG. So all that's missing are the user options and STLink JTAG support. I created #593 to track this.

You could hack in JTAG support by changing [pyocd/coresight/dap.py:91](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/coresight/dap.py#L91) to pass [`DebugProbe.Protocol.JTAG`](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/probe/debug_probe.py#L26). However, this has not been tested in quite some time, probably since #110 was introduced. So no telling if there are other issues that will prevent it from working. If you do encounter other problems, please create GitHub issues so we can get them fixed.

Sorry for my incorrect first response!

jiez, 2019-03-25:
It's OK. Thanks for you detailed replies. I have further questions. Instead of hack, what's the proper way to add a JTAG only target? Does pyOCD already support multiple JTAG devices on the same scan chain now?

flit, 2019-03-25:
It looks like right now it's hard coded for a single device with an IR length = 4. This is in the [`DAPAccessCMSISDAP.swj_sequence()`](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py#L674) function where it calls `self._protocol.jtag_configue(4)`. If you changed this call you can adjust IR length and number of devices (device count is the second optional param).

The proper way to implement this is detailed in #593. Basically it needs new user options to control JTAG vs SWD, IR length, and device count.

The actual support for multiple devices on the scan chain is handled entirely in the CMSIS-DAP firmware. The DAP_JTAG_Configure command just tells it the number.

flit, 2019-04-13:
Is it ok to close this issue now that #593 has been added to track implementing the rest of the features required to properly support JTAG? Thanks.

closed 2019-04-25 by jiez