pyocd/pyOCD #601: New target: MBED FF-LPC546XX

j3ll3, 2019-03-31:
Please add support for the MBED FF-LPC546XX board (NXP LPC54606).
Board ID 8081 is currently not supported so it is not possible to flash the target.

flit, 2019-04-04:
The LPC54608 is a supported target, completely compatible with the LPC54606. Did you try setting the target type?

j3ll3, 2019-04-07:
Yes I modified the /pyocd/board/board_ids.py file and added the following line:
 "8081": BoardInfo(  "mbed LTEK FF-LPC546xx","lpc54608",         "l1_lpc54608.bin",     ),

This in my opinion not the nicest way since the board uses a LPC54606 controller instead of a LPC54608 controller. If the l1_lpc54608.bin is compatible with the complete LPC546xx series is it a possibility to refactor the file and code to l1_lpc546xx.bin? I can make a pull request for this.

flit, 2019-05-16:
You can use a CMSIS-Pack to add for specifically the LPC54606. Just run `pyocd pack -i lpc54606`. Then just set the target type to "lpc54606" either via the command line or config file.

PR #704 (2019-07-28): Handful of additional board IDs
Closes #601
closed 2019-07-28 by flit