pyocd/pyOCD #604: Support for S5JS100 Needed

andrewc-arm, 2019-04-03:
cc @flit @mmahadevan108

Please add support for debugging S5JS100 (id: 3701) with pyOCD. This would enable debugging with Eclipse, GDB Server, and scripting with Python.

flit, 2019-04-05:
Thanks! I forgot the name of the S111 board was S5JS100, so I was thinking it might be a different MCU. 😄

andrewc-arm, 2019-04-08:
Hi, @flit 

From this file: https://github.com/ARMmbed/mbed-os-rjio/blob/25be2e52948409727f29d353637beb28c9b2f192/targets/TARGET_Samsung/S111_FW_Write_190103_mbed.zip

There is OpenOCD way to debug as S111_FW_Write_190103_mbed\OpenOCD\openocd\windows_64\fusing_dbg.bat
This enables OpenOCD debugging. Any chance you can reference that OpenOCD example and make it work for PyOCD?

If you need S111 reference manual, any chance you can elaborate which information is needed? As far as I know, Samsung may not be able to provide the reference manual but is able to answer any HW/embedded related questions.

flit, 2019-04-08:
Thanks, I'll look.

We require that the reference manual be _publicly available_ for any device added to pyOCD (or DAPLink), since this is an open source project. Anyone in the community should be able to see where the target support data came from.

As far as information from the RM, I need the detailed block diagram, memory map, and debug architecture details. Often other information is useful, but you don't know until you start testing and debugging.

flit, 2019-04-08:
The OpenOCD scripts help, but I need the reference manual to understand what they are doing and why. For instance, what does the `cmu_init` routine do? What registers is it writing, and why? I need to be able to distinguish between code in the OpenOCD scripts that is required for basic target connection and control, versus code used for flashing the firmware. The latter would be handled by an external script, not part of pyOCD itself.

flit, 2019-04-13:
Hi @andrewc-arm , I'll also need an SVD file for the S111.

andrewc-arm, 2019-04-15:
Hi, @flit 

I searched about how to create SVD file. 
This is the file type you are looking for?
https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_Format_pg.html

I will discuss with Samsung on creating this file.
Do you have any tips on creating this file?
Do you think SVD is related with CMSIS-Pack?

Thanks,

andrewc-arm, 2019-04-15:
Hi, @jh6186
We might need your assistance on getting these files to enable pyOCD + S111.
* S111 reference manual (register map, debugging level)
* SVD file

Thanks,

jh6186, 2019-04-15:
Hi,

We already shared S111 reference manual, kimmo or joseph may get this long ago. please double check.
and
what is SVD file?

flit, 2019-04-15:
The link @andrewc-arm provided above documents the SVD file format. It is a standard format in the Cortex-M ecosystem for describing memory mapped registers. All Cortex-M debuggers support this format, and it will shortly become a requirement for pyOCD target support.

andrewc-arm, 2019-04-19:
Hi, @jh6186 
There are Arm Keil related tools at this location.
https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/Utilities/Win32
I observe that SVDConv.exe is there.

flit, 2019-04-19:
@jh6186 SVD is the standard format for memory mapped register definitions in the Arm ecosystem. Pretty much all debuggers support this format. The docs are here: http://arm-software.github.io/CMSIS_5/SVD/html/index.html.

With #618, all new builtin targets in pyOCD need an SVD file.

andrewc-arm, 2019-04-23:
Hi, @flit 
Since Samsung provided us way to debug with OpenOCD + DAPLink. There is no urgent need for this enablement. If possible, could you please consider this issue instead?
https://github.com/ARMmbed/Samsung_projects/issues/3
Thanks!

40Grit, 2019-05-23:
@andrewc-arm

If I'm not mistaken standard headers for the memory mapped peripherals are also generated from the .svd does this target not have cmsis files generated?

I'd assume it would be a bit hard to have mbed os support unless the standard headers were generated. 

It would not bode well to me if Samsung truly didn't know about .svd.

juhhov, 2019-08-21:
Hi @flit, @0xc0170

What are the current plans with this board? Is there ongoing to effort to support it?
Thanks.

andrewc-arm, 2019-08-22:
Hi, @juhhov 
In the separate discussion, @flit and @0xc0170 mentioned that we need a CMSIS-pack from Samsung. Unfortunately, they are not ready yet. When they get it ready, maybe we can resume this ticket.

RomanSaveljev, 2019-08-22:
@andrewc-arm I can read from the comments above that Samsung has provided a way to debug through OpenOCD. Is there any intention to provide a pack, which will essentially enable the same feature, but with pyOCD?

andrewc-arm, 2019-08-23:
Hi, @RomanSaveljev 
I answered your question from our [JIRA case](https://jira.arm.com/browse/IOTCUST-281).

40Grit, 2019-08-23:
New pyOCD?

andrewc-arm, 2019-08-23:
@40Grit 
I just updated my previous comment about 'new pyOCD'. It was found to be incorrect. 😊

0xc0170, 2019-12-05:
I set this in progress, making progress to enable this one.

40Grit, 2019-12-05:
Do they have their packs in order now?

andrewc-arm, 2019-12-06:
Hi. The pack is ready but the Samsung team is going through legal checks on the pack if it can be publicized. The pack is available in Arm for now.

loverdeg, 2020-08-18:
@flit @andrewc-arm I beleive this can be closed?

closed 2020-10-10 by flit