pyocd/pyOCD #611: STLink error crash while wifi test with DISCO_L475VG_IOT01A

jeromecoutant, 2019-04-12:
Context:
- mbed-os tls test in wifi with DISCO_L475VG_IOT01A
- ARMC6

pyOCD is crashing when wifi transfer is started

````
$ pyocd gdbserver
0000536:INFO:board:Target type is cortex_m
0000566:INFO:dap:DP IDR = 0x2ba01477
0000583:INFO:dap:AP#0 IDR = 0x24770011
0000599:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=415)
0000615:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000615:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0000630:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0000630:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0000646:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0000646:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0000662:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0000677:INFO:cortex_m:FPU present: FPv4-SP
0000677:INFO:dwt:4 hardware watchpoints
0000694:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000740:INFO:server:Semihost server started on port 4444
0000740:INFO:gdbserver:GDB server started on port 3333
0008457:INFO:gdbserver:One client connected!
0008852:INFO:gdbserver:Attempting to load Argon
0008868:INFO:gdbserver:Attempting to load RTX5
0008883:INFO:gdbserver:RTX5 loaded successfully
STLink error (17): AP fault
Traceback (most recent call last):
  File "build\bdist.win32\egg\pyocd\gdbserver\gdbserver.py", line 714, in resume
    if self.target.get_state() == Target.TARGET_HALTED:
  File "build\bdist.win32\egg\pyocd\coresight\cortex_m.py", line 914, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
  File "build\bdist.win32\egg\pyocd\coresight\cortex_m.py", line 604, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "build\bdist.win32\egg\pyocd\probe\stlink_probe.py", line 224, in read_memory
    result = conversion.byte_list_to_u32le_list(self._link.read_mem32(addr, 4, self._apsel))[0]
  File "build\bdist.win32\egg\pyocd\probe\stlink\stlink.py", line 334, in read_mem32
    return self._read_mem(addr, size, Commands.JTAG_READMEM_32BIT, self.MAXIMUM_TRANSFER_SIZE, apsel)
  File "build\bdist.win32\egg\pyocd\probe\stlink\stlink.py", line 291, in _read_mem
    raise exc
TransferFaultError: SWD/JTAG Transfer Fault @ 0xe000edf0-0xe000edf7
0021806:ERROR:gdbserver:Unhandled exception in handle_message: SWD/JTAG Transfer Fault @ 0xe000ed30-0xe000ed37
Traceback (most recent call last):
  File "build\bdist.win32\egg\pyocd\gdbserver\gdbserver.py", line 538, in handle_message
    reply = handler(msg[msgStart:])
  File "build\bdist.win32\egg\pyocd\gdbserver\gdbserver.py", line 767, in v_command
    return self.v_cont(cmd)
  File "build\bdist.win32\egg\pyocd\gdbserver\gdbserver.py", line 818, in v_cont
    return self.resume(None)
  File "build\bdist.win32\egg\pyocd\gdbserver\gdbserver.py", line 734, in resume
    val = ('S%02x' % self.target_facade.get_signal_value()).encode()
  File "build\bdist.win32\egg\pyocd\gdbserver\context_facade.py", line 144, in get_signal_value
    if self._context.core.is_debug_trap():
  File "build\bdist.win32\egg\pyocd\coresight\cortex_m.py", line 1254, in is_debug_trap
    debugEvents = self.read_memory(CortexM.DFSR) & (CortexM.DFSR_DWTTRAP | CortexM.DFSR_BKPT | CortexM.DFSR_HALTED)
  File "build\bdist.win32\egg\pyocd\coresight\cortex_m.py", line 604, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "build\bdist.win32\egg\pyocd\probe\stlink_probe.py", line 224, in read_memory
    result = conversion.byte_list_to_u32le_list(self._link.read_mem32(addr, 4, self._apsel))[0]
  File "build\bdist.win32\egg\pyocd\probe\stlink\stlink.py", line 334, in read_mem32
    return self._read_mem(addr, size, Commands.JTAG_READMEM_32BIT, self.MAXIMUM_TRANSFER_SIZE, apsel)
  File "build\bdist.win32\egg\pyocd\probe\stlink\stlink.py", line 291, in _read_mem
    raise exc
TransferFaultError: SWD/JTAG Transfer Fault @ 0xe000ed30-0xe000ed37
````

Here is information in gdb:

````
(gdb) c
Continuing.
warning: Remote failure reply: E01
Remote failure reply: E01
(gdb)
````

flit, 2019-04-13:
How odd! Thanks for reporting the bug.

Can you please provide a link to the test code?
What version of pyOCD is this?

rqg0717, 2019-04-29:
I have encountered the similar issue. I am using pyOCD 0.19.0 and a mbed sample code here:[mbed-os-example-nfc](https://github.com/ARMmbed/mbed-os-example-nfc/tree/master/NFC_EEPROM)
Please advise. Thank you.

flit, 2019-04-29:
@rqg0717 Thanks for the info, I'll try to reproduce.

flit, 2019-04-29:
@rqg0717 Did the mbed-os-example-nfc project build without changes for you? I had to add `#include <stdio.h>` to `main.cpp`. I'm using the GCC_ARM toolchain.

flit, 2019-04-29:
With pyOCD 0.20.0 I cannot reproduce this. I'm able to flash the NFC example, run the gdbserver, connect with gdb, resume execution, halt, etc.

Btw, I noticed in the original log above that the target type was `cortex_m`. Prior to 0.20.0, you need to set the target type when using STLink probes. That doesn't account for the error—I still can't reproduce it when setting the target to `cortex_m`.

rqg0717, 2019-05-03:
@flit Thank you for letting me know. I will upgrade pyOCD from 0.19.0 to 0.20.0 and try it again.

jeromecoutant, 2019-08-20:
No more crash with pyOCD in latest branch

closed 2019-08-20 by jeromecoutant