pyocd/pyOCD #612: STM32F401RE CMSIS-Pack fails to load

flit, 2019-04-12:
The log from the failure:

```
> pyocd gdb -t stm32f401re --pack /Users/matost01/Downloads/CMSIS/STM32F4xx_DFP.2.13.0.pack
0000355:INFO:pack_target:Loading CMSIS-Pack: /Users/matost01/Downloads/CMSIS/STM32F4xx_DFP.2.13.0.pack
0000457:INFO:board:Target type is stm32f401re
0000464:WARNING:flash_algo:Not enough space for flash algorithm
0000464:ERROR:__main__:uncaught exception: unsupported operand type(s) for -: 'tuple' and 'int'
Traceback (most recent call last):
File "/Users/matost01/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/__main__.py", line 304, in run
self._commands[self._args.cmd]()
File "/Users/matost01/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/__main__.py", line 463, in do_gdbserver
**sessionOptions)
File "/Users/matost01/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/core/helpers.py", line 196, in session_with_chosen_probe
session = Session(allProbes[0], options=options, **kwargs)
File "/Users/matost01/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/core/session.py", line 127, in __init__
or Board(self, self._options.get('target_override', None))
File "/Users/matost01/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/board/board.py", line 47, in __init__
self.target = TARGET[self._target_type](session)
File "/Users/matost01/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/target/pack/pack_target.py", line 31, in _pack_target__init__
super(self.__class__, self).__init__(session, self._pack_device.memory_map)
File "/Users/matost01/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/target/pack/cmsis_pack.py", line 454, in memory_map
self._build_flash_regions()
File "/Users/matost01/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/target/pack/cmsis_pack.py", line 386, in _build_flash_regions
length = nextStart - start
TypeError: unsupported operand type(s) for -: 'tuple' and 'int'```

matt-oster, 2019-04-15:
@flit 

I quickly changed line 385 of target/pack/cmsis_pack.py from:
```python
nextStart = packAlgo.sector_sizes[i + 1]
```
to:
```python
nextStart, nextBlocksize = packAlgo.sector_sizes[i + 1]
```
to work around the issue. Whether that is a sensible change in the grand scheme of things...

flit, 2019-04-15:
@matt-oster, thanks! This is ultimately caused by the 128 kB sectors on this device being so large that the flash algo and data buffer can't fit in available RAM. This will be fixed shortly when I update the pack algo support to take advantage of support for different erase/program sizes.

PR #627 (2019-04-25): Flash enhancements and fixes primarily for STM32 support
This PR includes a number of flash related fixes. The most important is making use of erase sector vs program page sizes for .FLM flash algos coming from CMSIS Packs. This fixes issues with STM32 devices that have such large sector sizes that sector buffer won't fit in RAM. Now pyOCD will use the page size declared in the algo for buffers. This also enabled double buffer support, increasing flash programming speed.

Other included changes.
- The `debug.log_flm_info` user option will cause details of an .FLM flash algo to be logged when i [... cut]
closed 2019-04-25 by flit