pyocd/pyOCD #613: UTF decode error for STLink interface name

flit, 2019-04-12:
(Reported by the Mbed Studio team.)

There is currently an issue with STLink versions and PyOCD. So far we have observed that any version starting with V2J28Mxx is OK when using `pyocd list -p`. Any version prior to that causes PyOCD to crash. Oddly, one board had version V2J28Mxx (with a low number after the M) that worked with PyOCD, but didn't work with Mbed ls.
 
TODO: create a ticket to create documentation directing ST users to update their STLink if we show support for their board, but it is still not working.
 
This is what PyOCD shows when attempting `pyocd list -p` with older (V2J27 and below) STLink versions:

```
pyocd list -p
0000387:ERROR:__main__:uncaught exception: 'utf16' codec can't decode byte 0x6c in position 30: truncated data
Traceback (most recent call last):
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/__main__.py", line 304, in run
self._commands[self._args.cmd]()
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/__main__.py", line 331, in do_list
ConnectHelper.list_connected_probes()
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/core/helpers.py", line 100, in list_connected_probes
allProbes = ConnectHelper.get_all_connected_probes(blocking=False)
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/core/helpers.py", line 78, in get_all_connected_probes
allProbes = DebugProbeAggregator.get_all_connected_probes(unique_id=unique_id)
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/probe/aggregator.py", line 32, in get_all_connected_probes
probes += cls.get_all_connected_probes()
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/probe/cmsis_dap_probe.py", line 73, in get_all_connected_probes
return [cls(dev) for dev in DAPAccess.get_connected_devices()]
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 449, in get_connected_devices
all_interfaces = _get_interfaces()
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 56, in _get_interfaces
interfaces += INTERFACE[USB_BACKEND_V2].get_all_connected_interfaces()
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/probe/pydapaccess/interface/pyusb_v2_backend.py", line 178, in get_all_connected_interfaces
for board in all_devices:
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/usb/core.py", line 1249, in device_iter
if _interop._all(tests) and (custom_match is None or custom_match(d)):
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/probe/pydapaccess/interface/pyusb_v2_backend.py", line 274, in __call__
cmsis_dap_interface = usb.util.find_descriptor(config, custom_match=match_cmsis_dap_interface_name)
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/usb/util.py", line 190, in find_descriptor
return _interop._next(desc_iter(**args))
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/usb/_interop.py", line 66, in _next
return next(iter)
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/usb/util.py", line 183, in desc_iter
if _interop._all(tests) and (custom_match is None or custom_match(d)):
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/pyocd/probe/pydapaccess/interface/pyusb_v2_backend.py", line 256, in match_cmsis_dap_interface_name
interface_name = usb.util.get_string(desc.device, desc.iInterface)
File "/Users/matost01/Library/Python/2.7/lib/python/site-packages/usb/util.py", line 330, in get_string
return buf[2:buf[0]].tostring().decode('utf-16-le')
File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/encodings/utf_16_le.py", line 16, in decode
return codecs.utf_16_le_decode(input, errors, True)
UnicodeDecodeError: 'utf16' codec can't decode byte 0x6c in position 30: truncated data```

flit, 2019-04-12:
@schstm Do you happen to have any insight into this problem? It seems like the USB interface name on these versions of STLink is truncated to an odd number of bytes. Could this be related to #607? Thanks!

flit, 2019-04-13:
@matt-oster Could you please try out the changes in #616 to see if this bug is fixed? Thanks!

matt-oster, 2019-04-15:
Thanks @flit, this fixes the issue that I was seeing with multiple ST boards!

flit, 2019-04-15:
Thanks for testing!

closed 2019-04-15 by flit