pyocd/pyOCD #614: "ValueError: The device has no langid" caused by Android phone

flit, 2019-04-13:
The below exception traceback was reported when running `pyocd gdbserver`. After an Android phone was unplugged, the board was successfully connected to.

This seems likely to be a regression after #595 was merged.

```
$ pyocd gdbserver
0000506:ERROR:__main__:uncaught exception: The device has no langid
Traceback (most recent call last):
  File "build\bdist.win32\egg\pyocd\__main__.py", line 300, in run
    self._commands[self._args.cmd]()
  File "build\bdist.win32\egg\pyocd\__main__.py", line 460, in do_gdbserver
    **sessionOptions)
  File "build\bdist.win32\egg\pyocd\core\helpers.py", line 141, in session_with_chosen_probe
    allProbes = ConnectHelper.get_all_connected_probes(blocking=blocking, unique_id=board_id)
  File "build\bdist.win32\egg\pyocd\core\helpers.py", line 78, in get_all_connected_probes
    allProbes = DebugProbeAggregator.get_all_connected_probes(unique_id=unique_id)
  File "build\bdist.win32\egg\pyocd\probe\aggregator.py", line 32, in get_all_connected_probes
    probes += cls.get_all_connected_probes()
  File "build\bdist.win32\egg\pyocd\probe\cmsis_dap_probe.py", line 73, in get_all_connected_probes
    return [cls(dev) for dev in DAPAccess.get_connected_devices()]
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 456, in get_connected_devices
    all_interfaces = _get_interfaces()
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 56, in _get_interfaces
    v2_interfaces = INTERFACE[USB_BACKEND_V2].get_all_connected_interfaces()
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\interface\pyusb_v2_backend.py", line 179, in get_all_connected_interfaces
    for board in all_devices:
  File "c:\python27\lib\site-packages\usb\core.py", line 1249, in device_iter
    if _interop._all(tests) and (custom_match is None or custom_match(d)):
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\interface\pyusb_v2_backend.py", line 275, in __call__
    cmsis_dap_interface = usb.util.find_descriptor(config, custom_match=match_cmsis_dap_interface_name)
  File "c:\python27\lib\site-packages\usb\util.py", line 190, in find_descriptor
    return _interop._next(desc_iter(**args))
  File "c:\python27\lib\site-packages\usb\_interop.py", line 66, in _next
    return next(iter)
  File "c:\python27\lib\site-packages\usb\util.py", line 183, in desc_iter
    if _interop._all(tests) and (custom_match is None or custom_match(d)):
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\interface\pyusb_v2_backend.py", line 271, in match_cmsis_dap_interface_name
    interface_name = usb.util.get_string(desc.device, desc.iInterface)
  File "c:\python27\lib\site-packages\usb\util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
```

Reported by @jeromecoutant.

flit, 2019-04-13:
@jeromecoutant Could you please test with the changes in #616? I don't have an Android phone to try to reproduce this, but I've added a workaround. Thanks!

jeromecoutant, 2019-05-14:
Issue is no more reproducible with master branch

flit, 2019-05-15:
@jeromecoutant Excellent, thanks very much for testing!

closed 2019-05-15 by flit